ARMmbed/mbed-os #6205: Hardware interrupts not working with EFM32 and tickless

amq, 2018-02-25:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
Tested on EFM32PG12

**Toolchain:**
GCC_ARM

**Toolchain version:**
6-2017-q2-update

**mbed-cli version:** 
1.3.0

**mbed-os sha:**
caeaa49d68c67ee00275cece10cd88e0ed0f6ed3

**Expected behavior**

Hardware interrupts should wake the MCU from deep sleep.

**Actual behavior**

Tickless actually works quite well on EFM32. Even complex and time-sensitive communications are fine, with i2c, spi, uart. Even cellular. But the problem came from an unexpected thing: regular hardware interrupts.

**Steps to reproduce**

```
InterruptIn interrupt(Pin);
motion.rise(callback(this, &isr));
```

`isr()` never gets called while the MCU is in deep sleep. As soon as it wakes and stays in a longer communication (e.g. cellular), it gets called without issues.

amq, 2018-02-25:
| Approach                                              | Result                           |
|-------------------------------------------------------|----------------------------------|
| `MBED_TICKLESS`                                       | InterruptIn not working, else OK |
| `MBED_TICKLESS` + `Thread::attach_idle_hook(&sleep);` | OK                               |
| `Thread::attach_idle_hook(&sleep);`                   | Crash on first Thread::wait()    |

0xc0170, 2018-02-26:
@ARMmbed/team-silabs Please review

stevew817, 2018-02-28:
@amq Do you have a full reproducer you can point me to?

amq, 2018-02-28:
@stevew817

`mbed_app.json`
```
{
    "target_overrides": {
        "EFM32PG12_STK3402": {
            "target.macros_add": [
                "MBED_TICKLESS"
            ],
            "events.use-lowpower-timer-ticker": true
        }
    }
}
```

`main.cpp`
```
#include "mbed.h"

InterruptIn interrupt(BUTTON1);
DigitalOut led(LED1);

void isr() {
    led = !led; // never reached unless you uncomment Thread::attach_idle_hook()
}

int main() {
    interrupt.rise(callback(&isr));

    // Thread::attach_idle_hook(&sleep);
}
```

amq, 2018-03-07:
@stevew817 does it reproduce for you?

stevew817, 2018-03-08:
Sorry, I've been out sick the last couple of days. Will try to reproduce now.

stevew817, 2018-03-08:
@amq I'm unable to reproduce using your reproducer. The button interrupt fires as it should.

stevew817, 2018-03-08:
See energy trace: 
[image]

Energy usage is as expected, with ~4uA in deepsleep with LED off, and ~0.5mA in deepsleep with LED on. You can see the current spikes when it wakes up to handle the button press.

Why there is a spike every 40ms, I'm not sure.

stevew817, 2018-03-08:
@amq If you're seeing interrupts fail as part of a bigger circuit, it would be helpful to know for which pins you've registered interrupt handlers. There is a limitation on Silicon Labs parts as to which pins can be used simultaneously for interrupts on mbed.
short version: you can't register handlers for pins on different ports that have the same pin number. I.e. GPIO IRQ's on PA0 and PB1 is fine, but not PA0 and PB0.

stevew817, 2018-03-08:
For reference, I did a "mbed new ." and then just created the two files you pasted in this ticket. Works both on regular and debug build, and with or without the attach_idle_hook call.
This is on mbed-os [91e6db1](https://github.com/ARMmbed/mbed-os/commit/91e6db1ea251ffcc973001ed90477f42fdca5751). Reverting to caeaa49 didn't cause any failures either.

amq, 2018-03-08:
@stevew817 thank you for looking into this so closely! I'm enabling these pins as interrupts: PC8, PC9, PI3. The strangest thing is that when the MCU exits EM2, the interrupts are firing, so it's not like the devices which should be creating them couldn't be configured because of some communication issue.

amq, 2018-03-08:
Maybe relevant, here is my `device_peripherals.h`:

```
/* us ticker */
#define US_TICKER_TIMER         TIMER0
#define US_TICKER_TIMER_CLOCK   cmuClock_TIMER0
#define US_TICKER_TIMER_IRQ     TIMER0_IRQn

/* PWM */
#define PWM_TIMER        TIMER1
#define PWM_TIMER_CLOCK  cmuClock_TIMER1
#define PWM_ROUTE        TIMER_ROUTE_LOCATION_LOC1

/* Crystal calibration */
#if !defined(CMU_HFXOINIT_STK_DEFAULT)
#define CMU_HFXOINIT_STK_DEFAULT CMU_HFXOINIT_DEFAULT
#endif

#define EMU_NO_DCDC
```

stevew817, 2018-03-08:
You checked you're not calling sleep() anywhere from within a critical/atomic section? If the end of the critical section [here](https://github.com/ARMmbed/mbed-os/blob/79bd2638b8a2626905fdb1f542946b34cc9dce04/hal/mbed_sleep_manager.c#L193) doesn't make the counter reach zero (and thus re-enable interrupts), then you wouldn't see the ISR until whatever is still holding a critical section lock gets triggered into releasing it.

amq, 2018-03-08:
Looks like you are spot on.

I don't have `sleep()` anywhere, but what I'm seeing is that when an interrupt is generated (this is a motion sensor, so when I move it), a breakpoint [here](https://github.com/ARMmbed/mbed-os/blob/caeaa49d68c67ee00275cece10cd88e0ed0f6ed3/platform/mbed_critical.c#L98) gets triggered, with `interrupt_enable_counter` equal to 1.

This is before a breakpoint [here](https://github.com/ARMmbed/mbed-os/blob/caeaa49d68c67ee00275cece10cd88e0ed0f6ed3/rtos/TARGET_CORTEX/mbed_rtx_idle.cpp#L207) gets triggered, after which `interrupt_enable_counter` is equal to 0.

amq, 2018-03-08:
Possibly relevant: https://github.com/ARMmbed/mbed-os/pull/5346

stevew817, 2018-03-08:
Well, another way for this to manifest itself would be waiting a thread... If the kernel suspends a thread because the thread is e.g. waiting for an event or a kernel-provided timeout, then it activates the idle handler which at this point will go ahead and call sleep(). If you hold a critical section while asking your thread to wait (e.g. suspend your thread from a kernel's POV), it will cause the behaviour you described.

amq, 2018-03-08:
Maybe there is a simpler explanation to what I described. The idle loop looks like this:

```
core_util_critical_section_enter
core_util_critical_section_enter
hal_deepsleep
core_util_critical_section_exit
core_util_critical_section_exit
```

So no wonder that when I first hit a breakpoint in `core_util_critical_section_exit`, the `interrupt_enable_counter` is 1.

To describe the problem differently: I can see that the MCU exits EM2 on interrupts, but for some reason, the ISR is not executed, even though `interrupt_enable_counter` is 0 after the second `core_util_critical_section_exit`. Maybe the whole process is not fast enough and the device manages to clear the interrupt?

stevew817, 2018-03-08:
Good questions. Can you check whether the function called `GPIO_EVEN_IRQHandler` (or ODD, depending on the pin number) actually gets executed? If yes, then we're looking at a different issue then if it is not called at all. the _IRQHandler functions are the IRQ's entry point from the NVIC.

amq, 2018-03-08:
Yes, `GPIO_EVEN_IRQHandler()` is called.

amq, 2018-03-08:
And even `InterruptIn::_irq_handler()` is called, but no switch case is triggered (???)

amq, 2018-03-08:
The problem was in using `interrupt.rise()` instead of `interrupt.fall()`.

@stevew817, I'm terribly sorry for your time, I owe you a beer.

amq, 2018-03-08:
Seems to be timing after all. The peripheral's interrupt is active high (confirmed that `read()` returns 0 when there is no interrupt) and by default, makes a 50 µs pulse, which is not detected when using tickless and the default idle loop.

Possible workarounds:
- configure the peripheral to keep the interrupt longer or until it's cleared. Clearing can be dangerous, because if for some reason we miss the interrupt handler or the queue gets full, the clearing may not happen, and the interrupt will stay active forever
- use the falling edge, which sounds like a bad option for many cases, since we are losing the information between rise and fall

A side question: does Silicon Labs plan to support LPTIM?

stevew817, 2018-03-09:
50 us might indeed be a short enough pulse to trip up the logic [here](https://github.com/ARMmbed/mbed-os/blob/f67fe4a7a31666469d5487bc81b2cfc2dec16edc/targets/TARGET_Silicon_Labs/TARGET_EFM32/gpio_irq_api.c#L66). The reasoning behind having that logic there is to not fire interrupts out-of-sequence, but perhaps that logic should just be removed there and the burden of checking your pin state matches what you expect moved over to the user.
Can you remove that line on your local copy (and the corresponding checks on isRise) and verify it starts behaving as expected?

To your side question: yes, when the time comes and the API is finalised, we will support any new API that is relevant to the featureset of the chips we have supported on mbed.

amq, 2018-03-09:
Removing that line and checks for `isRise` solves my problem.

stevew817, 2018-03-09:
Would you be willing to submit a PR?

amq, 2018-03-09:
Done.

closed 2018-03-15 by cmonr