ARMmbed/mbed-os #6206: undefined reference to `__wrap__free_r' and others

ensonic, 2018-02-25:
### Description
- Type: Bug / Question
- Related issue: `#3673`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
LPC11U68

**Toolchain:**
GCC_ARM

**Toolchain version:**
/usr/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919

**mbed-cli version:** 
1.4.0

**Steps to reproduce**
```
sudo apt install gcc-arm-none-eabi
sudo pip2 install mbed-cli
mbed config -G ARM_PATH "/usr/bin/"
mbed import https://ensonic@os.mbed.com/teams/Pokitto-Community-Team/code/Bitmap/
cd Bitmap
mbed toolchain GCC_ARM
mbed target LPC11U68
mbed compile
Building project Bitmap (LPC11U68, GCC_ARM)
Scan: .
Scan: env
Scan: mbed
Link: Bitmap
./BUILD/LPC11U68/GCC_ARM/PokittoLib/mbed-pokitto/common/error.o: In function `error':
error.c:(.text.error+0x6): undefined reference to `__wrap_exit'
./BUILD/LPC11U68/GCC_ARM/PokittoLib/mbed-pokitto/common/retarget.o: In function `__cxa_pure_virtual':
retarget.cpp:(.text.__cxa_pure_virtual+0x4): undefined reference to `__wrap_exit'
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-malloc.o): In function `malloc':
/build/newlib-XAuz1P/newlib-2.4.0.20160527/build_nano/arm-none-eabi/armv6-m/newlib/libc/stdlib/../../../../../../newlib/libc/stdlib/malloc.c:215: undefined reference to `__wrap__malloc_r'
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-malloc.o): In function `free':
/build/newlib-XAuz1P/newlib-2.4.0.20160527/build_nano/arm-none-eabi/armv6-m/newlib/libc/stdlib/../../../../../../newlib/libc/stdlib/malloc.c:222: undefined reference to `__wrap__free_r'
collect2: error: ld returned 1 exit status
[ERROR] ./BUILD/LPC11U68/GCC_ARM/PokittoLib/mbed-pokitto/common/error.o: In function `error':
error.c:(.text.error+0x6): undefined reference to `__wrap_exit'
./BUILD/LPC11U68/GCC_ARM/PokittoLib/mbed-pokitto/common/retarget.o: In function `__cxa_pure_virtual':
retarget.cpp:(.text.__cxa_pure_virtual+0x4): undefined reference to `__wrap_exit'
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-malloc.o): In function `malloc':
/build/newlib-XAuz1P/newlib-2.4.0.20160527/build_nano/arm-none-eabi/armv6-m/newlib/libc/stdlib/../../../../../../newlib/libc/stdlib/malloc.c:215: undefined reference to `__wrap__malloc_r'
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-malloc.o): In function `free':
/build/newlib-XAuz1P/newlib-2.4.0.20160527/build_nano/arm-none-eabi/armv6-m/newlib/libc/stdlib/../../../../../../newlib/libc/stdlib/malloc.c:222: undefined reference to `__wrap__free_r'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/ensonic/projects/pokitto/Bitmap/.temp/tools/make.py -t GCC_ARM -m LPC11U68 --source . --build ./BUILD/LPC11U68/GCC_ARM" in "/home/ensonic/projects/pokitto/Bitmap"
---
```

Please note that I am new to mbed. I am an experienced linux/embedded developer though. Feel free to ask me about additional details as needed.

0xc0170, 2018-02-26:
> DO NOT import mbed-src anymore, a better version is now included inside PokittoLib

From the lib published on the portal, this library uses mbed-classic https://os.mbed.com/users/Pokitto/code/PokittoLib/file/6d5a63b6d5b6/mbed-pokitto/module.json/, is that correct?

I would recommend asking in the https://os.mbed.com/users/Pokitto/code/PokittoLib/issues/ as this is related to that library.

Retarget wrapping is defined in `mbed_retarget.cpp`. And `mbed_alloc_wrappers.cpp` contains implementation for wrapping allocator calls. These 2 files however might have a different name for older libraries, `retarget.cpp` and possibly alloc wrappers might not exist there, and the implementation might be in the retarget code file (it used to be there previously).

ensonic, 2018-03-08:
Indeed there is
```
find . -name "*.cpp" -exec grep -Hn "__wrap_" {} \;
./PokittoLib/mbed-pokitto/common/retarget.cpp:438:extern "C" int __wrap_main(void) {
```
and it does not have the things from
https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_alloc_wrappers.cpp
I'll need to figure how to either tweak the linker flags or how the default build setup assumes those.

I've asked on the pokitto forum, but they just confirm that I should not have those extra linker flags.

0xc0170, 2018-03-09:
As noted, this might get better support in the pokitto forum for this type of problem (it's in their libraries  that they use).

closed 2018-03-10 by ensonic

JojoS62, 2018-03-11:
the pokitto project uses an very old mbed lib version 107, actual is 159...

I would start a new mbed-classic project and copy the pokitto files to the new mbed project:
```
mbed new . --mbedlib
mbed add https://os.mbed.com/users/mbed_official/code/mbed-dev/
```