ARMmbed/mbed-os #6207: nRF52832 Sluggish App - I2C Timeout?

AGlass0fMilk, 2018-02-26:
### Description
- Type: Bug/Question
- Related issue: #5187
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
TARBGET_MCU_NRF52832
TARGET_NRF5

**Toolchain:**
GCC_ARM

I'm experiencing an issue where after some time of running, my application begins to run _extremely_ slowly. Button presses that should take 3 seconds take 20 seconds to register.

I believe I narrowed this issue down to a few lines of code in the i2c_api.c here:

https://github.com/ARMmbed/mbed-os/blob/107976773c22271617c51b3d037683535b91c7c4/targets/TARGET_NORDIC/TARGET_NRF5/i2c_api.c#L55-L61

When the issue happens, a logic analyzer shows the I2C transactions having 1 second delays between them. Since my application uses blocking I2C reads, this slows that thread down dramatically, sometimes causing watchdog resets.

Can anyone elaborate what would cause this I2C to get stuck in a timeout like this? And why is the default timeout _1 second?_ That seems extremely long for any real time system.

0xc0170, 2018-02-26:
@marcuschangarm Please review

marcuschangarm, 2018-02-26:
@AGlass0fMilk 

The current I2C driver in master has some issues. We've re-implemented it as part of our SDK 14.2 update and it should hit the feature branch either this or next week. However, I don't know when it will hit master.

marcuschangarm, 2018-02-26:
Yes, you can see the feature branch here: https://github.com/ARMmbed/mbed-os/tree/feature-nrf528xx

We are debating how best to communicate the progress of that branch to the community.

AGlass0fMilk, 2018-02-26:
@marcuschangarm That's an exciting update, it looks like a lot of my complaints with the mbed/nRF52x platform have been addressed in the feature branch. I have been working with mbed and nRF52 extensively the past few months and may have a few suggestions.

An interesting idea would be a post-binary hook that reads values in from the mbed_app.json file and lets you set UICR values there (for things like bootloader addr, etc).

AGlass0fMilk, 2018-02-26:
I have confirmed the app slowdown issue was caused by an I2C bus error that then triggered cascading timeouts. Shortening the default I2C timeout made the issue less noticeable, but for some reason every I2C transaction starts failing once it happens once.

I'll be waiting for an official fix from the nRF feature branch.

AGlass0fMilk, 2018-03-07:
@marcuschangarm Can you share the current status of the nrf528xx feature branch? What drivers are still being ported? I see the build warnings for the I2C and SPI. My application needs a fix for existing I2C issues.

marcuschangarm, 2018-03-07:
@AGlass0fMilk 

I have literally opened a PR with the new I2C and SPI driver! 😄 
https://github.com/ARMmbed/mbed-os/pull/6296

Please give it a spin and see if it resolves your issue!

We've created this forum post to keep track of the current progress and solicit feedback from the community: https://os.mbed.com/forum/upcoming-features/topic/29477/

AGlass0fMilk, 2018-03-07:
@marcuschangearm Awesome, thanks! I'll comment on the PR if I notice any issues.

marcuschangarm, 2018-03-07:
Thank you! I only have a handful of devices to test the driver on after all! 😄

AGlass0fMilk, 2018-03-08:
@marcuschangarm I merged in PR #6308 and now my app works as expected. Haven't seen the I2C issue pop up yet and it's been running longer than usual. Fingers crossed but it looks good so far! This update couldn't have come at a better time.

marcuschangarm, 2018-03-08:
Excellent! Glad it is working out for you! Please do let us know if anything pops up!

AGlass0fMilk, 2018-03-15:
Resolved by #6296

closed 2018-03-15 by AGlass0fMilk