ARMmbed/mbed-os #6208: Managed Bootloaders for Nordic Platforms

AGlass0fMilk, 2018-02-26:
### Description
- Type: Question
- Related issue: #5950 
- Priority: Minor

What is currently preventing managed bootloaders from being used on the Nordic nRF platform? It is my understanding that mbed-managed bootloaders have to lie before the main app in flash. This contradicts the expectation of the Nordic soft device.

I have spent a few days creating tools for inserting appdata headers into my binaries and would like to take advantage of the new feature in #5950 instead...

0xc0170, 2018-02-26:
@marcuschangarm Please review

marcuschangarm, 2018-02-26:
@AGlass0fMilk 

With the new SDK 14.2 you'll be able to add a custom bootloader in between the SoftDevice and application. You could use the managed bootloader feature to combine the bootloader with application, but you'll then have to combine the new image with the SoftDevice yourself. 

Right now, we are treating the SoftDevice as a managed bootloader and using the new feature for merging it with the application.

SenRamakri, 2018-09-24:
managed bootloader is supported on NRF52 - Chanes from @marcuschangarm are in 5.9

closed 2018-09-24 by SenRamakri