ARMmbed/mbed-os #6213: default_idle_hook - why is deepsleep locked?

MikeDK, 2018-02-26:
### Description
- Type: Question
- Priority: Major

---------------------------------------------------------------

## Question

Some months ago I made a pull request #5328 which fixes energy consumption on SiLabs devices when using EventQueue.

I just re-checked with newer sources, and I found that the device is not entering deep sleep again ... After some research, I found out that my fix is broken since #5581 was merged in.

Now if sleep() gets called, it should be automatically decided if hal_sleep() or hal_deepsleep() gets called, depending on whether deepsleep is locked, or not.

Now since low power event queue usage was still not working, I dug a little bit deeper into the code, and found out that in default_idle_hook() function, there is a sleep_manager_lock_deep_sleep() call right before the sleep() - that way, the device will never go into deep sleep - If I remove the sleep_manager_lock_deep_sleep() call, then using EventQueue will work again with low power.
(I also found out that when using MBED_TICKLESS, some different mechanism is used for the idle hook, and EventQueue usage is working with deepsleep too)

So now my question: Is there some specific reason why deep sleep is locked explicitly in the default_idle_hook() function?

0xc0170, 2018-02-26:
Please see https://github.com/ARMmbed/mbed-os/issues/6199

MikeDK, 2018-02-26:
Wow that was fast ;-)

Thanks for the answer - I guess I will the Thread::attach_idle_hook(&sleep) approach, it seems it is working well here.

closed 2018-02-26 by MikeDK