ARMmbed/mbed-os #6224: mbed_hal-flash tests failing inconsistently for NRF52840_DK

studavekar, 2018-02-27:
### Description
- Type: Bug 
- Related issue: https://github.com/ARMmbed/mbed-os/pull/5831 
- Priority: Minor

---------------------------------------------------------------
## Bug
**Target**
NRF52840_DK

**Toolchain:**
GCC_ARM|ARM|IAR

Failure history :

http://mbed-ci-master-2.austin.arm.com:8081/view/mbed-os-raas/job/text_matrix_campeci/24/target=NRF52840_DK,toolchain=GCC_ARM/testReport/history/

cc : @marcuschangarm

cmonr, 2018-04-11:
For reference, the specific failing test is in the progress of being disabled: https://github.com/ARMmbed/mbed-os/pull/6535

IOTMORF-2289 has also been created internally.

0xc0170, 2018-04-20:
> For reference, the specific failing test is in the progress of being disabled: #6535

No disabling, investigation is needed for this one, latest failure here http://mbed-ci-master-2.austin.arm.com:8081/job/test_matrix_raas/1614/

0xc0170, 2018-05-22:
I believe this one can be closed, resolved via latest updates to nrf codebase. We haven't seen the error since.

cc @marcuschangarm

closed 2018-05-22 by 0xc0170