ARMmbed/mbed-os #6242: CMake export binary output

vidavidorra, 2018-03-01:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement
When exporting to CMake, the compile step does not generate a binary output file. This output type is needed to flash the program on, for example a NUCLEO_F103RB.

**Reason to enhance or problem with existing solution**
At the moment extra steps in a `project.cmake` are needed to create binary output file, which is required for flashing some boards (NUCLEO_F103RB).

**Suggested enhancement**
Two options for enhancement:
1. Add two lines to the current output generation for the .hex ([here](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/cmake/CMakeLists.txt.tmpl#L69-L72)) that generates the binary output
2. Add extra custom command after the .hex output generation here ([here](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/cmake/CMakeLists.txt.tmpl#L69-L72)) that generates the binary output

I think option 1 might be the cleanest as there remains one custom command for output generations, but both would do just fine. If this enhancement is ok, I'd be happy to make a pull request for it.

I've tried it in my `project.cmake` as shown below and it works perfectly. Although it would be cleaner to have this in the `CMakeList.txt.tmpl` by default.

```cmake
add_custom_command(TARGET ${CMAKE_PROJECT_NAME} POST_BUILD
        COMMAND ${ELF2BIN} -O binary $<TARGET_FILE:${CMAKE_PROJECT_NAME}> $<TARGET_FILE:${CMAKE_PROJECT_NAME}>.bin
        COMMAND ${CMAKE_COMMAND} -E echo "-- built: $<TARGET_FILE:${CMAKE_PROJECT_NAME}>.bin"
        )
```

**Pros**
- Binary output by default that can be used for programming

**Cons**

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools Please review

0xc0170, 2021-01-15:
I'll close this. We are slowly moving to CMake (it's already on Master branch). This should be fixed there or if not, a new issue report should be provided.

closed 2021-01-15 by 0xc0170