ARMmbed/mbed-os #6247: USART6 receive interrupt not working on NUCLEO-F401RE

daid, 2018-03-01:
### Description
- Type: Bug
- Priority: Major

When trying to use USART6 (PA_11, PA_12) with a receive interrupt on the NUCLEO-F401RE it never gets the interrupt.
The source of this bug seems to be that the target specific implementation of the STM32F4 assumes the USARTS are in a sequence without gaps. But the STM32F401 has USART1, USART2 and USART6.
The code actually tries to attach to the interrupt of USART3, which does not exist, and fails silently.

---------------------------------------------------------------
## Bug

**Target**
NUCLEO-F401RE (possibly more?)

**Steps to reproduce**

Example, the following code should echo back, but it does not for USART6, it does so correctly for USART2.
```
RawSerial serial(PA_11, PA_12, 115200)

void interrupt()
{
  while(raw_serial.readable())
    serial.putc(serial.getc());
}

int main()
{
    serial.attach(interrupt);
}
```

Quick hack fix, when I apply this it works around the bug and I can use the receive interrupt. But it's not a proper fix.
```
--- a/src/mbed-os/targets/TARGET_STM/TARGET_STM32F4/serial_device.c
+++ b/src/mbed-os/targets/TARGET_STM/TARGET_STM32F4/serial_device.c
@@ -111,7 +111,11 @@ static void uart5_irq(void)
 #if defined(USART6_BASE)
 static void uart6_irq(void)
 {
+#if UART_NUM == 3
+    uart_irq(2);//HACK to get USART6 to work
+#else
     uart_irq(5);
+#endif
 }
 #endif

@@ -187,6 +191,9 @@ void serial_irq_set(serial_t *obj, SerialIrq irq, uint32_t enable)
             break;
 #endif
 #if defined(USART6_BASE)
+#if UART_NUM == 3
+        case 2://HACK to get USART6 to work
+#endif
         case 5:
             irq_n = USART6_IRQn;
             vector = (uint32_t)&uart6_irq;
@@ -394,6 +401,9 @@ static IRQn_Type serial_get_irq_n(serial_t *obj)
             break;
 #endif
 #if defined(USART6_BASE)
+#if UART_NUM == 3
+        case 2://HACK to get USART6 to work
+#endif
         case 5:
             irq_n = USART6_IRQn;
             break;

```

bcostm, 2018-03-01:
It is strange because a fix for USART indexes has been done in PR #5962.

Looking at your code it looks you are not on master. Can you verify this ?

daid, 2018-03-01:
I'm indeed running release 5.7.4, fix is included in 5.7.5 (I tried searching for the bug, but results came up empty)

closed 2018-03-01 by daid