ARMmbed/mbed-os #6253: fatfs: Integrate into config system

geky, 2018-03-02:
From here:
https://os.mbed.com/questions/80344/mbed-os5-includes-user-configurable-head/

ChanFS provides a handful of configuration options in a header file. These should be provided to application developers.
https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/ChaN/ffconf.h

dannybenor, 2018-06-26:
I think it is enough that users can change https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/ChaN/ffconf.h

dannybenor, 2018-06-26:
I think that it is enough that users can change https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/ChaN/ffconf.h. I think this issue can be closed

geky, 2018-06-26:
Unfortunately, users can't change ffconf.h when mbed-os is a binary. Which is the case for the online compiler.

From [here](https://os.mbed.com/questions/80344/mbed-os5-includes-user-configurable-head/):
> Now I cannot modify the file in the Mbed Web online system (I don't want to use off-line compiler).

dannybenor, 2018-07-02:
In my opinion, the online compiler is good for evaluation. When going to real work you should go to the full source

0xc0170, 2018-07-02:
@geky we do not release it as binary, only mbed 2 is being released as binary (it does not include the filesystem). Therefore they should be able to edit the file. I could not locate an example they use to reproduce the problem

yossi2le, 2018-07-02:
@0xc0170 I have looked into blinky example in the online compiler and mbed-os sources are not part of it.

[image]

geky, 2018-07-02:
We should also consider consistency. The standard method of providing configuration in Mbed OS is through the config system.

This is a low priority item, but we should be tracking it somewhere.

SeppoTakalo, 2019-08-26:
I'm closing the internal ticket, we are not going to work on this in the given time.
It can be considered as a future feature requests.

loverdeg-ep, 2019-08-26:
@AGlass0fMilk

closed 2019-08-26 by ciarmcom