ARMmbed/mbed-os #6256: Documentation clarification for EventQueue::call_every

pauluap, 2018-03-02:
I was just working on adding tests to equeue and noticed a behavior that I did not expect.

In context of `events/equeue/tests.c`

I tried running this test - 
```
void dispatch_time_small_does_dispatch(void) {
    equeue_t q;
    int err = equeue_create(&q, 2048);
    test_assert(!err);

    bool touched = false;
    equeue_call_every(&q, 10, simple_func, &touched);

    // normal zero dispatch, verify that call occurs
    equeue_dispatch(&q, 0);
    test_assert(touched);
}
```

At first, I thought that it was because of the dispatch(0) that was doing an early exit, but digging in, it appears that the first call for the event is scheduled in the future, not immediately.

This behavior makes sense for `EventQueue::call_in` but it was surprising to me for `call_every`.  I looked at the documentation at https://os.mbed.com/docs/v5.7/mbed-os-api-doxy/classevents_1_1_event_queue.html#ae2edd53b4d12ab3f01a3c9bda58310f0 and it doesn't say anything one way or another.

I don't know if I have a preference, there's bound to be people in both camps, so probably the best solution is to add to the documentation to make clear when the first event dispatch will actually occur.

0xc0170, 2018-03-05:
@geky Please review

geky, 2018-03-05:
Good point, it's not very clear.

The current behaviour (with immediate delay) matches the behaviour of the hardware ticker classes, so I think we should keep it.

You _can_ create more customized events with both the C++ and C api, although this requires a bit more control over the memory allocation. Here's how you might rewrite your test for periodic events:
``` c
void simple_func_indirect(void *indirect_ptr) {
    bool **ptr = indirect_ptr;
    **ptr = true;
}

void dispatch_time_small_does_dispatch(void) {
    equeue_t q;
    int err = equeue_create(&q, 2048);
    test_assert(!err);

    bool touched = false;
    // allocate event, memory is inlined in event, so indirect pointer is required
    bool **event = equeue_alloc(&queue, sizeof(bool*));
    *event = &touched;
    equeue_event_delay(event, 0);  // configure no delay
    equeue_event_delay(event, 10); // configure 10ms period
    equeue_post(&queue, simple_func_indirect, event);

    // normal zero dispatch, verify that call occurs
    equeue_dispatch(&q, 0);
    test_assert(touched);
}
```

How does this look to you? #6274 

Also cc @kjbracey-arm.

pauluap, 2018-03-05:
Thanks!  Made a comment on the PR, this is good for me, injecting more information in the ecosystem.

kjbracey-arm, 2018-03-06:
I think the current behaviour is what I would naturally expect - "call_every" being a variation on "call_in" that repeats, rather than something with fundamentally different behaviour.

And given the tick-based nature, the "call_in" will be rounding down to N ticks. If the current time is 0.5 ticks, and you say "call_in(10)", the first call will be at 10.0. You've already missed 0.0, so having call_every make an immediate call would give a slight start-up period stutter. call_every has the advantage of a precise tick period otherwise.

(Then we could go nuts and make the first call on next tick 1.0 and second call at 11.0. Getting a bit hard to explain there, and probably not for general API.)

I spent a while on similar calls over in another event library in-tree - read the comments on https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/nanostack-event-loop/eventOS_event_timer.h for my own picky take on "call_in", "call_after", "call_every" and "call_at". Went through very specifically when each thing would occur - (I found phrasing everything in terms of absolute time helped.)

pauluap, 2018-03-06:
Ah, I see.  A matter of assumptions.  I had thought that the event would kick at the moment of call and be exactly periodic thereafter.  But of course, we don't have an attosecond clock but rather a tick clock, so it all makes sense now.

deepikabhavnani, 2018-12-14:
Closing since query was resolved, and documentation was updated, re-open if you feel otherwise

closed 2018-12-14 by deepikabhavnani