ARMmbed/mbed-os #6258: GR-Peach Compile Error Message

Meano, 2018-03-03:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue:  
- Priority:  Major

---------------------------------------------------------------
## Bug

**Target**
RZ_A1H|GR_LYCHEE

**Toolchain:**
GCC_ARM

**Toolchain version:**
GCC ARM 6-2017-Q2

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
25321961f (Since Mbed-os-5.7.4)

**DAPLink version:**
Unconcerned

**Expected behavior**
No Error Message

**Actual behavior**
GR-Peach Compile Show error message Like this 
[image]

**Steps to reproduce**
Compile GR-Peach Program by GCC ARM Toolchain

0xc0170, 2018-03-05:
@ARMmbed/team-renesas Please review

TomoYamanaka, 2018-03-09:
Sorry for late reply. I will check this. I'll get back to you as soon as I know something.

theotherjimmy, 2018-03-14:
@Meano Why is this a "Major" priority? This bug is purely cosmetic.

Meano, 2018-03-15:
@theotherjimmy I've corrected it.

toyowata, 2018-03-15:
@Meano This is not compile error, but error message from memap.py build script here.
https://github.com/ARMmbed/mbed-os/blob/master/tools/memap.py#L134

I've quick checked this with blinky example and I got correct binary image like below:

```
$ mbed import mbed-os-example-blinky
[mbed] Importing program "mbed-os-example-blinky" from "https://github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #91e6db1ea251
$ cd mbed-os-example-blinky/
$ mbed ls
mbed-os-example-blinky (#0b37629a15c8)
`- mbed-os (#91e6db1ea251, tags: latest, mbed-os-5.7.6)
$ mbed compile -m RZ_A1H -t GCC_ARM
Building project mbed-os-example-blinky (RZ_A1H, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Compile [  0.3%]: irq_ctrl_gic.c
[Warning] irq_ctrl_gic.c@183,7: 'cfg' may be used uninitialized in this function [-Wmaybe-uninitialized]
Compile [  0.6%]: AnalogIn.cpp
Compile [  1.0%]: BusIn.cpp
Compile [  1.3%]: BusOut.cpp
Compile [  1.6%]: BusInOut.cpp

(snip)

Compile [ 99.7%]: spi_api.c
[Warning] spi_api.h@35,31: large integer implicitly truncated to unsigned type [-Woverflow]
Compile [100.0%]: test_env.cpp
Link: mbed-os-example-blinky
Elf2Bin: mbed-os-example-blinky
Unknown object name found in GCC map file: linker stubs
Unknown object name found in GCC map file: LONG 0x1800f868 __etext
Unknown object name found in GCC map file: LONG 0x20020000 __data_start__
Unknown object name found in GCC map file: LONG 0xa7c (__data_end__ - __data_start__)
Unknown object name found in GCC map file: LONG 0x180102e4 __etext2
Unknown object name found in GCC map file: LONG 0x20900000 __nc_data_start
Unknown object name found in GCC map file: LONG 0x0 (__nc_data_end - __nc_data_start)
Unknown object name found in GCC map file: LONG 0x20020a80 __bss_start__
Unknown object name found in GCC map file: LONG 0x2f08 (__bss_end__ - __bss_start__)
Unknown object name found in GCC map file: LONG 0x20900000 __nc_bss_start
Unknown object name found in GCC map file: LONG 0x0 (__nc_bss_end - __nc_bss_start)
Text 46.1KB Data 2.62KB BSS 11.8KB ROM 48.7KB RAM 14.4KB
Image: ./BUILD/RZ_A1H/GCC_ARM/mbed-os-example-blinky.bin
```

Can you check whether you got binary imag in your BUILD directory?

I am not sure why the memap.py generate parse error in this case.

cc @geky @theotherjimmy

theotherjimmy, 2018-03-15:
@toyowata It's gerenating a parse error because it's looking for object names and finding other garbage.

toyowata, 2018-03-16:
@theotherjimmy Thank you for your response.  The memap.py doesn't handle mapfile information as below:

```
.copy.table     0x000000001800f840       0x18
                0x000000001800f840                . = ALIGN (0x4)
                0x000000001800f840                __copy_table_start__ = .
                0x000000001800f840        0x4 LONG 0x1800f868 __etext
                0x000000001800f844        0x4 LONG 0x20020000 __data_start__
                0x000000001800f848        0x4 LONG 0xa7c (__data_end__ - __data_start__)
                0x000000001800f84c        0x4 LONG 0x180102e4 __etext2
                0x000000001800f850        0x4 LONG 0x20900000 __nc_data_start
                0x000000001800f854        0x4 LONG 0x0 (__nc_data_end - __nc_data_start)
                0x000000001800f858                __copy_table_end__ = .

.zero.table     0x000000001800f858       0x10
                0x000000001800f858                . = ALIGN (0x4)
                0x000000001800f858                __zero_table_start__ = .
                0x000000001800f858        0x4 LONG 0x20020a80 __bss_start__
                0x000000001800f85c        0x4 LONG 0x2f08 (__bss_end__ - __bss_start__)
                0x000000001800f860        0x4 LONG 0x20900000 __nc_bss_start
                0x000000001800f864        0x4 LONG 0x0 (__nc_bss_end - __nc_bss_start)
                0x000000001800f868                __zero_table_end__ = .
                0x000000001800f868                __etext = .
```

This data table is created by RZ_A1H linker script for GCC_ARM [here](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_RENESAS/TARGET_RZ_A1XX/TARGET_RZ_A1H/device/TOOLCHAIN_GCC_ARM/RZA1H.ld#L98-L120).

JanneKiiskila, 2018-03-22:
Same issue pops up with all apps that you compile for RZ_A1H.

theotherjimmy, 2018-03-22:
I would expect it to pop up for every app + RZ_A1H. I'm glad that you found that to be true.

theotherjimmy, 2018-03-22:
@TomoYamanaka Should these values contribute to the size of the application? Are they safe to just ignore?

TomoYamanaka, 2018-03-23:
@theotherjimmy 
Since the data table processes are need in any application of RZ_A1H, I hope that these values are handled in map.py.

theotherjimmy, 2018-03-23:
@TomoYamanaka There seems to be some misunderstanding here:
 * It's memap.py
 * memap.py generates the rom/ram summary numbers
 * memap.py does not effect the output binary in any way
 * memap.py could crash, burn and return success and your program would still run just fine.

So: do these numbers contribute to the RAM or ROM size of the application in any way? if so, how do they contribute and what section should they contribute to?

TomoYamanaka, 2018-03-27:
Sorry for late reply.

These processes are mandatory for any application, is expended in ROM memory(4 Byte per LONG proc).
Those are used as ROMization information (ex: table of copy information of variable with initial value).
I think that this size should be considered in ROM/RAM memory analytics proc of memap.py.
But it seems that these are recognized as section by mistake and warning message occurs as a result of parsing. Could you change memap.py?

theotherjimmy, 2018-03-27:
> But it seems that these are recognized as section by mistake and warning message occurs as a result of parsing

They're not recognized as sections, they're recognized as objects. The format is normally:
```
[optional section] [address] [size] [optional type] [object file name]
```
In this context, I have no file name to use, as gcc places the linker script snippet in place of the file name.

0xc0170, 2018-05-22:
@ARMmbed/team-renesas Is this issue still relevant ? Does it require any fixes?

toyowata, 2018-05-24:
I agreed with @TomoYamanaka comment.

* Its not an error message, but looks like critical error message for user point of view
* It appears for **all** build code for GR-PEACH and GR-LYCHEE

So, my suggestion is not to show the message "Unknown object name found..." as previous version.

PR #8271 (2018-10-15): Tools, memap: Silence warnings that we handled correctly
### Description

Memap will create a bunch of warnings about "Unknown object name" when
parsing a map file created for the RZ_A1H and the GR_Peach. Theses 
warnings are sperious; the information in these sections does not belong
to an object file. This PR silences these warnings.

Fixes #6258

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
commit d8b7b3a2 (2018-10-15): Tools, memap: Silence warnings that we handled correctly

### Description

Memap will create a bunch of warnings about "Unknown object name" when
parsing a map file created for the RZ_A1H and the GR_Peach. Theses
warnings are sperious; the information in these sections does not belong
to an object file. This PR silences these warnings.

Fixes #6258

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-15 by cmonr