ARMmbed/mbed-os #6259: Error during export MBED to uVision for ARM STMicro Nucleo f103RB

juicerman, 2018-03-04:
### Description
- Type: Bug 

---------------------------------------------------------------
## Bug

**Target**
Nucleo-F103RB

**Toolchain:**
uVision 5

[image]
[image]

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools Please review

theotherjimmy, 2018-05-22:
@juicerman That "error dialog" has been updated to include the error text. Could you run the export again?

cmonr, 2018-06-21:
@juicerman Would you mind looking into the error again?