ARMmbed/mbed-os #6261: MAX32625PICO target missing

Ferdi265, 2018-03-04:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

There is no target for the [MAX32625PICO](https://os.mbed.com/platforms/MAX32625PICO/) device.

Intersting note: the DAPLINK "MBED.HTM" file for the device links to the page for the [MAX32630FTHR](https://os.mbed.com/platforms/MAX32630FTHR/) device, which seems wrong, as the devices are not the same, and a special page for the MAX32625 _does_ exist.

**Target**
MAX32625PICO

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
tag: mbed-os-5.7.6, sha: 91e6db1ea251ffcc973001ed90477f42fdca5751

**Expected behavior**

A MAX32625PICO target should exist, as stated on the mbed platform webpage for the board. (see above)

Quote:
> To compile a program for this board using Mbed CLI, use max32625pico as the target name.

**Actual behavior**

No such target exists. Targets for MAX32625MBED or MAX32630FTHR exist however.

**Steps to reproduce**

- Install mbed-cli
- create a new empty project
- set the target to MAX32625PICO
- try to run mbed compile

0xc0170, 2018-03-05:
@ARMmbed/team-maximintegrated Please review

jessexm, 2018-03-05:
The MAX32625PICO does not yet exist as a mbed platform and the [MAX32625PICO](https://os.mbed.com/platforms/MAX32625PICO/) platform page is not public. Although the MAX32625PICO as a platform pull request is a few days out. The PICO as a platform will ship with the proper platform ID (0416) and embedded links.

I believe the PICO you have is part of a MAX32630FTHR kit and a such is identifying correctly.

When made public the PICO platform page will include DAPLink firmware that can be used to re-target a kit PICO.

Ferdi265, 2018-05-11:
Since the PR (#6779) was merged today I think we can close this.

closed 2018-05-11 by Ferdi265