ARMmbed/mbed-os #6272: STM32L0xx: Timing issues using RTC as lowpower ticker timer

marcemmers, 2018-03-05:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32L073RZ

**mbed-os sha:**
91e6db1ea251ffcc973001ed90477f42fdca5751

----------------------------------------------------------------

We are experiencing some issues when using the RTC as lowpower ticker timer by setting:
```
#define MBED_CONF_EVENTS_USE_LOWPOWER_TIMER_TICKER        1
#define MBED_CONF_TARGET_LOWPOWERTIMER_LPTIM              0
```

The maximum baudrate we can currently achieve is 9600 baud because if we go higher we start to drop frames to overruns. This however is only a result of a deeper issue in the library which could cause a number of issues.

There are two main problems as far as i have seen:

- The number of times the RTC wakeup is configured each time the processor comes out of sleep. It looks like the RTC is written to up to 5 times each time when coming out of sleep.
- The duration of the functions called.

Writing to the RTC wakeup timer is a costly operation because of the synchronization to the LSE/LSI clock domain. Calling `HAL_RTCEx_SetWakeUpTimer_IT()` takes approx 30 - 60us depending on the time it takes to sync the first clock.
On the Cortex M0 and M1 this is even slower because divisions take a lot of time to calculate and the `rtc_set_wake_up_timer(uint32_t delta)` function in rtc_api.c uses division in a while loop. A single pass through the while can take up to 50us when the delta is big enough.

When a character at 115200 takes approx 87us you can see how some characters might be missed.

I have written an improvement to the `rtc_set_wake_up_timer(uint32_t delta)` function but this does not really solve the issues here.

I do understand that this is really in the core of the RTOS and the microcontroller so i don't know if it would be even possible to fix these issues but it might be worthwhile to try and improve this.

0xc0170, 2018-03-05:
@marcemmers Would it be possible to create a small example that would help us to reproduce the problem? I assume sending/receiving characters plus going to sleep with specific baudrates?

@ARMmbed/team-st-mcd Please review

marcemmers, 2018-03-05:
Of course i can. Tomorrow I will attach some timing diagrams using gpio's.
This code could result in: "Tt1a2d8hC\n" when sent "Test1a2b3c4d5e6f7g8h9i0jABC\n" at 115200. It would probably change around a bit but it shows how much is missed.

Main.cpp:
```
#include "mbed.h"

RawSerial com(PB_6, PB_7, 115200);
EventQueue *gpQueue = mbed_event_queue();

// Used for tickless mode without systick
extern "C"
{
    IRQn_Type mbed_get_m0_tick_irqn()
    {
        return LCD_IRQn;
    }
}

char buffer[256] = {0};
uint8_t write = 0;
uint8_t read = 0;

void HandleRx(void)
{
    while(write != read)
    {
        com.putc(buffer[read]);
        read++;
    }
}

void RxCallback(void)
{
    while(com.readable())
    {
        char ch = com.getc();   
        buffer[write++] = ch;
        if(ch == '\n')
        {
            gpQueue->call(HandleRx);
        }
    }
}

int main()
{
    com.attach(RxCallback);
  
    gpQueue->dispatch_forever();
    return 0;
}
```

mbed_config.h:
```
#ifndef __MBED_CONFIG_DATA__
#define __MBED_CONFIG_DATA__

// Configuration parameters
#define MBED_CONF_APP_THREAD_STACK_SIZE                   1024        
#define MBED_CONF_PPP_CELL_IFACE_APN_LOOKUP               0           
#define MBED_CONF_EVENTS_PRESENT                          1           
#define MBED_LFS_PROG_SIZE                                64          
#define MBED_CONF_PLATFORM_STDIO_FLUSH_AT_EXIT            0           
#define MBED_CONF_TARGET_LOWPOWERTIMER_LPTIM              0           
#define MBED_CONF_DRIVERS_UART_SERIAL_RXBUF_SIZE          256         
#define MBED_CONF_APP_MAIN_STACK_SIZE                     2048        
#define MBED_CONF_FILESYSTEM_PRESENT                      0           
#define MBED_CONF_PPP_CELL_IFACE_BAUD_RATE                115200      
#define MBED_LFS_BLOCK_SIZE                               512         
#define MBED_CONF_PPP_CELL_IFACE_AT_PARSER_BUFFER_SIZE    256         
#define MBED_CONF_PLATFORM_FORCE_NON_COPYABLE_ERROR       0           
#define MBED_CONF_PLATFORM_STDIO_BAUD_RATE                9600        
#define CLOCK_SOURCE                                      USE_PLL_HSI 
#define MBED_CONF_EVENTS_SHARED_HIGHPRIO_STACKSIZE        1024        
#define MBED_LFS_READ_SIZE                                64          
#define MBED_LFS_ENABLE_INFO                              0           
#define MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE       9600        
#define MBED_CONF_RTOS_PRESENT                            1           
#define MBED_CONF_EVENTS_SHARED_DISPATCH_FROM_APPLICATION 1           
#define MBED_CONF_EVENTS_SHARED_EVENTSIZE                 1024        
#define MBED_CONF_PPP_CELL_IFACE_AT_PARSER_TIMEOUT        8000        
#define MBED_CONF_EVENTS_SHARED_STACKSIZE                 2048        
#define MBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE          256         
#define MBED_LFS_LOOKAHEAD                                512         
#define MBED_CONF_EVENTS_USE_LOWPOWER_TIMER_TICKER        1           
#define MBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES         0           
#define MBED_CONF_TARGET_LSE_AVAILABLE                    1           
#define MBED_CONF_NSAPI_PRESENT                           1           
#define MBED_CONF_EVENTS_SHARED_HIGHPRIO_EVENTSIZE        512         

// Macros
#define MBED_TICKLESS                                                 
#define NO_SYSTICK                                                    
#define UNITY_INCLUDE_CONFIG_H                                        
#define MBEDTLS_MD5_C                                                 

#endif

```

marcemmers, 2018-03-06:
I have attached a timing diagram when sending "Test1a2b3c4d5e6f7g8h9i0jABC\n" at 115200 baud.

CH 0 is the tx of the processor
CH 1 is the rx of the processor
CH 2 is a irq disable flag by setting on the disable irq and resetting on enable irq
CH 3 is set on entry of `rtc_set_wake_up_timer()` and reset on exit
CH 4 is set before `line 1185 __HAL_RTC_WRITEPROTECTION_DISABLE(hrtc);` and reset after `line 1231 while(__HAL_RTC_WAKEUPTIMER_GET_FLAG(hrtc, RTC_FLAG_WUTWF) == RESET)` in stm32l0xx_hal_rtc_ex.c function `HAL_RTCEx_SetWakeUpTimer_IT()`

Pin setting and resetting is done directly in the gpio register for minimal overhead.
This is all compiled in IAR 8.11.2 with optimization set to high, balanced and without instruction scheduling.

[image]

jeromecoutant, 2018-03-14:
Hi

About the duration of HAL_RTCEx_SetWakeUpTimer_IT, it's should not be possible to reduce.
Check Reference Manual of STM32 chip :

> Programming the wakeup timer
> Poll WUTWF until it is set in RTC_ISR to make sure the access to wakeup auto-reload
> counter and to WUCKSEL[2:0] bits is allowed. It takes around 2 RTCCLK clock cycles
> (due to clock synchronization).

About the time spent in the while loop, I agree we need to optimize.

I tried to measure this time:
- with 1s delta value => 8s
- with 5s delta value => 14s
- with big delta value => 35s

I applied your proposal:
- with 1s delta value => 9s
- with 5s delta value => 10s
- with big delta value => 10s

This is better !

I tried to remove the loop:

    if (delta < (0x10000 * 2 / RTC_CLOCK * 1000000) ) {
        RTC_clock_us = ( (uint64_t)RTC_CLOCK << 32 ) / 1000000;
        WakeUpCounter = (((uint64_t)delta * RTC_clock_us) >> 32) >> 1 ;
        WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV2;
    } else if (delta < (0x10000 * 4 / RTC_CLOCK * 1000000) ) { // 0x10000 (0xFFFF + 1) * 4 (RTCCLK_DIV4) * 1000000 (1s) / 0x8000 (RTC_CLOCK)
        RTC_clock_us = ( (uint64_t)RTC_CLOCK << 32 ) / 1000000;
        WakeUpCounter = (((uint64_t)delta * RTC_clock_us) >> 32) >> 2 ;
        WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV4;
    } else if (delta < (0x10000 * 8 / RTC_CLOCK * 1000000) ) { // 0x10000 (0xFFFF + 1) * 8 (RTCCLK_DIV8) * 1000000 (1s) / 0x8000 (RTC_CLOCK)
        RTC_clock_us = ( (uint64_t)RTC_CLOCK << 32 ) / 1000000;
        WakeUpCounter = (((uint64_t)delta * RTC_clock_us) >> 32) >> 3 ;
        WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV8;
    } else if (delta < (0x10000 * 16 / RTC_CLOCK * 1000000) ) { // 0x10000 (0xFFFF + 1) * 16 (RTCCLK_DIV16) * 1000000 (1s) / 0x8000 (RTC_CLOCK)
        RTC_clock_us = ( (uint64_t)RTC_CLOCK << 32 ) / 1000000;
        WakeUpCounter = (((uint64_t)delta * RTC_clock_us) >> 32) >> 4 ;
        WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV16;
    } else {
        WakeUpCounter = (delta / 1000000) ;
        WakeUpClock = RTC_WAKEUPCLOCK_CK_SPRE_16BITS;
    }

- with 1s delta value => 5s
- with 5s delta value => 5s
- with big delta value => 9s

What do you think ?

marcemmers, 2018-03-14:
I totally agree that it is not possible to reduce the time spent in the SetWakeUpTimer. My biggest issue there is the amount of times it is set per cycle which causes extra delay. This is operating system level and probably just a limit we have to live with for now.

I checked my code when testing last week and the compiler performed the loop unrolling. However, I do agree that your function is cleaner because the unrolling is explicit and not optimization related. Seems to compile into fairly compact code, i like it!

Don't need the RTC_clock_us variable though so would suggest to use a define of some sort:

```
#define RTC_CLOCK_US (((uint64_t)RTC_CLOCK << 32 ) / 1000000)
if (delta < (0x10000 * 2 / RTC_CLOCK * 1000000) ) { // 0x10000 (0xFFFF + 1) * 2 (RTCCLK_DIV2) * 1000000 (1s) / 0x8000 (RTC_CLOCK)
    WakeUpCounter = (((uint64_t)delta * RTC_CLOCK_US) >> 32) >> 1 ;
    WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV2;
} else if (delta < (0x10000 * 4 / RTC_CLOCK * 1000000) ) { // 0x10000 (0xFFFF + 1) * 4 (RTCCLK_DIV4) * 1000000 (1s) / 0x8000 (RTC_CLOCK)
    WakeUpCounter = (((uint64_t)delta * RTC_CLOCK_US) >> 32) >> 2 ;
    WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV4;
} else if (delta < (0x10000 * 8 / RTC_CLOCK * 1000000) ) { // 0x10000 (0xFFFF + 1) * 8 (RTCCLK_DIV8) * 1000000 (1s) / 0x8000 (RTC_CLOCK)
    WakeUpCounter = (((uint64_t)delta * RTC_CLOCK_US) >> 32) >> 3 ;
    WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV8;
} else if (delta < (0x10000 * 16 / RTC_CLOCK * 1000000) ) { // 0x10000 (0xFFFF + 1) * 16 (RTCCLK_DIV16) * 1000000 (1s) / 0x8000 (RTC_CLOCK)
    WakeUpCounter = (((uint64_t)delta * RTC_CLOCK_US) >> 32) >> 4 ;
    WakeUpClock = RTC_WAKEUPCLOCK_RTCCLK_DIV16;
} else {
    WakeUpCounter = (delta / 1000000) ;
    WakeUpClock = RTC_WAKEUPCLOCK_CK_SPRE_16BITS;
}
```

jeromecoutant, 2018-03-21:
Hi

As #6370 is merges, do you agree to close this issue,
or something more is missing ?

Thx

closed 2018-03-21 by 0xc0170