ARMmbed/mbed-os #6284: mbed-os 5.8 fails at parsing mbed_lib.json

LiyouZhou, 2018-03-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: ?

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
(`git log -n1 --oneline`)
6ff720302 Add Mbed version block for new minor release.

Try to parse the following mbed_lib.json
```
{
    "name": "blinky-shield",
    "macros": ["__nop()=asm(\"nop\")"],
    "config": {
        "ws2812_zero_high": 0,
        "ws2812_zero_low": 0,
        "ws2812_one_high": 0,
        "ws2812_one_low": 0
    },
    "target_overrides": {
        "K64F": {
            "ws2812_zero_high": 0,
            "ws2812_zero_low": 5,
            "ws2812_one_high": 5,
            "ws2812_one_low": 0
        }
    }
}
```
but fails, and the error printing did not work properly hence the cli crashed with no useful information.
 I added print statements in the python code to find out which file it failed on.
```
mbed compile -m K64F -t GCC_ARM -vvvv
[mbed] Working path "/Users/liyzho01/work/mbed-cloud-client-example-internal" (program)
[mbed] Exec "/usr/local/opt/python/bin/python2.7 -u /Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v" in /Users/liyzho01/work/mbed-cloud-client-example-internal
Building project mbed-cloud-client-example-internal (K64F, GCC_ARM)
Scan: .
Scan: env
Scan: mbed
Traceback (most recent call last):
  File "/Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/make.py", line 298, in <module>
    stats_depth=options.stats_depth)
  File "/Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/build_api.py", line 574, in build_project
    resources = scan_resources(src_paths, toolchain, inc_dirs=inc_dirs)
  File "/Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/build_api.py", line 484, in scan_resources
    resources = toolchain.config.load_resources(resources)
  File "/Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/config/__init__.py", line 1018, in load_resources
    self.add_config_files(resources.json_files)
  File "/Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/config/__init__.py", line 511, in add_config_files
    for x in errors))
  File "/Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/config/__init__.py", line 511, in <genexpr>
    for x in errors))
  File "/Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/config/__init__.py", line 395, in format_validation_error
    path, str(".".join(error.absolute_path)), error.message)
TypeError: sequence item 1: expected string or Unicode, int found
[mbed] ERROR: "/usr/local/opt/python/bin/python2.7" returned error code 1.
[mbed] ERROR: Command "/usr/local/opt/python/bin/python2.7 -u /Users/liyzho01/work/mbed-cloud-client-example-internal/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v" in "/Users/liyzho01/work/mbed-cloud-client-example-internal"
```

LiyouZhou, 2018-03-06:
@theotherjimmy

theotherjimmy, 2018-03-06:
@LiyouZhou 5.8 is not released yet, so you're probably talking about RC1.

------

That's an error error! Nice. Let me track that down.

theotherjimmy, 2018-03-06:
Real error message: ./mbed_lib.json element macros.0: u'__nop()=asm("nop")' does not match u'(^[\\w]+$|^[\\w]+=.+$)'

theotherjimmy, 2018-03-06:
I think the `()` are the failing bits.

LiyouZhou, 2018-03-06:
Do I need to escape them? are they not allowed in json format?

LiyouZhou, 2018-03-06:
but they are inside string, so should be fine. What is this regex trying to validate?

theotherjimmy, 2018-03-06:
> Do I need to escape them?

You're fine. It's a bug.

> are they not allowed in json format?

At the moment, yes. That's bug 2

> What is this regex trying to validate?

The values in the `macros` key.

theotherjimmy, 2018-03-06:
I have a fix. PR in a moment.

theotherjimmy, 2018-03-06:
BAM. Check out #6285

theotherjimmy, 2018-03-06:
```
"macros": ["__nop()=asm(\"nop\")"],
```

All that being said: Why do you need this?

LiyouZhou, 2018-03-06:
i am importing a hg repo from developer.mbed.com. it was designed to work with online compiler and hence ARMCC only. So i need to coax it into working with GCC_ARM.

PR #6285 (2018-03-07): Improve json schema and schema errors
Resolves #6284
closed 2018-03-07 by cmonr