ARMmbed/mbed-os #6305: compiler warnings on several NXP targets

bmcdonnell-ionx, 2018-03-08:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
Many NXP targets.

**mbed-os sha:**
`6ff720302 (tag: mbed-os-5.8.0-rc1, origin/mbed-os-5.8, mbed-os-5.8) Add Mbed version block for new minor release.`

**Description**

For many NXP targets (e.g. LPC4088), the `serial_format()` function in `serial_api.c` doesn't handle invalid values of the `SerialParity` parameter correctly, leading to a compiler warning:

```
../mbed-os/targets/TARGET_NXP/TARGET_LPC408X/TARGET_LPC4088/serial_api.c: In function 'serial_format':
../mbed-os/targets/TARGET_NXP/TARGET_LPC408X/TARGET_LPC4088/serial_api.c:236:43: warning: 'parity_select' may be used uninitialized in this function [-Wmaybe-uninitialized]
                    | parity_select        << 4;
                      ~~~~~~~~~~~~~~~~~~~~~^~~~
../mbed-os/targets/TARGET_NXP/TARGET_LPC408X/TARGET_LPC4088/serial_api.c:235:43: warning: 'parity_enable' may be used uninitialized in this function [-Wmaybe-uninitialized]
                    | parity_enable        << 3
                      ~~~~~~~~~~~~~~~~~~~~~^~~~
```

Also, the handling is not uniform. Various observed correct handling methods include:

- `return` on the `default` `switch` `case`
- set default values in the `default` `case`
- set default values before the `switch` statement

To see all the variants (good and bad), you can do this at a POSIX command line:

```
for f in `find targets/TARGET_NXP -name serial_api.c`; do echo $f; awk '/void +serial_format/,/^\}/' $f; echo; done | less
```

**Suggested changes**

Choose a method and apply it uniformly to all. Also make the `MBED_ASSERT`s consistent (add to or remove from all).

0xc0170, 2018-03-09:
@ARMmbed/team-nxp please review

maclobdell, 2018-08-16:
@mmahadevan108  Any update on this?

mmahadevan108, 2018-09-05:
PR https://github.com/ARMmbed/mbed-os/pull/8010 should fix this issue.

closed 2018-09-24 by maclobdell