ARMmbed/mbed-os #6306: Re-compile issue with ARM Compiler 5 and REALTEK_RTL8195AM platform on example-wifi

jamesbeyond, 2018-03-08:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**Toolchain:**
ARM

**Toolchain version:**
ARM Compiler 5 update 3

**mbed-cli version:** 
1.3.0

**mbed-os sha:**
6ff720302 (mbed-os-5.8.0-rc1)

**DAPLink version:**

**Expected behavior**
Re-compile `mbed-os-example-wifi` example should not throw errors.

**Actual behavior**
Re-compile `mbed-os-example-wifi` example on REALTEK_RTL8195AM thrown errors:
```
Elf2Bin: mbed-os-example-wifi
Error: Q0147E: Failed to create Directory .\BUILD\REALTEK_RTL8195AM\ARM\mbed-os-example-wifi.bin\IMAGE2_TABLE: File exists
Finished: 0 information, 0 warning and 1 error messages.
[ERROR] Error: Q0147E: Failed to create Directory .\BUILD\REALTEK_RTL8195AM\ARM\mbed-os-example-wifi.bin\IMAGE2_TABLE: File exists
Finished: 0 information, 0 warning and 1 error messages.
```

**Steps to reproduce**

1. import example and update to mbed-os 5.8.0-oob
`mbed import mbed-os-example-wifi`
`cd mbed-os-example-wifi` 
`mbed update mbed-os-5.8.0-oob`
2.  Compiler the example first time
`mbed compile -t ARM -m REALTEK_RTL8195AM`
3. modify wifi password in *mbed_app.json*
4. Re-compile the example again
`mbed compile -t ARM -m REALTEK_RTL8195AM`

The problem only happens on Re-compile. if I do a clean first, the compiling would be fine.
Only with  REALTEK_RTL8195AM board, other boards are fine.
Only with ARM Compiler 5. Gcc and IAR are fine.

theotherjimmy, 2018-03-08:
@jamesbeyond Is this bug specific to example wifi? does this affect any other examples?

jamesbeyond, 2018-03-08:
Just had a quick try, other examples have the same problem. @theotherjimmy

theotherjimmy, 2018-03-09:
@0xc0170 This is not related to online deployment. What gave you that impression?

adbridge, 2018-03-12:
@theotherjimmy It doesn't look like a fix for this will make it into RC2 so we may need a known issue adding to the docs for this ?

theotherjimmy, 2018-03-12:
That's true. Known issue coming.

samchuarm, 2018-03-13:
@ARMmbed/team-realtek

yangkunming99, 2018-03-13:
Hi, this is because ARM compiler always create a temp folder with the same name as the output binary. from my observation, the only way is to os.path.exists() and remove() in arm.py:link(). @0xc0170 @samchuarm

tung7970, 2018-03-13:
I can think of two solutions:

1. like @yangkunming99 mentioned, a path removal call such as shutil.rmtree() before elf2bin
2. add --bincombined option to fromelf so it generates plain file instead of directory for multi-load regions

tung7970, 2018-03-13:
Is this ok ?

https://github.com/ARMmbed/mbed-os/pull/6344

theotherjimmy, 2018-03-20:
@thegecko FYI, Realtek will likely download more than one file in the online compiler. I don't know if you were aware, or if you handle it.

thegecko, 2018-03-20:
I'm not aware of a way to serve more than one binary, is this a new requirement?

theotherjimmy, 2018-03-20:
@tung7970 I'm not able to answer @thegecko 's question. Could you?

tung7970, 2018-03-20:
@thegecko Still one download. Binary files are post-processed beforehand.

theotherjimmy, 2018-03-20:
Thanks! @thegecko Sorry for the noise.

adbridge, 2018-03-27:
fixed by #6344

closed 2018-03-27 by adbridge