ARMmbed/mbed-os #6307: Test: attach stub RTC fails for MAX32630FTHR

studavekar, 2018-03-08:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
MAX32630FTHR

**Toolchain:**
GCC_ARM

**mbed-os sha:**

f1d493dd0c5786b3ae3e63c16e3a21f16c0909af

**Steps to reproduce**

Use command run :  mbed  test -m MAX32630FTHR -t GCC_ARM -n tests-mbed_drivers-rtc -v

[MAX32630FTHR_GCC_ARM_tests_mbed_drivers_rtc.log](https://github.com/ARMmbed/mbed-os/files/1794629/MAX32630FTHR_GCC_ARM_tests_mbed_drivers_rtc.log)

CC : @maclobdell @bentcooke

bentcooke, 2018-03-09:
@ARMmbed/team-maximintegrated

jessexm, 2018-10-04:
Unable to reproduce error with steps: 
mbed new bug
revert 1224b3894d1115acd1140c02ba0ec99ac15afafc
mbed test -m MAX32630FTHR -t GCC_ARM -n mbed-os-tests-mbed_drivers-rtc -v

[630fthr-rtc-bug.log](https://github.com/ARMmbed/mbed-os/files/2446852/630fthr-rtc-bug.log)

studavekar, 2018-10-05:
few features are disabled for MAX32630FTHR on master, enabled it and below are results 

```
mbedgt: test suite report:
+----------------------+---------------+------------------------------+--------+--------------------+-------------+
| target               | platform_name | test suite                   | result | elapsed_time (sec) | copy_method |
+----------------------+---------------+------------------------------+--------+--------------------+-------------+
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | OK     | 62.35              | default     |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc           | OK     | 88.77              | default     |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc_reset     | OK     | 55.56              | default     |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc_time      | OK     | 48.66              | default     |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc_time_conv | OK     | 86.08              | default     |
+----------------------+---------------+------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 5 OK
mbedgt: test case report:
+----------------------+---------------+------------------------------+-----------------------------------------------------------------+--------+--------+--------+--------------------+
| target               | platform_name | test suite                   | test case                                                       | passed | failed | result | elapsed_time (sec) |
+----------------------+---------------+------------------------------+-----------------------------------------------------------------+--------+--------+--------+--------------------+
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Functional Test: RTC counts seconds.                            | 1      | 0      | OK     | 10.22              |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Functional Test: set time - CUSTOM_TIME_0.                      | 1      | 0      | OK     | 0.08               |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Functional Test: set time - CUSTOM_TIME_1.                      | 1      | 0      | OK     | 0.09               |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Functional Test: set time - CUSTOM_TIME_2.                      | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: attach original RTC functions.                       | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: attach stub RTC functions.                           | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: set_time() - RTC functions are defined.              | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: set_time() - init RTC function is undefined.         | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: set_time() - write RTC function is undefined.        | 1      | 0      | OK     | 0.33               |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: time() - RTC functions are defined, RTC is disabled. | 1      | 0      | OK     | 0.09               |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: time() - RTC functions are defined, RTC is enabled.  | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: time() - isenabled RTC function is undefined.        | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: time() - read RTC function is undefined.             | 1      | 0      | OK     | 0.06               |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_drivers-rtc       | Unit Test: time() - result is stored in given buffer.           | 1      | 0      | OK     | 0.09               |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc           | RTC - accuracy                                                  | 1      | 0      | OK     | 9.7                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc           | RTC - enabled                                                   | 1      | 0      | OK     | 0.1                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc           | RTC - glitch                                                    | 1      | 0      | OK     | 3.91               |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc           | RTC - init                                                      | 1      | 0      | OK     | 0.0                |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc           | RTC - persist                                                   | 1      | 0      | OK     | 3.48               |
[... cut ...]
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc_time_conv | test make time and local time - RTC leap years full support     | 1      | 0      | OK     | 19.23              |
| MAX32630FTHR-GCC_ARM | MAX32630FTHR  | tests-mbed_hal-rtc_time_conv | test make time and local time - RTC leap years partial support  | 1      | 0      | OK     | 17.14              |
+----------------------+---------------+------------------------------+-----------------------------------------------------------------+--------+--------+--------+--------------------+
```

closed 2018-10-05 by studavekar