ARMmbed/mbed-os #6312: Regenerate arm_pack_manager/index.json failed

ccli8, 2018-03-09:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**OS**
Win7

**Console**
git version 2.10.1.windows.1/Git Bash

**python**
2.7.11

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
79bd263 Merge pull request #6285 from theotherjimmy/fix-6284

**Steps to reproduce**

I would like to regenerate `mbed-os/tools/arm_pack_manager/index.json` to integrate CMSIS pack for new chip. By following the [link](https://os.mbed.com/docs/v5.6/reference/porting-a-new-target-for-exporters.html), I run the command:
```
mbed export -i uvision --update-packs
```
And I get the result and hold there:
```
Exception in thread Thread-20:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 63, in run
    self.func(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 133, in cache_pdsc_and_pack
    self.cache_file(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 110, in cache_file
    stderr.write(e.reason)
TypeError: expected a string or other character buffer object

Exception in thread Thread-13:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 63, in run
    self.func(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 133, in cache_pdsc_and_pack
    self.cache_file(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 110, in cache_file
    stderr.write(e.reason)
TypeError: expected a string or other character buffer object

Exception in thread Thread-7:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 63, in run
    self.func(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 135, in cache_pdsc_and_pack
    self.cache_file(self.pdsc_to_pack(url))
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 108, in cache_file
    fd.write(urlopen(url).read())
  File "c:\python27\lib\socket.py", line 362, in read
    buf.write(data)
MemoryError: out of memory

[ ERROR ] http://cores.actel-ip.com/CMSIS-Pack/Microsemi.M2Sxxx.pdsc does not appear to be a conforming .pdsc file
Exception in thread Thread-12:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 63, in run
    self.func(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 135, in cache_pdsc_and_pack
    self.cache_file(self.pdsc_to_pack(url))
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 108, in cache_file
    fd.write(urlopen(url).read())
  File "c:\python27\lib\socket.py", line 362, in read
    buf.write(data)
MemoryError: out of memory

Exception in thread Thread-17:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 63, in run
    self.func(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 135, in cache_pdsc_and_pack
    self.cache_file(self.pdsc_to_pack(url))
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 108, in cache_file
    fd.write(urlopen(url).read())
  File "c:\python27\lib\socket.py", line 362, in read
    buf.write(data)
MemoryError: out of memory

Exception in thread Thread-10:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 63, in run
    self.func(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 135, in cache_pdsc_and_pack
    self.cache_file(self.pdsc_to_pack(url))
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 108, in cache_file
    fd.write(urlopen(url).read())
  File "c:\python27\lib\socket.py", line 362, in read
    buf.write(data)
MemoryError: out of memory

Exception in thread Thread-11:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 63, in run
    self.func(url)
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 135, in cache_pdsc_and_pack
    self.cache_file(self.pdsc_to_pack(url))
  File "C:\ccli8\iot\mbed\mbed-os\nu-secure\mbed-os\tools\arm_pack_manager\__init__.py", line 108, in cache_file
    fd.write(urlopen(url).read())
  File "c:\python27\lib\socket.py", line 362, in read
    buf.write(data)
MemoryError: out of memory

[ ERROR ] http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/msp432cmsis/latest/exports/TexasInstruments.MSP432E4_DFP.pdsc does not appear to be a conforming .pdsc file
[ ERROR ] http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/msp432cmsis/latest/exports/TexasInstruments.MSP432P4xx_DFP.pdsc does not appear to be a conforming .pdsc file
Caching Files 717/726
```

0xc0170, 2018-03-09:
I use cmder,  f67fe4a7a, mbed-cli 1.3.0 - I could not reproduce these problems.

@theotherjimmy Please review

theotherjimmy, 2018-03-09:
@ccli8 There's nothing I can do about OOM. Sorry.

0xc0170, 2018-05-22:
> @ccli8 There's nothing I can do about OOM. Sorry.

This can be closed as the issue is in the users environment ? (Just to confirm, please close if that is the case)

theotherjimmy, 2018-05-22:
I think so. @ccli8 You may have luck if you close your web browser for a bit before running the script.

closed 2018-05-22 by theotherjimmy

ccli8, 2018-05-23:
@theotherjimmy Closing web browser doesn't help. Actually, this issue also happens on another computer with 8GB DRAM. If you have any update, please let me know.

theotherjimmy, 2018-05-23:
@ccli8 I don't. Sorry.