ARMmbed/mbed-os #6318: test :features-nvstore-tests-nvstore-functionality is failing on EV_COG_AD3029LZ

studavekar, 2018-03-09:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
EV_COG_AD3029LZ

**Toolchain:**
GCC_ARM

**mbed-os sha:**

fd54e5eeb8535fb41874a30c3cd9e068d344e602

**Steps to reproduce**

Use command run :  mbed  test -m EV_COG_AD3029LZ -t GCC_ARM -n features-nvstore-tests-nvstore-functionality -v

```
[1520619942.40][GLRM][INF] remote resources reset...
[1520619945.41][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1520619945.42][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
[1520619945.42][CONN][INF] sending preamble 'e34b2c3c-b20c-432b-81e7-3bab106aace7'
[1520619945.42][GLRM][TXD] {{__sync;e34b2c3c-b20c-432b-81e7-3bab106aace7}}
[1520619950.45][CONN][INF] resending new preamble '639af901-6815-4004-af34-adfd4c89c2bb' after 5.04 sec
[1520619950.45][GLRM][TXD] {{__sync;639af901-6815-4004-af34-adfd4c89c2bb}}
[1520619950.98][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1520619951.78][CONN][RXD] >>> Running 3 test cases...
[1520619951.78][CONN][RXD] 
[1520619951.78][CONN][RXD] >>> Running case #1: 'NVStore: Basic functionality'...
[1520619951.78][CONN][RXD] NVStore areas:
[1520619951.78][CONN][RXD] Area 0: address 0x0003e000, size 4096 (0x1000)
[1520619951.78][CONN][INF] found SYNC in stream: {{__sync;639af901-6815-4004-af34-adfd4c89c2bb}} it is #1 sent, queued...
[1520619951.78][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1520619951.78][CONN][INF] found KV pair in stream: {{__timeout;120}}, queued...
[1520619951.78][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1520619951.78][CONN][INF] found KV pair in stream: {{__testcase_count;3}}, queued...
[1520619951.78][CONN][INF] found KV pair in stream: {{__testcase_name;NVStore: Basic functionality}}, queued...
[1520619951.78][CONN][INF] found KV pair in stream: {{__testcase_name;NVStore: Race test}}, queued...
[1520619951.78][CONN][INF] found KV pair in stream: {{__testcase_name;NVStore: Multiple thread test}}, queued...
[1520619951.78][HTST][INF] sync KV found, uuid=639af901-6815-4004-af34-adfd4c89c2bb, timestamp=1520619951.775586
[1520619951.78][HTST][INF] DUT greentea-client version: 1.3.0
[1520619951.78][HTST][INF] setting timeout to: 120 sec
[1520619951.78][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1520619951.78][HTST][INF] host test setup() call...
[1520619951.78][HTST][INF] CALLBACKs updated
[1520619951.78][HTST][INF] host test detected: default_auto
[1520619951.79][CONN][INF] found KV pair in stream: {{__testcase_start;NVStore: Basic functionality}}, queued...
[1520619952.46][CONN][RXD] Area 1: address 0x0003f000, size 4096 (0x1000)
[1520620072.76][HTST][INF] test suite run finished after 120.98 sec...
[1520620072.77][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1520620072.77][urllib3.connectionpool]Starting new HTTP connection (1): iron-gatekeeper
[1520620072.88][urllib3.connectionpool]http://iron-gatekeeper:8000 "PUT /resource/0602000041114e45001a3003b60f0012a6e1000097969900/disconnect HTTP/1.1" 200 2
[1520620072.88][urllib3.connectionpool]Starting new HTTP connection (1): iron-gatekeeper
[1520620076.16][urllib3.connectionpool]http://iron-gatekeeper:8000 "PUT /resource/0602000041114e45001a3003b60f0012a6e1000097969900/release HTTP/1.1" 200 66
[1520620076.16][HTST][INF] CONN exited with code: 0
[1520620076.16][HTST][INF] No events in queue
[1520620076.16][HTST][INF] stopped consuming events
[1520620076.16][HTST][INF] host test result(): None
[1520620076.16][HTST][WRN] missing __exit event from DUT
[1520620076.16][HTST][WRN] missing __exit_event_queue event from host test
[1520620076.16][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1520620076.16][HTST][INF] calling blocking teardown()
[1520620076.16][HTST][INF] teardown() finished
[1520620076.16][HTST][INF] {{result;timeout}}
```

CC : @maclobdell @bentcooke

bentcooke, 2018-03-09:
@ARMmbed/team-analogdevices

li-ho, 2018-03-15:
`page_size` of `flash_target_config_t` and `size` of `sector_info_t` in mbed-os\hal\TARGET_FLASH_CMSIS_ALGO\flash_data.h are misleading names.
In flash_data.h, `page_size` is meant to be flash's minimal programmable unit size.
In some documents, `page_size` is taken as sector size.
More texts for `flash_target_config_t` and `sector_info_t` would be appreciated

0xc0170, 2018-03-15:
@li-ho thanks for the feedback, can you send a pull request adding more info that would clarify it?

Here would be my proposal:

```
/** Target flash algorithm structure
 */
typedef struct {
    const uint32_t init;
    const uint32_t uninit; 
    const uint32_t erase_sector;
    const uint32_t program_page;
    const uint32_t static_base;
    uint32_t *algo_blob;
} flash_algo_t;

/** Sector information structure
 */
typedef struct {
    const uint32_t start; /**< Sector start address */
    const uint32_t size; /**< Sector size */
} sector_info_t;

/** Flash configuration structure
 */
typedef struct {
    const uint32_t page_size; /**< The minimum program page size */
    const uint32_t flash_start; /**< Program page size */
    const uint32_t flash_size; /**<  Flash size */
    const sector_info_t *sectors; /**<  List of sectors */
    const uint32_t sector_info_count; /**< Number of sectors */
} flash_target_config_t;

/** Target flash configuration
 */
struct flash_s {
    const flash_target_config_t *target_config;
    const flash_algo_t *flash_algo;
};
```

0xc0170, 2018-03-15:
@li-ho I referenced above the PR with docs fixes

li-ho, 2018-03-16:
`flash_start` is an address, not a size.

davidsaada, 2018-07-11:
@studavekar Is this issue still relevant after @li-ho's PR?

li-ho, 2018-07-12:
There have been changes in mbed CLI. According to mbed CLI version 1.7.3,
`mbed test -m EV_COG_AD3029LZ -t GCC_ARM -n features-nvstore-tests-nvstore-functionality` can build but fail to run.
`mbed test -m EV_COG_AD3029LZ -t GCC_ARM -n mbed-os-features-nvstore-tests-nvstore-functionality` can produce pass results.

davidsaada, 2018-07-12:
@li-ho So just to clarify - Changes in tools solve the problem, or is it a new problem? Didn't understand which version of the tools has the problem and which doesn't.

li-ho, 2018-07-12:
@davidsaada After the #6367 bug fix, it's now a new problem related to tools. It's just a matter where to run `mbed test -m EV_COG_AD3029LZ -t GCC_ARM`. I would not consider the new problem of tools to be a major  bug to fix.

0xc0170, 2018-07-13:
@li-ho Shall this be tracked separately? Or here? How to reproduce the failures?

li-ho, 2018-07-13:
@0xc0170 The test case name must be `mbed-os-features-nvstore-tests-nvstore-functionality` instead of `features-nvstore-tests-nvstore-functionality` for current version of mbed CLI. It's not about nvstore failure anymore.

davidsaada, 2018-07-13:
> @0xc0170 The test case name must be mbed-os-features-nvstore-tests-nvstore-functionality instead of features-nvstore-tests-nvstore-functionality for current version of mbed CLI. It's not about nvstore failure anymore.

This actually depends on which directory you run the test from. If you run it from within mbed-os, the test name shouldn't include the "mbed-os" prefix. If you run it from one level upper, this prefix should be added.

0xc0170, 2018-07-13:
@li-ho Thanks,  I haven't noticed that naming change.

>  @studavekar Is this issue still relevant after @li-ho's PR?

@studavekar Can we retest?

li-ho, 2018-07-13:
> This actually depends on which directory you run the test from. If you run it from within mbed-os, the 
> test name shouldn't include the "mbed-os" prefix. If you run it from one level upper, this prefix should be added.

Within or above mbed-os directory, "mbed-os" prefix must be included in test case name for current mbed CLI.

studavekar, 2018-08-21:
closing this issue, Thanks for the fix.

```
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'OK'
mbedgt: test on hardware with target id: DUMMY
mbedgt: test suite 'features-nvstore-tests-nvstore-functionality' .................................... OK in 47.19 sec
        test case: 'NVStore: Basic functionality' .................................................... OK in 0.39 sec
        test case: 'NVStore: Multiple thread test' ................................................... OK in 7.73 sec
        test case: 'NVStore: Race test' .............................................................. OK in 0.30 sec
mbedgt: test case summary: 3 passes, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.2617263269
mbedgt: test suite report:
+-------------------------+-----------------+----------------------------------------------+--------+--------------------+-------------+
| target                  | platform_name   | test suite                                   | result | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+----------------------------------------------+--------+--------------------+-------------+
| EV_COG_AD3029LZ-GCC_ARM | EV_COG_AD3029LZ | features-nvstore-tests-nvstore-functionality | OK     | 47.19              | default     |
+-------------------------+-----------------+----------------------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+-------------------------+-----------------+----------------------------------------------+-------------------------------+--------+--------+--------+--------------------+
| target                  | platform_name   | test suite                                   | test case                     | passed | failed | result | elapsed_time (sec) |
+-------------------------+-----------------+----------------------------------------------+-------------------------------+--------+--------+--------+--------------------+
| EV_COG_AD3029LZ-GCC_ARM | EV_COG_AD3029LZ | features-nvstore-tests-nvstore-functionality | NVStore: Basic functionality  | 1      | 0      | OK     | 0.39               |
| EV_COG_AD3029LZ-GCC_ARM | EV_COG_AD3029LZ | features-nvstore-tests-nvstore-functionality | NVStore: Multiple thread test | 1      | 0      | OK     | 7.73               |
| EV_COG_AD3029LZ-GCC_ARM | EV_COG_AD3029LZ | features-nvstore-tests-nvstore-functionality | NVStore: Race test            | 1      | 0      | OK     | 0.3                |
+-------------------------+-----------------+----------------------------------------------+-------------------------------+--------+--------+--------+--------------------+
```

closed 2018-08-21 by studavekar