ARMmbed/mbed-os #6319: test :Test insert_absolute zero is failing on EV_COG_AD3029LZ

studavekar, 2018-03-09:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
EV_COG_AD3029LZ

**Toolchain:**
GCC_ARM

**mbed-os sha:**

fd54e5eeb8535fb41874a30c3cd9e068d344e602

**Steps to reproduce**

Use command run :  mbed  test -m EV_COG_AD3029LZ -t GCC_ARM -n tests-mbed_drivers-timerevent -v

```
[1520618128.10][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1520618128.10][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
[1520618128.10][CONN][INF] sending preamble '22711742-fb5e-451c-9b02-512b07d39c5b'
[1520618128.10][GLRM][TXD] {{__sync;22711742-fb5e-451c-9b02-512b07d39c5b}}
[1520618133.10][CONN][INF] resending new preamble '43e65411-25b6-4481-a5b2-6da04642af7b' after 5.00 sec
[1520618133.11][GLRM][TXD] {{__sync;43e65411-25b6-4481-a5b2-6da04642af7b}}
[1520618133.89][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1520618133.89][CONN][RXD] >>> Running 7 test cases...
[1520618133.89][CONN][INF] found SYNC in stream: {{__sync;43e65411-25b6-4481-a5b2-6da04642af7b}} it is #1 sent, queued...
[1520618133.89][HTST][INF] sync KV found, uuid=43e65411-25b6-4481-a5b2-6da04642af7b, timestamp=1520618133.890527
[1520618133.89][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1520618133.89][HTST][INF] DUT greentea-client version: 1.3.0
[1520618133.89][CONN][INF] found KV pair in stream: {{__timeout;5}}, queued...
[1520618133.89][HTST][INF] setting timeout to: 5 sec
[1520618133.89][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1520618133.89][CONN][INF] found KV pair in stream: {{__testcase_name;Test insert}}, queued...
[1520618133.89][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1520618133.89][HTST][INF] host test setup() call...
[1520618133.89][HTST][INF] CALLBACKs updated
[1520618133.89][HTST][INF] host test detected: default_auto
[1520618133.97][CONN][INF] found KV pair in stream: {{__testcase_name;Test insert_absolute}}, queued...
[1520618133.97][CONN][INF] found KV pair in stream: {{__testcase_name;Test remove after insert}}, queued...
[1520618134.07][CONN][INF] found KV pair in stream: {{__testcase_name;Test remove after insert_absolute}}, queued...
[1520618134.07][CONN][INF] found KV pair in stream: {{__testcase_name;Test insert_absolute zero}}, queued...
[1520618134.16][CONN][INF] found KV pair in stream: {{__testcase_name;Test insert timestamp from the past}}, queued...
[1520618134.27][CONN][RXD] 
[1520618134.27][CONN][RXD] >>> Running case #1: 'Test insert'...
[1520618134.27][CONN][INF] found KV pair in stream: {{__testcase_name;Test insert_absolute timestamp from the past}}, queued...
[1520618134.27][CONN][INF] found KV pair in stream: {{__testcase_start;Test insert}}, queued...
[1520618134.37][CONN][INF] found KV pair in stream: {{__testcase_finish;Test insert;1;0}}, queued...
[1520618134.49][CONN][RXD] >>> 'Test insert': 1 passed, 0 failed
[1520618134.49][CONN][RXD] 
[1520618134.49][CONN][RXD] >>> Running case #2: 'Test insert_absolute'...
[1520618134.58][CONN][INF] found KV pair in stream: {{__testcase_start;Test insert_absolute}}, queued...
[1520618134.67][CONN][RXD] >>> 'Test insert_absolute': 1 passed, 0 failed
[1520618134.67][CONN][RXD] 
[1520618134.67][CONN][INF] found KV pair in stream: {{__testcase_finish;Test insert_absolute;1;0}}, queued...
[1520618134.77][CONN][RXD] >>> Running case #3: 'Test remove after insert'...
[1520618134.87][CONN][INF] found KV pair in stream: {{__testcase_start;Test remove after insert}}, queued...
[1520618135.07][CONN][RXD] >>> 'Test remove after insert': 1 passed, 0 failed
[1520618135.07][CONN][RXD] 
[1520618135.08][CONN][INF] found KV pair in stream: {{__testcase_finish;Test remove after insert;1;0}}, queued...
[1520618135.18][CONN][RXD] >>> Running case #4: 'Test remove after insert_absolute'...
[1520618135.28][CONN][INF] found KV pair in stream: {{__testcase_start;Test remove after insert_absolute}}, queued...
[1520618135.96][CONN][RXD] >>> 'Test remove after insert_absolute': 1 passed, 0 failed
[1520618135.96][CONN][RXD] 
[1520618135.96][CONN][RXD] >>> Running case #5: 'Test insert_absolute zero'...
[1520618135.96][CONN][RXD] :164::FAIL: Expected 1 Was 0
[1520618135.96][CONN][RXD] >>> 'Test insert_absolute zero': 0 passed, 1 failed with reason 'Assertion Failed'
[1520618135.96][CONN][RXD] 
[1520618135.96][CONN][RXD] >>> Test cases: 4 passed, 1 failed with reason 'Assertion Failed'
[1520618135.96][CONN][INF] found KV pair in stream: {{__testcase_finish;Test remove after insert_absolute;1;0}}, queued...
[1520618135.96][CONN][INF] found KV pair in stream: {{__testcase_start;Test insert_absolute zero}}, queued...
[1520618135.96][CONN][INF] found KV pair in stream: {{__testcase_finish;Test insert_absolute zero;0;1}}, queued...
[1520618136.00][CONN][RXD] >>> TESTS FAILED!
[1520618136.00][CONN][INF] found KV pair in stream: {{__testcase_summary;4;1}}, queued...
[1520618136.00][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1520618136.00][CONN][INF] found KV pair in stream: {{reserved_heap;11436}}, queued...
[1520618136.00][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1520618135.997675
[1520618136.00][HTST][ERR] orphan event in main phase: {{reserved_heap;11436}}, timestamp=1520618135.997681
[1520618136.08][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200416a8",600,4096}}, timestamp=1520618136.082991
[1520618136.08][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200416a8",600,4096}}, queued...
[1520618136.08][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200004b4",72,512}}, timestamp=1520618136.082998
[1520618136.08][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200004b4",72,512}}, queued...
[1520618136.20][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200004fc",112,768}}, timestamp=1520618136.197725
[1520618136.20][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200004fc",112,768}}, queued...
[1520618136.20][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1520618136.20][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1520618136.21][HTST][INF] __exit(0)
[1520618136.21][HTST][INF] __notify_complete(False)
[1520618136.21][HTST][INF] __exit_event_queue received
```

CC : @maclobdell @bentcooke

bentcooke, 2018-03-09:
@ARMmbed/team-analogdevices

studavekar, 2018-03-09:
@fkjagodzinski any idea here this tests seems to be failing for a couple of hardware 

- EV_COG_AD3029LZ 
- EV_COG_AD4050LZ
- MAX32630FTHR

fkjagodzinski, 2018-03-12:
Looks like the `us_ticker_fire_interrupt()` function needs an update for these targets. The test case that failed `'Test insert_absolute zero'` expects a ticker interrupt to fire instantly when an event with  us delay is attached.

https://github.com/ARMmbed/mbed-os/blob/f67fe4a7a31666469d5487bc81b2cfc2dec16edc/TESTS/mbed_drivers/timerevent/main.cpp#L154-L158

A couple of boards have already been fixed by @0xc0170 in 2661f6fe401ea2d0bd866bdb861d71e13bfa4d47.

sbrewer1, 2018-03-14:
I am a bit confused by this test.  Can you please clarify if this is a relative or absolute timestamp. i.e. when calling  us_ticker.c::us_ticker_set_interrupt(timestamp_t timestamp) is timestamp==0u, or timestamp==get_current_time()+0u?

fkjagodzinski, 2018-03-14:
`us_ticker_set_interrupt()` is not called in this case, the `us_ticker_fire_interrupt()` is called instead: https://github.com/ARMmbed/mbed-os/blob/f67fe4a7a31666469d5487bc81b2cfc2dec16edc/hal/mbed_ticker_api.c#L230-L235

`TimerEvent::insert_absolute()` --> `ticker_insert_event_us()` --> `schedule_interrupt()` --> `us_ticker_fire_interrupt()`

sbrewer1, 2018-03-15:
Ok thanks for the clarification.  The mystery is why the interrupt is not called quick enough to pass the test.  As far as I can see it should happen immediately.  Maybe IRQs are disabled?  Maybe 'largecnt' is non-zero?  We will need to investigate from our side.

li-ho, 2018-03-16:
The following code will pass the test:
```diff
void us_ticker_fire_interrupt(void)
{
-    NVIC_SetPendingIRQ(TMR2_EVT_IRQn);
+    adi_tmr_Enable(ADI_TMR_DEVICE_GP2, false);
+    largecnt = 0;
+    us_ticker_irq_handler();
}
```

dave-wu, 2018-03-16:
I thought the NVIC_SetPendingIRQ would trigger the interrupt and we are not supposed to call the irq handler directly here.

0xc0170, 2018-03-20:
That is correct, should not call the handler directly, dangerous to do.

@li-ho Why set pending IRQ is not effective? Is there driver API that would retrigger TMR2 IRQ handler?

li-ho, 2018-03-20:
It's clear `NVIC_SetPendingIRQ(TMR2_EVT_IRQn);` does not work as we expected.
`GP2CallbackFunction` is the callback of TMR2 IRQ handler.

`adi_tmr_Enable(ADI_TMR_DEVICE_GP2, false);` disable timer.

`tmr2Config.nLoad        = cnt;`
`tmr2Config.nAsyncLoad   = cnt;`
`adi_tmr_ConfigTimer(ADI_TMR_DEVICE_GP2, tmr2Config);`
`adi_tmr_Enable(ADI_TMR_DEVICE_GP2, true);` restart the timer.

0xc0170, 2018-03-20:
Does that fixes the issue? Can you send PR?

li-ho, 2018-03-21:
Within `us_ticker_fire_interrupt()`, TMR2 timer cannot be restarted after being disabled, and `NVIC_SetPendingIRQ(TMR2_EVT_IRQn)` is not effective either.
The workable solution so far is to disable TMR2 timer and then call `us_ticker_irq_handler();` 

The proposed change to https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Analog_Devices/TARGET_ADUCM4X50/TARGET_ADUCM4050/api/us_ticker.c is as below:

```diff
void us_ticker_fire_interrupt(void)
{
-   NVIC_SetPendingIRQ(TMR2_EVT_IRQn);
+   largecnt = 0;
+   event_timer();
}
```
@sbrewer1 @dave-wu @0xc0170  Is it OK to submit a PR?

0xc0170, 2018-03-21:
`largecnt  = 0` would mean you invoke handler directly? If yes, then this can lead to stack overflow (we already had issues like this that was one of the drive for having fire interrupt function).

sbrewer1, 2018-03-21:
Unfortunately I think this points to another problem in the code.  `event_timer()` is called from `us_ticker_set_interrupt(timestamp_t timestamp)`.  It is possible that `us_ticker_irq_handler()` is called directly from `event_timer()` if `largecnt==0`.  Which would mean the irq handler could cause the stack overflow problem you mention.  This code should fix the problem (untested).

```
static void event_timer()
{
    if (largecnt) {
        uint32_t cnt = largecnt;

        if (cnt > 65535u) {
            cnt = 0u;
        } else
            cnt = 65536u - cnt;

        tmr2Config.nLoad        = cnt;
        tmr2Config.nAsyncLoad   = cnt;
        adi_tmr_ConfigTimer(ADI_TMR_DEVICE_GP2, tmr2Config);
        adi_tmr_Enable(ADI_TMR_DEVICE_GP2, true);
      }
//    } else {
//       us_ticker_irq_handler();
//    }
}

static void GP2CallbackFunction(void *pCBParam, uint32_t Event, void  * pArg)
{
    if (largecnt >= 65536u) {
        largecnt -= 65536u;
    } else
        largecnt = 0;

    if (largecnt < 65536u) {
        adi_tmr_Enable(ADI_TMR_DEVICE_GP2, false);
        if (largecnt) {
            event_timer();
        } else {
                     us_ticker_irq_handler();
        }
    }
}
```

sbrewer1, 2018-03-21:
As Edmund correctly points out, the reason the test is failing is that when the timer is disabled via `adi_tmr_enable(xxxx, false)`, then interrupts are disabled for that timer.

Because of my previous comment and bugfix, we can use a variation of Edmund's solution.  The (untested) solution is to call event_timer() with a largecnt value, then initiate an interrupt manually after forcing largecnt back to zero.
```
void us_ticker_fire_interrupt(void)
{
    largecnt = 65534u;    // set a long interval so interrupt does not fire while we are hacking largecnt below
    event_timer();      // enable the timer and interrupt
    largecnt = 0u;     // clear largecnt so Callback immediately runs handler 
    NVIC_SetPendingIRQ(TMR2_EVT_IRQn);   // force the interrupt (GP2CallbackFunction) to fire
}
```

sbrewer1, 2018-03-22:
Hmm, I spotted a bug with my above changes.  When calling us_ticker_set_interrupt(), if largecnt is zero, then my bugfix code doesn't initiate any interrupt.

```
static void event_timer()
{
    if (largecnt) {
        uint32_t cnt = largecnt;

        if (cnt > 65535u) {
            cnt = 0u;
        } else
            cnt = 65536u - cnt;

        tmr2Config.nLoad        = cnt;
        tmr2Config.nAsyncLoad   = cnt;
        adi_tmr_ConfigTimer(ADI_TMR_DEVICE_GP2, tmr2Config);
        adi_tmr_Enable(ADI_TMR_DEVICE_GP2, true);
    } else {
        tmr2Config.nLoad        = 65535u;
        tmr2Config.nAsyncLoad   = 65535u;
        adi_tmr_ConfigTimer(ADI_TMR_DEVICE_GP2, tmr2Config);
        adi_tmr_Enable(ADI_TMR_DEVICE_GP2, true);
    }
}

static void GP2CallbackFunction(void *pCBParam, uint32_t Event, void  * pArg)
{
    if (largecnt >= 65536u) {
        largecnt -= 65536u;
    } else
        largecnt = 0;

    if (largecnt < 65536u) {
        adi_tmr_Enable(ADI_TMR_DEVICE_GP2, false);
        if (largecnt) {
            event_timer();
        } else {
                     us_ticker_irq_handler();
        }
    }
}

void us_ticker_fire_interrupt(void)
{
    largecnt = 65534u;    // set a long interval so interrupt does not fire while we are hacking largecnt below
    event_timer();      // enable the timer and interrupt
    largecnt = 0u;     // clear largecnt so Callback immediately runs handler 
    NVIC_SetPendingIRQ(TMR2_EVT_IRQn);   // force the interrupt (GP2CallbackFunction) to fire
}
```

li-ho, 2018-03-23:
@0xc0170 @sbrewer1 TMR2 is pre-scaled to tick at the frequency of 26MHz/256. Isn't 10 us too long to wait to fire the interrupt? Were you thinking to fire the the interrupt after one tick at the frequency of 26MHz (0.04 us)?

0xc0170, 2018-03-23:
> I have tried to restart timer to tick for 0.04 us instead of 10 us within us_ticker_fire_interrupt(), and it works (I mean all pass). The only thing is that NVIC_SetPendingIRQ(TMR2_EVT_IRQn) still does not make sense to me.

SetPending will be replaced then with your fix (to set tick for 0.04) ? 

> @0xc0170 @sbrewer1 TMR2 is pre-scaled to tick at the frequency of 26MHz/256. Isn't 10 us too long to wait to fire the interrupt? Were you thinking to fire the the interrupt after one tick at the frequency of 26MHz (0.04 us)?

It is too long, should be immediate. The event is in the past already in this case.

cc @fkjagodzinski @maciejbocianski

sbrewer1, 2018-03-23:
@li-ho

This code does not want the interrupt to run, it just enables the timer so the interrupt will occur.
 
```    largecnt = 65534u;    // set a long interval so interrupt does not fire while we are hacking largecnt below
    event_timer();      // enable the timer and interrupt
```

This code enables forces the interrupt

```NVIC_SetPendingIRQ(TMR2_EVT_IRQn)```

0xc0170, 2018-04-11:
Resolved via 6490

closed 2018-04-11 by 0xc0170