ARMmbed/mbed-os #6320: test : tests-mbed_platform-critical_section is failing on EV_COG_AD3029LZ

studavekar, 2018-03-09:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
EV_COG_AD3029LZ

**Toolchain:**
GCC_ARM

**mbed-os sha:**

fd54e5eeb8535fb41874a30c3cd9e068d344e602

**Steps to reproduce**

Use command run :  mbed  test -m EV_COG_AD3029LZ -t GCC_ARM -n tests-mbed_platform-critical_section -v

```
[1520621251.86][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
[1520621251.86][CONN][INF] sending preamble '70012f47-b5cf-471d-828e-f089a62f5b07'
[1520621251.86][GLRM][TXD] {{__sync;70012f47-b5cf-471d-828e-f089a62f5b07}}
[1520621253.05][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1520621253.05][CONN][INF] found SYNC in stream: {{__sync;70012f47-b5cf-471d-828e-f089a62f5b07}} it is #0 sent, queued...
[1520621253.05][HTST][INF] sync KV found, uuid=70012f47-b5cf-471d-828e-f089a62f5b07, timestamp=1520621253.054578
[1520621253.16][HTST][INF] DUT greentea-client version: 1.3.0
[1520621253.16][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1520621253.16][HTST][INF] setting timeout to: 10 sec
[1520621253.16][CONN][INF] found KV pair in stream: {{__timeout;10}}, queued...
[1520621253.16][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1520621253.16][HTST][INF] host test setup() call...
[1520621253.16][HTST][INF] CALLBACKs updated
[1520621253.16][HTST][INF] host test detected: default_auto
[1520621253.16][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1520621253.27][CONN][RXD] >>> Running 6 test cases...
[1520621253.27][CONN][INF] found KV pair in stream: {{__testcase_name;Test critical section C API}}, queued...
[1520621253.36][CONN][INF] found KV pair in stream: {{__testcase_name;Test critical section C API nested lock}}, queued...
[1520621253.46][CONN][INF] found KV pair in stream: {{__testcase_name;Test critical section C++ API constructor/destructor}}, queued...
[1520621253.56][CONN][INF] found KV pair in stream: {{__testcase_name;Test critical section C++ API constructor/destructor nested lock}}, queued...
[1520621253.66][CONN][RXD] 
[1520621253.66][CONN][INF] found KV pair in stream: {{__testcase_name;Test critical section C++ API enable/disable}}, queued...
[1520621253.66][CONN][INF] found KV pair in stream: {{__testcase_name;Test critical section C++ API enable/disable nested lock}}, queued...
[1520621253.76][CONN][RXD] >>> Running case #1: 'Test critical section C API'...
[1520621254.05][CONN][RXD] :88::FAIL: Expected TRUE Was FALSE
[1520621254.05][CONN][INF] found KV pair in stream: {{__testcase_start;Test critical section C API}}, queued...
[1520621254.05][CONN][INF] found KV pair in stream: {{__testcase_finish;Test critical section C API;0;1}}, queued...
[1520621254.07][CONN][RXD] >>> 'Test critical section C API': 0 passed, 1 failed with reason 'Assertion Failed'
[1520621254.07][CONN][RXD] 
[1520621254.07][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1520621254.16][CONN][RXD] >>> TESTS FAILED!
[1520621254.16][CONN][INF] found KV pair in stream: {{__testcase_summary;0;1}}, queued...
[1520621254.16][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1520621254.159823
[1520621254.16][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1520621254.26][HTST][ERR] orphan event in main phase: {{reserved_heap;11588}}, timestamp=1520621254.257781
[1520621254.26][CONN][INF] found KV pair in stream: {{reserved_heap;11588}}, queued...
[1520621254.26][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200045a0",1048,4096}}, timestamp=1520621254.257786
[1520621254.26][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200045a0",1048,4096}}, queued...
[1520621254.26][HTST][ERR] orphan event in main phase: {{__thread_info;"0x02004042c",64,512}}, timestamp=1520621254.257789
[1520621254.26][CONN][INF] found KV pair in stream: {{__thread_info;"0x02004042c",64,512}}, queued...
[1520621254.36][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200403e4",104,768}}, timestamp=1520621254.357784
[1520621254.36][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200403e4",104,768}}, queued...
[1520621254.36][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1520621254.36][HTST][INF] __notify_complete(False)
[1520621254.36][HTST][INF] __exit_event_queue received
[1520621254.36][HTST][INF] test suite run finished after 1.20 sec...
[1520621254.36][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
```

CC : @maclobdell @bentcooke

bentcooke, 2018-03-09:
@ARMmbed/team-analogdevices

li-ho, 2018-03-20:
In line 48 of https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbed_platform/critical_section/main.cpp
The 'timeout_time_us' of 'timeout.attach_us(callback(tiemout_callback), timeout_time_us);' is 1.
The typical value of 'timeout_time_us' from other test cases ranges from 100 to 500000.
1 microsecond is too harsh for a 26 MHz MCU.

0xc0170, 2018-03-20:
> In line 48 of https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbed_platform/critical_section/main.cpp
The timeout_time_us of timeout.attach_us(callback(tiemout_callback), timeout_time_us); is set to 1.
The typical value of timeout_time_us from other test cases ranges from 100 to 500000.
1 microsecond is too harsh for a 26 MHz MCU.

@maciejbocianski

li-ho, 2018-03-20:
The tested achievable values for EV_COG_ADxxxx boards are 
`const us_timestamp_t timeout_time_us = 50;`
`const int wait_time_us = timeout_time_us * 1;`

maciejbocianski, 2018-03-20:
I think it's the same problem as reported here #6319 
frankly calling `timeout.attach_us(..., 0);` and `timeout.attach_us(..., 1);` is practically the same scenario

studavekar, 2018-03-20:
@maciejbocianski thanks should be close this issue and track progress on #6319?

maciejbocianski, 2018-03-21:
Yes, could be closed
@studavekar

closed 2018-03-21 by 0xc0170