ARMmbed/mbed-os #6324: REALTEK_RTL8195AM exporting is disabled on all the IDEs/Projects

jamesbeyond, 2018-03-09:
### Description
- Type:  Enhancement / Question
- Priority:  Minor

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
 `mbed export -S` shows REALTEK_RTL8195AM was disabled for all the type of export.

**Suggested enhancement**
The OOB is requiring those features.
Should those exporting features be supported?

samchuarm, 2018-03-13:
The target does not have a CMSIS PACK associated with it.

0xc0170, 2018-05-22:
> The target does not have a CMSIS PACK associated with it.

@ARMmbed/team-realtek Just to confirm. If that is correct this can be closed as won't fix for now? Or any plans?

samchuarm, 2018-09-26:
@M-ichae-l , can you please confirm this?

samchuarm, 2018-10-03:
@M-ichae-l , please confirm if there won't be CMSIS-Pack support for this target? Or any plans? Thanks.

M-ichae-l, 2018-10-05:
@samchuarm Sorry for the late reply. It is confirmed that no plan on CMSIS-Pack support for RTL8195AM currently.

samchuarm, 2018-10-07:
Hi @0xc0170 , since Realtek confirmed they're not going to add CMSIS-Pack support for this target, should this be closed as won't fix? Thx

AdamZhang0124, 2019-01-09:
Hi @0xc0170 , just checking if this issue can be closed. We are trying to minimize the remaining open issues.
Thanks.

0xc0170, 2019-01-09:
Closing this as won't fix.

closed 2019-01-09 by 0xc0170

AdamZhang0124, 2019-01-10:
I understand, thanks!