ARMmbed/mbed-os #6325: LittleFs poor performance

YarivCol, 2018-03-11:
Hello!

Ive started using LittleFs after reading about its powerloss resilient features.

However, ive encountered  realy poor performance compared to FAT.

The same transaction took the FAT fs 2 read cycles and 2 write cycles while with the LFS it took 254 read cycles and 45 write cycles. 

Is that a standard behavior?

Im using an eeprom with ths size of 0X1FFFF bytes, and i formated the LFS with 512 bytes for read,progg and erase size and with 32-64 lookhead.

Thx for helpers :)

geky, 2018-03-12:
Hi @YarivCol, that's interesting.

LittleFS can be slower than FAT because of the power resilience and because it prioritizes memory consumption. But we found in the worst case this was ~2:3 not ~2:254.

It's hard to guess why you're seeing such terrible performance without knowing how you're using LittleFS. You don't happen to have a code snippet that shows this you could post do you?

I do have a few questions:
- When you say eeprom, do you mean a part like the AT24C?
- You said 512 bytes for read/prog/erase in format, do you also pass these sizes to the constructor (for mounting)?
- How many files do you have on your filesystem? Are they large?
- How much do you write each boot?
- Do you seek in files often? Do you write to the middle of files? Do you rewrite files?

The large amount of read cycles could make sense and would probably be improved with a larger lookahead (keep in mind the lookahead size is in bits, a lookahead of 512 only takes 64 bytes).

But I'm confused by the massive difference in write cycles. LittleFS and FAT should be very similar in this area. I wonder if LittleFS is doing a bunch of small writes while FAT is only doing one big write.

YarivCol, 2018-03-14:
Hi, thanks for the reply!

About your question:
1. Yes
2. Yes, i do pass the same arguments to Format and to the c'tor
3. I only have 3-4 files, the largest one is 12k in size the others are 3k and below that... need to say that ive encountered the problem with the 3k file.
4. I only create the files if the FS is formatted, otherwise i write nothing when booting...
5. Yeah i do write mainly to middle of files as the files are mostly configuration files

dannybenor, 2018-05-23:
@YarivCol  what block device are you using?

michalpasztamobica, 2019-08-08:
Hi, @YarivCol. Is the issue still relevant?

closed 2019-10-31 by ciarmcom