ARMmbed/mbed-os #6331: Power-saving behaviour of Nanostack

stevew817, 2018-03-12:
### Description
- Type: Question

@SeppoTakalo 

-----------------------------------------------------------------
## Question
I've been test-driving Nanostack on EFR32 for a bit now, and wanted to ask about its power-saving behaviour. When configured as a sleepy end-node, I can see it turns off the radio during the MAC downtime. So far, so good.
However, during the sleepy time, the processor is still being woken up every 10ms. I traced this back to the nanostack event loop in the PAL. Is there any good reason why Nanostack needs a 10ms system ticker, when the underlying RTOS should be able to support all of Nanostack's timing requirements in a tickless way? Is there any work being done to resolve this, and integrate Nanostack into mbed instead of running its own tick-based scheduler on top of the RTOS' scheduler?

0xc0170, 2018-03-12:
cc @kjbracey-arm

kjbracey-arm, 2018-03-12:
It's not an area of active work.

Nanostack's scheduler is designed to support sleep in a bare metal environment - see net_sleep.h and eventOS_scheduler_sleep(). I can't find the "reference implementation of idle" referred to there publicly. But the basic idea is for the platform's `eventOS_scheduler_idle` call to check from Nanostack and scheduler whether it can sleep and how long, and if it can, to sleep the device.

Part of the issue is that just being "tickless" isn't sufficient - there is stuff that the stack would regularly do as housekeeping because the device is on. Most of the ticky stuff is that. Even if implemented tickless, that would still lead the RTOS to the device to do unneeded stuff. The answer to the question "when do I next need to wake up?" is different to the answer to "when do you next have something to do, if I am awake?". 

For the Nanostack's intended sleep mechanism the distinction is that its manually-specified (ie non-tick) timers are treated as wake events (reported via `eventOS_event_timer_shortest_active_timer`). The ticks are not wake events. Anything done with the ticks that needs waking is reported via `arm_net_check_enter_deep_sleep_possibility`.

If the rest of the system is tickless enough to sleep when idle, modifying the mbed OS `eventOS_scheduler_idle` to shut off ticks and sleep the event dispatch thread for that long would probably do the job.

stevew817, 2018-03-12:
Okay, thanks for the rapid reply. If I were to make such changes, would you guys be open to a PR?

kjbracey-arm, 2018-03-12:
Do feel free to experiment.

I would just give one warning - it's been years since the Nanostack sleep functionality has been fully exercised - we support sleepy-mode devices in terms of network protocols and testing, as it's a core part of the stack functionality - but that testing (which is mostly performed in simulation) is limited to shutting down the radio.

We have no current real-world platforms using the system power control on top of that, and that's why we've never spent the time porting those mechanisms to mbed OS. So I have a suspicion that current versions of Nanostack might end up either sleeping too much (because `arm_net_check_enter_deep_sleep_possibility` isn't checking enough reasons to wake) or not enough (because people are putting regular work into wake-uppable event timers rather than ticks).

kjbracey-arm, 2018-03-12:
If it helps, here's the "reference" bare-metal version of idle, to show intended behaviour - this was included in previous standalone versions of documentation:

```
void eventOS_scheduler_idle(void)
{
    uint32_t sleep_possible = arm_net_check_enter_deep_sleep_possibility();
    if (sleep_possible) {
        uint32_t system_timer_next_tick_time = eventOS_event_timer_shortest_active_timer();
        if (system_timer_next_tick_time) {
            if (system_timer_next_tick_time < sleep_possible) {
                sleep_possible = system_timer_next_tick_time;    //Select shorter next event
            }
        }
        if (arm_net_enter_sleep() == 0) {
            if (eventOS_scheduler_timer_stop() != 0) {
                tr_debug("EventOS timer sleep fail");
            }

            sleep_possible = eventOS_scheduler_sleep(sleep_possible);
            //Enable Data Polling after sleep & Synch Times
            arm_net_wakeup_and_timer_synch(sleep_possible);

            //Update Runtime ticks and event timers
            if (eventOS_scheduler_timer_synch_after_sleep(sleep_possible) != 0) {
                tr_debug("Timer wakeUP Fail");
            }
        }
    } else {
        eventOS_scheduler_wait();
    }
}
```
That then relies on two platform calls `eventOS_scheduler_wait` and `eventOS_scheduler_sleep`, as described in `eventOS_scheduler.h`.

closed 2018-08-10 by kjbracey-arm