ARMmbed/mbed-os #6334: Pretty bars feature optimisation

jeromecoutant, 2018-03-12:
### Description
- Type: Bug
- Related issue: #5929 
- Priority: Minor

## Bug

Pretty bars feature doesn't work perfectly with dual RAM targets.

I can see 2 sub cases:

- check #5171
IRAM1 and IRAM2 have been concatenated into "IRAM1"

- check #6172 
the stack is now located in IRAM2

geky, 2018-03-12:
Hi @jeromecoutant, thanks for opening an issue.

It should've been noted that pretty bars has since been removed:
https://github.com/ARMmbed/mbed-os/pull/6197

We may add it back at a later date, but it's been established that pretty bars won't replace the pretty table output. If we do add pretty bars back we'll make sure that dual RAM targets are properly supported this time around.

0xc0170, 2018-05-22:
@geky I'll close this as won't fix

closed 2018-05-22 by 0xc0170