ARMmbed/mbed-os #6335: Exporting to Makefilefile fails to build on larger project in Windows

SeppoTakalo, 2018-03-12:
### Description
- Type: Bug
- Priority: Minor

## Bug

**Target**
Any

**Toolchain:**
Any Makefile based export.

**Expected behavior**
`mbed export -m <board> -i make_gcc_arm`
`make`

**Actual behavior**
Error code:
`make (e=87): The parameter is incorrect.`

This is produced because the final link command is 44,252 characters long.

Windows CreateProcess() only accepts command line maximum length to be 32,768 characters. [See here.](https://msdn.microsoft.com/en-us/library/windows/desktop/ms682425(v=vs.85).aspx)

Similar issue described here: https://stackoverflow.com/questions/12598933/ndk-build-createprocess-make-e-87-the-parameter-is-incorrect

**Steps to reproduce**
Import mbed-os-example-mesh-minimal
`mbed export -m <board> -i make_gcc_arm`

**Workarounds**
1. Don't use Makefile builds on Windows.

theotherjimmy, 2018-03-12:
Yeah, there's not a whole lot I can do about that. Known issue?

MarceloSalazar, 2018-03-19:
FYI @adbridge

theotherjimmy, 2018-03-19:
@MarceloSalazar What do you want Anna to do about this? Is there something I can do?

MarceloSalazar, 2018-03-19:
I've just wanted Anna to be aware of this, as may need to go to the known issue list.

theotherjimmy, 2018-03-19:
I can add it. Would you like a known issue?

theotherjimmy, 2018-03-19:
Known issue added.

sg-, 2018-03-26:
Is there a workaround for this based on how close to root you place the exported project or are all these paths relative to the extract point on disc and due to the amount of features and source code and path expansion in Mbed OS for a particular target?

theotherjimmy, 2018-03-26:
@sg- I'd have to check, but I think you may be able to  work around this by extracting into an already short path. That being said, if the project exceeds the path limit without any prefixes, then there is no workaround (other than use another exporter, or use Linux)

SeppoTakalo, 2018-03-26:
I cannot remember whether it used relative or absolute path names...
I would assume that relative, which would mean that no workarounds exists, other than using proper operating system.

sg-, 2018-03-26:
> proper operating system.

🤣

theotherjimmy, 2018-03-26:
XD "proper" operating system

theotherjimmy, 2018-04-24:
https://github.com/ARMmbed/mbed-os-example-bootloader/issues/58 is a duplicate.

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools Can this be closed or still expecting fixes?

SeppoTakalo, 2018-05-31:
@0xc0170 Why would you close issues without fixing them?

Eventually this will pop up again, and we should be able to link to existing issue.

SeppoTakalo, 2018-06-01:
There are various workaround to fix this.

For ARM linker there is `--via` option to allow specifying the filenames in a file: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0493e/CHDJEJCA.html

For GCC this probably needs working with the linker script file, as the equivalent of `--via` does not exist: http://ftp.gnu.org/old-gnu/Manuals/ld-2.9.1/html_mono/ld.html#SEC5

For IAR, the linker has `-f` option that allows specifying all command line parameters in a file.
http://netstorage.iar.com/SuppDB/Public/UPDINFO/006738/arm/doc/EWARM_DevelopmentGuide.ENU.pdf

theotherjimmy, 2018-06-01:
Yes @SeppoTakalo That's how things compile in `mbed compile`. It's a bit of work to include these fixes into the makefiles.

robmosys, 2018-07-16:
A colleague of mine just hit this bug.
For gcc, it should be possible to use the `@file` option to gcc to pass this to linker.
( https://gcc.gnu.org/onlinedocs/gcc-8.1.0/gcc/Overall-Options.html#Overall-Options )
Will test this solution and report back.

Also this page may be of importance:
https://community.arm.com/processors/f/discussions/4748/is-there-a-maximum-length-of-compile-path
Some people may be using old versions of the make command.

theotherjimmy, 2018-07-16:
@robmosys note that `mbed compile` already does many optimizations for all compilers to get around these limits. The make exporters need to be updated to include the same code.

robmosys, 2018-07-18:
In summary both my workaround and the mbed compile command use response files.
I also found out why workaround didn't work on wsl:
My wsl was an old Ununtu with 3.8 version of make. I think the file command was introduced with the 4.0 version of make in 2013.

theotherjimmy, 2018-07-19:
@robmosys Yep. (I have yet to see any offiical documentation for "Response files" I have seen them called "command files" (personal favorite term) or "via files").

russetrob, 2018-07-22:
Regarding the term "Response files" took this from the source, after a quick google for validity: It's used but not so commonly.
See `self.RESPONSE_FILES` usage in the code snippets you pointed at above.

robmosys, 2018-07-22:
The make code snippet for linking above wouldn't work on WSL due to space before comma.
Here's an update that should work on wsl.
Only change is removal of space between `in` and `,`
```make
$(PROJECT).elf: $(OBJECTS) $(SYS_OBJECTS) $(PROJECT).link_script.ld 
	$(file > $@.in, $(filter %.o, $^))
	+@echo "link: $(notdir $@)"
	$(LD)   $(LD_FLAGS) -T $(filter-out %.o, $^) $(LIBRARY_PATHS) --output $@ @$@.in $(LIBRARIES) $(LD_SYS_LIBS) 
````

SeppoTakalo, 2018-07-23:
Why the use of `$(filter ...)` instead of just `$(OBJECTS) $(SYS_OBJECTS)`?

Those macros might contain (not sure) other than object files as well. For example binary libraries `.a` or `.ar`

theotherjimmy, 2018-07-23:
@SeppoTakalo So that you can add more dependencies to the elf file, and expect them to work. Those macros do not contain the binary libraries.

theotherjimmy, 2018-07-23:
@robmosys Thanks for the prototype. I'll send a PR soon doing the same thing for all of the compilers shortly.

theotherjimmy, 2018-07-23:
see  #7583

JojoS62, 2018-08-27:
I'm getting the error 87 now also with an export to gnuarmeclipse.  Is there any workaround known? As far as I understand the problem is solved by using a response file for the linker, but this is not yet implemented for the gnuarmeclipse exporter?

EduardShaid, 2018-08-28:
All of the above methods do not solve anything on the merits !

robmosys, 2018-08-30:
@theotherjimmy in your pull request using response files, you had difficulty finding a modern make on Windows. I got my version of make from here: https://gnu-mcu-eclipse.github.io/windows-build-tools/

EduardShaid, 2018-08-31:
thanks for the answer, but it also did not help, because there is still a bug at the level of the CreateProcess(...)

ConductedForce, 2018-09-22:
Tried @robmosys fix. Works with Make GCC Arm toolchain export. Had this problem when I imported direct from github into Mbed Online and didn't use stock example projects.

closed 2018-10-02 by adbridge

robmosys, 2018-11-02:
Can someone tell me what the future of this bug is?
I'm trying to integrate a fix in the most mbed compatible way, for our company, and patch  #7559 looks good.
But as this patch hasn't been accepted and this bug is closed, I'm wondering how long the patch will be applicable to the tree. Currently I'm inclined to make a local fork, apply the patch and maintain it as necessary.

cmonr, 2018-12-14:
@robmosys Sorry we missed your comment.

If this is still an issue, would you mind opening a new issue?

srberard, 2018-12-15:
I'm experiencing the same issue as well, not sure why this issue was closed without the fix being merged in.

cmonr, 2018-12-17:
@srberard @robmosys This PR was likely closed because at the time, the issue was resolved.

Because Windows filepaths are fundamentally shorter than *nix limitations, Windows has been rather suceptable to this problem. We're working on a more permanent fix, but it tends to come back every once in a while.

Please open new issues with steps to help us reproduce. We're working on a couple of issues and tests to prevent this from regressing.

cmonr, 2018-12-18:
@srberard I'm confused. https://github.com/ARMmbed/mbed-os/pull/7559 was accepted and merged before their comment was made: https://github.com/ARMmbed/mbed-os/issues/6335#issuecomment-435458936.

As for the issue, we prefer to have issues opened a-new instead of having older issues ressurected (see https://github.com/ARMmbed/mbed-os/issues/6335#event-1879426534). It's the same reason that forums generally frown upon users that try to ressurect dead forum threads.

I'd be interested to see the diff of your fix, as well as to confirm your tool versions. Part of the finding fixes for this is that we only support a certain set of tools, so one solution that works might not even exist in an older version.

ConductedForce, 2019-01-31:
Faced this again using an Eclipse_GCC_ARM export from online compiler. Came back because I knew a fix was here.