ARMmbed/mbed-os #6339: 2 Questions about "targets.json"

0x6d61726b, 2018-03-12:
## Question

- Is it intended that the `"Super_Target"` does not contain the defintion `"public": false`?
- Are `"target_overrides"` expected to be present here instead within the corresponding library?

0xc0170, 2018-03-13:
> Is it intended that the "Super_Target" does not contain the definition "public": false?

It inherits it so public is set  to false.

> Are "target_overrides" expected to be present here (see "HI2110") instead within the corresponding library?

I believe that should be a regular config not target overrides. We should fix that

0xc0170, 2018-03-13:
@RobMeades Can you please review the second question? `target_overrides` in the HI2110 target

RobMeades, 2018-03-13:
That line was put in to reduce the image size, though I have a feeling it might not actually do what I intended.  Anyway, is the suggestion that, rather than:

```
        "target_overrides": {
            "*": {
                "core.stdio-flush-at-exit": false
            }
        },
```

...it should be:

`"core.stdio-flush-at-exit": false,`

0x6d61726b, 2018-03-13:
> It inherits it so public is set to false.

@0xc0170: I had the feeling that the "public" property is excluded from inheritance, because:

`LPC1768` (does not have a "public" property) but inherits `LPCTarget` (that sets `"public": false`) that inherits `Target` (that sets `"public": false`). 

`NUCLEO_F429ZI` (does not have a "public" property) but inherits `FAMILY_STM32` (that sets `"public": false`) that intherits `Target` (that sets `"public": false`).

RobMeades, 2018-03-13:
I just dredge up the comment from Sam Grove when I originally put this value in, which was that it should actually be `platform.stdio-flush-at-exit`.  In a build for another randomly chosen platform, which doesn't fiddle with the value, this lands up as `true` in an `mbed_config.h`:

`#define MBED_CONF_PLATFORM_STDIO_FLUSH_AT_EXIT            1               // set by library:platform`

0x6d61726b, 2018-03-13:
@RobMeades: Thanks for the explanation. I was confused about the `core.`.

cmonr, 2018-06-21:
Closing since it appears that this question was answered already. Thanks @bridadan!

closed 2018-06-21 by cmonr