ARMmbed/mbed-os #6342: Fail to run Greentea test on inherited target

ccli8, 2018-03-13:
### Description
- Type: Enhancement
- Priority: Major

-----------------------------------------------------------------
## Enhancement

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
ARM

**mbed-cli version:** 
1.4.0

**python**
2.7.1

**mbed-os sha:**
f67fe4a Merge pull request #6257 from SenRamakri/sen_FaultHandlerFixes

-----------------------------------------------------------------

If I run on original target **NUMAKER_PFM_NUC472**:
```
mbed test -m NUMAKER_PFM_NUC472 -t ARM
```
Everything goes fine.

-----------------------------------------------------------------

If I add target **NUMAKER_PFM_NUC472_NOXRAM** inheriting **NUMAKER_PFM_NUC472** in `mbed_app.json`:
```
    "NUMAKER_PFM_NUC472_NOXRAM": {
        "inherits": ["NUMAKER_PFM_NUC472"],
        "extra_labels_remove": ["NU_XRAM_SUPPORTED"],
        "extra_labels_add": ["NU_XRAM_UNSUPPORTED"]
    }
```
And run:
```
mbed test -m NUMAKER_PFM_NUC472_NOXRAM -t ARM
```
I get error:
```
usage: test.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-D MACROS]
               [-j JOBS] [--source SOURCE_DIR] [--build BUILD_DIR] [-l]
               [-p PATHS] [-f FORMAT] [--continue-on-build-fail] [-n NAMES]
               [--test-config TEST_CONFIG] [--test-spec TEST_SPEC]
               [--build-report-junit BUILD_REPORT_JUNIT]
               [--build-data BUILD_DATA] [-v] [--stats-depth STATS_DEPTH]
test.py: error: argument -m/--mcu: NUMAKER_PFM_NUC472_NOXRAM is not a supported MCU. Supported MCUs are:
ARCH_BLE,              ARCH_BLE_BOOT,         ARCH_BLE_OTA,
ARCH_GPRS,             ARCH_LINK,             ARCH_LINK_BOOT,
ARCH_LINK_OTA,         ARCH_MAX,              ARCH_PRO,
ARM_BEETLE_SOC,        ARM_CM3DS_MPS2,        ARM_IOTSS_BEID,
ARM_MPS2_M0,           ARM_MPS2_M0P,          ARM_MPS2_M1,
ARM_MPS2_M3,           ARM_MPS2_M4,           ARM_MPS2_M7,
B96B_F446VE,           BLUEPILL_F103C8,       DELTA_DFBM_NQ620,
DELTA_DFCM_NNN40,      DELTA_DFCM_NNN40_BOOT, DELTA_DFCM_NNN40_OTA,
DELTA_DFCM_NNN50,      DELTA_DFCM_NNN50_BOOT, DELTA_DFCM_NNN50_OTA,
DISCO_F051R8,          DISCO_F100RB,          DISCO_F303VC,
DISCO_F334C8,          DISCO_F401VC,          DISCO_F407VG,
DISCO_F413ZH,          DISCO_F429ZI,          DISCO_F469NI,
DISCO_F746NG,          DISCO_F769NI,          DISCO_L053C8,
DISCO_L072CZ_LRWAN1,   DISCO_L475VG_IOT01A,   DISCO_L476VG,
EFM32GG_STK3700,       EFM32HG_STK3400,       EFM32LG_STK3600,
EFM32PG12_STK3402,     EFM32PG_STK3401,       EFM32WG_STK3800,
EFM32ZG_STK3200,       ELEKTOR_COCORICO,      ELMO_F411RE,
EV_COG_AD3029LZ,       EV_COG_AD4050LZ,       FF1705_L151CC,
FF_LPC546XX,           GR_LYCHEE,             HEXIWEAR,
HRM1017,               HRM1017_BOOT,          HRM1017_OTA,
K20D50M,               K22F,                  K64F,
K66F,                  K82F,                  KL05Z,
KL25Z,                 KL26Z,                 KL27Z,
KL43Z,                 KL46Z,                 KL82Z,
KW24D,                 KW41Z,                 LPC1114,
LPC11C24,              LPC11U24,              LPC11U24_301,
LPC11U34_421,          LPC11U35_401,          LPC11U35_501,
LPC11U35_501_IBDAP,    LPC11U35_Y5_MBUG,      LPC11U37H_401,
LPC11U37_501,          LPC11U68,              LPC1347,
LPC1549,               LPC1768,               LPC1769,
LPC4088,               LPC4088_DM,            LPC4330_M0,
LPC4330_M4,            LPC4337,               LPC54114,
LPC546XX,              LPC810,                LPC812,
LPC824,                LPCCAPPUCCINO,         MAX32600MBED,
MAX32620HSP,           MAX32625MBED,          MAX32625NEXPAQ,
MAX32630FTHR,          MAXWSNENV,             MBED_CONNECT_ODIN,
MCU_LPC4088,           MCU_LPC546XX,          MICRONFCBOARD,
MIMXRT1050_EVK,        MODULE_UBLOX_ODIN_W2,  MOTE_L152RC,
MTB_ADV_WISE_1510,     MTB_ADV_WISE_1570,     MTB_LAIRD_BL600,
MTB_LAIRD_BL652,       MTB_MTS_DRAGONFLY,     MTB_MTS_XDOT,
MTB_MURATA_ABZ,        MTB_MXCHIP_EMW3166,    MTB_UBLOX_NINA_B1,
MTB_UBLOX_ODIN_W2,     MTM_MTCONNECT04S,      MTM_MTCONNECT04S_BOOT,
MTM_MTCONNECT04S_OTA,  MTS_DRAGONFLY_F411RE,  MTS_GAMBIT,
MTS_MDOT_F405RG,       MTS_MDOT_F411RE,       NCS36510,
NRF51822,              NRF51822_BOOT,         NRF51822_OTA,
NRF51822_Y5_MBUG,      NRF51_DK,              NRF51_DK_BOOT,
NRF51_DK_LEGACY,       NRF51_DK_OTA,          NRF51_DONGLE,
NRF51_DONGLE_BOOT,     NRF51_DONGLE_LEGACY,   NRF51_DONGLE_OTA,
NRF51_MICROBIT,        NRF51_MICROBIT_B,      NRF51_MICROBIT_BOOT,
NRF51_MICROBIT_B_BOOT, NRF51_MICROBIT_B_OTA,  NRF51_MICROBIT_OTA,
NRF52840_DK,           NRF52_DK,              NUCLEO_F030R8,
NUCLEO_F031K6,         NUCLEO_F042K6,         NUCLEO_F070RB,
NUCLEO_F072RB,         NUCLEO_F091RC,         NUCLEO_F103RB,
NUCLEO_F207ZG,         NUCLEO_F302R8,         NUCLEO_F303K8,
NUCLEO_F303RE,         NUCLEO_F303ZE,         NUCLEO_F334R8,
NUCLEO_F401RE,         NUCLEO_F410RB,         NUCLEO_F411RE,
NUCLEO_F412ZG,         NUCLEO_F413ZH,         NUCLEO_F429ZI,
NUCLEO_F439ZI,         NUCLEO_F446RE,         NUCLEO_F446ZE,
NUCLEO_F746ZG,         NUCLEO_F756ZG,         NUCLEO_F767ZI,
NUCLEO_L011K4,         NUCLEO_L031K6,         NUCLEO_L053R8,
NUCLEO_L073RZ,         NUCLEO_L152RE,         NUCLEO_L432KC,
NUCLEO_L433RC_P,       NUCLEO_L476RG,         NUCLEO_L486RG,
NUCLEO_L496ZG,         NUCLEO_L496ZG_P,       NUMAKER_PFM_M453,
NUMAKER_PFM_M487,      NUMAKER_PFM_NANO130,   NUMAKER_PFM_NUC472,
NZ32_SC151,            OC_MBUINO,             OSHCHIP,
RBLAB_BLENANO,         RBLAB_BLENANO2,        RBLAB_BLENANO_BOOT,
RBLAB_BLENANO_OTA,     RBLAB_NRF51822,        RBLAB_NRF51822_BOOT,
RBLAB_NRF51822_OTA,    REALTEK_RTL8195AM,     RO359B,
RZ_A1H,                RZ_A1XX,               SAKURAIO_EVB_01,
SAMD21G18A,            SAMD21J18A,            SAMG55J19,
SAML21J18A,            SAMR21G18A,            SARA_NBIOT_EVK,
SEEED_TINY_BLE,        SEEED_TINY_BLE_BOOT,   SEEED_TINY_BLE_OTA,
SILICA_SENSOR_NODE,    SSCI824,               STEVAL_3DP001V1,
Super_Target,          TB_SENSE_1,            TB_SENSE_12,
TEENSY3_1,             TMPM066,               TY51822R3,
TY51822R3_BOOT,        TY51822R3_OTA,         UBLOX_C027,
UBLOX_C030_N211,       UBLOX_C030_R410M,      UBLOX_C030_U201,
UBLOX_EVA_NINA,        UBLOX_EVK_NINA_B1,     UBLOX_EVK_ODIN_W2,
UBRIDGE,               USENSE,                VBLUNO51,
VBLUNO51_BOOT,         VBLUNO51_LEGACY,       VBLUNO51_OTA,
VBLUNO52,              VK_RZ_A1H,             WALLBOT_BLE,
WALLBOT_BLE_BOOT,      WALLBOT_BLE_OTA,       WIZWIKI_W7500,
WIZWIKI_W7500ECO,      WIZWIKI_W7500P,        XADOW_M0,
XBED_LPC1768,          XDOT_L151CC
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\test.py -t ARM -m NUMAKER_PFM_NUC472_NOXRAM --source C:\ccli8\iot\mbed\mbed-os\nu-greentea --build C:\ccli8\iot\mbed\mbed-os\nu-greentea\BUILD\tests\NUMAKER_PFM_NUC472_NOXRAM\ARM --test-spec C:\ccli8\iot\mbed\mbed-os\nu-greentea\BUILD\tests\NUMAKER_PFM_NUC472_NOXRAM\ARM\test_spec.json" in "C:\ccli8\iot\mbed\mbed-os\nu-greentea"
---
```

-----------------------------------------------------------------

If I add target **NUMAKER_PFM_NUC472_NOXRAM** inheriting **NUMAKER_PFM_NUC472** in `mbed-os/targets/targets.json`:
```
{
   "NUMAKER_PFM_NUC472_NOXRAM": {
        "inherits": ["NUMAKER_PFM_NUC472"],
        "extra_labels_remove": ["NU_XRAM_SUPPORTED"],
        "extra_labels_add": ["NU_XRAM_UNSUPPORTED"]
    }
}
```
And run:
```
mbed test -m NUMAKER_PFM_NUC472_NOXRAM -t ARM
```
I get error:
```
mbedgt: greentea test automation tool ver. 1.3.1
mbedgt: test specification file 'C:\ccli8\iot\mbed\mbed-os\nu-greentea\BUILD\tests\NUMAKER_PFM_NUC472_NOXRAM\ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\ccli8\iot\mbed\mbed-os\nu-greentea\BUILD\tests\NUMAKER_PFM_NUC472_NOXRAM\ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
[1520919809.70][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1520919810.22][root]Found htm target id, 1302000000C60021011DAC8700000066, for usb target id 1302000000C60021011DAC8700000066
mbedgt: detected 1 device
mbedgt: processing target 'NUMAKER_PFM_NUC472_NOXRAM' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.1704255073
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.66 sec
mbedgt: exited with code -110
[mbed] ERROR: "mbedgt" returned error code -110.
[mbed] ERROR: Command "mbedgt --test-spec C:\ccli8\iot\mbed\mbed-os\nu-greentea\BUILD\tests\NUMAKER_PFM_NUC472_NOXRAM\ARM\test_spec.json" in "C:\ccli8\iot\mbed\mbed-os\nu-greentea"
---

```

0xc0170, 2018-03-13:
I can not reproduce. I added lines

```
    "NUMAKER_PFM_NUC472_NOXRAM": {
        "inherits": ["NUMAKER_PFM_NUC472"],
        "extra_labels_remove": ["NU_XRAM_SUPPORTED"],
        "extra_labels_add": ["NU_XRAM_UNSUPPORTED"]
    },
```

after `NUMAKER_PFM_NUC472` and mbed test does compile for me. Even if I do only `mbed test -m NUMAKER_PFM_NUC472_NOXRAM -t GCC_ARM --list`, it prints available tests.

cmonr, 2018-06-21:
@ccli8 Could you answer @0xc0170 request for a reproduction of the issue? If not, this issue will be closed soon.

theotherjimmy, 2018-06-21:
@ccli8 You can't add targets in `mbed_app.json` please use `custom_targets.json` instead.

theotherjimmy, 2018-06-21:
@ccli8 For the second, greatea error, you need to tell `mbedls` how to identify the target correctly, and the target must be plugged in. You will have to use `mbedls`'s mock functionality, https://github.com/ARMmbed/mbed-ls#mocking-renaming-platforms

ccli8, 2018-06-22:
> @ccli8 Could you answer @0xc0170 request for a reproduction of the issue? If not, this issue will be closed soon.

@0xc0170 @cmonr Sorry, I miss it.

@theotherjimmy Checking on last mbed-os e8ec3614b0a2573b397c709b6a40e66e3c0c830f, I try `custom_targets.json` but fail:

**custom_targets.json:**
```
{
   "NUMAKER_PFM_NUC472_NOXRAM": {
        "inherits": ["NUMAKER_PFM_NUC472"],
        "extra_labels_remove": ["NU_XRAM_SUPPORTED"],
        "extra_labels_add": ["NU_XRAM_UNSUPPORTED"]
    }
}
```

Run Greentea test:
```
mbed test -m NUMAKER_PFM_NUC472_NOXRAM -t ARM -n mbed-os-tests-mbedmicro-mbed-cpp
```

But fail with:
```
$ mbed test -m NUMAKER_PFM_NUC472_NOXRAM -t ARM -n mbed-os-tests-mbedmicro-mbed-cpp
Scan: .
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Building library mbed-build (NUMAKER_PFM_NUC472_NOXRAM, ARM)
Scan: nu-greentea
Scan: ARM
Traceback (most recent call last):
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\test_api.py", line 2195, in build_test_worker
    bin_file = build_project(*args, **kwargs)
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\build_api.py", line 554, in build_project
    app_config=app_config, build_profile=build_profile, ignore=ignore)
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\build_api.py", line 328, in prepare_toolchain
    config = config or Config(target, src_paths, app_config=app_config)
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\config\__init__.py", line 477, in __init__
    self.app_config_data.get("custom_targets", {}), tgt)
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\targets\__init__.py", line 122, in generate_py_target
    return target(name, total_data)
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\targets\__init__.py", line 101, in target
    resolution_order = get_resolution_order(json_data, name, [])
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\targets\__init__.py", line 93, in get_resolution_order
    parents = json_data[target_name].get("inherits", [])
KeyError: u'NUMAKER_PFM_NUC472_NOXRAM'
Traceback (most recent call last):
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\test.py", line 247, in <module>
    ignore=options.ignore)
  File "C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\test_api.py", line 2302, in build_tests
    report_entry[worker_result['kwargs']['project_id'].upper()][0][0]['output'] = new_notify.get_output()
KeyError: 'MBED-OS-TESTS-MBEDMICRO-MBED-CPP'
[ERROR] 'MBED-OS-TESTS-MBEDMICRO-MBED-CPP'
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\ccli8\iot\mbed\mbed-os\nu-greentea\mbed-os\tools\test.py -t ARM -m NUMAKER_PFM_NUC472_NOXRAM --source C:\ccli8\iot\mbed\mbed-os\nu-greentea --build C:\ccli8\iot\mbed\mbed-os\nu-greentea\BUILD\tests\NUMAKER_PFM_NUC472_NOXRAM\ARM --test-spec C:\ccli8\iot\mbed\mbed-os\nu-greentea\BUILD\tests\NUMAKER_PFM_NUC472_NOXRAM\ARM\test_spec.json -n mbed-os-tests-mbedmicro-mbed-cpp" in "C:\ccli8\iot\mbed\mbed-os\nu-greentea"
---
```

theotherjimmy, 2018-06-22:
It looks like you did not build a test there. Could you check that? If you don't build a test, you may run into that trace back.

ccli8, 2018-06-25:
@theotherjimmy Yes, the test is not built. But I expect the test would be automatically built even though I run test on custom target:
<pre>
mbed test -m <b>NUMAKER_PFM_NUC472_NOXRAM</b> -t ARM -n mbed-os-tests-mbedmicro-mbed-cpp
</pre>

Compared to above, if I run test on native target, it is OK. 
<pre>
mbed test -m <b>NUMAKER_PFM_NUC472</b> -t ARM -n mbed-os-tests-mbedmicro-mbed-cpp
</pre>

theotherjimmy, 2018-06-25:
@ccli8 I can't reproduce this with the latest master.

mattbrown015, 2018-06-29:
@theotherjimmy I wonder if you can help me?  I came to this issue because I was getting the `KeyErr` when I tried to build for a custom target.  I started trying understand `TESTS` but have now got completely lost.

I added `TESTS\group\case\main.cpp` to our project.  `main.cpp` only contains 3 lines, `int main() {}`!

Initially I tried to specify our custom target and got the `KeyError` so I just changed to specifying `NUCLEO_L433RC_P`.

I just did this from the project directory:

    `mbed test --compile -m NUCLEO_L433RC_P -n tests-group-case`

but this fails because of linker errors.  `mbed test` is building the project root and finding the application's `main`.

Is testing supposed to work for applications or can it only be used for libs?

I've been starring at [Testing Applications](https://os.mbed.com/docs/v5.9/tools/testing-applications.html) but quite clearly I'm missing something!  Perhaps I'm having a Friday afternoon moment.  Thanks.

`mbed --version 1.7.3` mbed-os is 5bf483e

theotherjimmy, 2018-06-29:
@mattbrown015 I was able to add the following `custom_targets.json` and compile tests successfully:
```
{
    "MY_TARGET": {
        "inherits": ["NUCLEO_F411RE"]
    }
}
```
I cannot reproduce the `KeyError` you are seeing.

----------------------------

WRT the duplicate main function, yes that's a know issues with placing tests in the same project as an application.

mattbrown015, 2018-06-29:
@theotherjimmy Hhhmm, that's odd my custom target looks exactly like yours except I inherit from `NUCLEO_L432KC`.  Could it be target specific?  Could it be tool version specific?

But, it's a bit academic if I can't create tests for an application.  Any thoughts about work arounds?  Use a macro defined as `main` for the normal build and a second config that defines it has `not_main` for the test build??  Somewhat cryptic!  I might try on Monday.

theotherjimmy, 2018-06-29:
That'll probably work for you.

theotherjimmy, 2018-06-29:
Make sure that you spelled the target correctly in the inherits, and that it's a list. Those may cause keyerrors.

mattbrown015, 2018-07-02:
@theotherjimmy I copied your `custom_targets.json` from above, still no joy.

The error I see is almost exactly what @ccli8 has posted above so I haven't bothered posting it again.

AFAICT, I'm using a different `mbed-cli` and repo version to @ccli8. So no clues there.

I'm building an app so I have a `mbed_app.json` rather than an `mbed_lib.json`. Could it be anything to do with that?

My only other thoughts was that I using Windows.

mattbrown015, 2018-07-02:
@theotherjimmy As regards the possibly more important issue of creating a test framework for an application I've decided it's more complicated that just having multiple definitions of `main`.

Any statically allocated objects from the application code will also be crated and initialised and that's not really what I want.

I was hoping to use the test framework to create some small tests to test, in isolation, some small parts of our application.

It's quite possible that was I thinking of doing doesn't really make any sense.  I shall think about it in the background!

theotherjimmy, 2018-07-02:
@ccli8 I used:
```
commit e8ec3614b0a2573b397c709b6a40e66e3c0c830f
Merge: 95d2b3d55 8efe3b5bc
Author: Cruz Monrreal <Cruz.Monrreal@arm.com>
Date:   Thu Jun 21 18:23:07 2018 -0500

    Merge pull request #7291 from aqib-ublox/master
    
    adding USTICKER label for C027

```
, the exact commit you mentioned in your second comment. I also used the `custom_targets.json` from my comment. I compiled with `mbed test --compile -t arm -m MY_TARGET`, and I could not reproduce the problem you reported. You might have stale `.pyc` file? you could try running `git clean -xf` in your `mbed-os` dir.

mattbrown015, 2018-07-02:
I did `git clean -xf` but it didn't make any difference. It's a mystery!

`git clean -xf` is a good tip though, I might do it more often in future. :-)

mattbrown015, 2018-07-02:
Just seen [Tests not possible with costum target #7395](https://github.com/ARMmbed/mbed-os/issues/7395) which seems to be discussing the problem with custom targets.

ccli8, 2018-07-03:
@theotherjimmy I still meet the error even though I clean up `*.pyc` files with `git clean -xf`.

EmbedEngineer, 2018-07-03:
@theotherjimmy As @mattbrown015 mentioned, I stumbled upon the same problem, that I cannot compile any test if I use a custom defined target. Please have a look at my issue report #7395.

I my opinion it is independend from the mbed cli version and just is a bug in the mbed os tools.

EmbedEngineer, 2018-07-03:
For tracking: my custom_targets.json looks like this:
```
{
    "MY_TARGET": {
        "inherits": [
            "EFM32PG12B500F1024GL125"
        ],
        "device_has": [
            "ANALOGIN",
            "I2C",
            "I2CSLAVE",
            "I2C_ASYNCH",
            "INTERRUPTIN",
            "LPTICKER",
            "PORTIN",
            "PORTINOUT",
            "PORTOUT",
            "PWMOUT",
            "RTC",
            "SERIAL",
            "SERIAL_ASYNCH",
            "SLEEP",
            "SPI",
            "SPISLAVE",
            "SPI_ASYNCH",
            "STDIO_MESSAGES",
            "USTICKER",
            "TRNG",
            "FLASH"
        ],
        "forced_reset_timeout": 2,
        "OUTPUT_EXT": "hex",
        "config": {
            "hf_clock_src": {
                "help": "Value: HFXO for external crystal, HFRCO for internal RC oscillator",
                "value": "HFXO",
                "macro_name": "CORE_CLOCK_SOURCE"
            },
            "hfxo_clock_freq": {
                "help": "Value: External crystal frequency in hertz",
                "value": "40000000",
                "macro_name": "HFXO_FREQUENCY"
            },
            "lf_clock_src": {
                "help": "Value: LFXO for external crystal, LFRCO for internal RC oscillator, ULFRCO for internal 1KHz RC oscillator",
                "value": "LFXO",
                "macro_name": "LOW_ENERGY_CLOCK_SOURCE"
            },
            "lfxo_clock_freq": {
                "help": "Value: External crystal frequency in hertz",
                "value": "32768",
                "macro_name": "LFXO_FREQUENCY"
            },
            "hfrco_clock_freq": {
                "help": "Value: Frequency in hertz, must correspond to setting of hfrco_band_select",
                "value": "38000000",
                "macro_name": "HFRCO_FREQUENCY"
            },
            "hfrco_band_select": {
                "help": "Value: One of cmuHFRCOFreq_1M0Hz, cmuHFRCOFreq_2M0Hz, cmuHFRCOFreq_4M0Hz, cmuHFRCOFreq_7M0Hz, cmuHFRCOFreq_13M0Hz, cmuHFRCOFreq_16M0Hz, cmuHFRCOFreq_19M0Hz, cmuHFRCOFreq_26M0Hz, cmuHFRCOFreq_32M0Hz, cmuHFRCOFreq_38M0Hz. Be sure to set hfrco_clock_freq accordingly!",
                "value": "cmuHFRCOFreq_38M0Hz",
                "macro_name": "HFRCO_FREQUENCY_ENUM"
            },
            "board_controller_enable": {
                "help": "Pin to pull high for enabling the USB serial port",
                "value": "PA5",
                "macro_name": "EFM_BC_EN"
            }
        },
        "target_overrides": {
            "events.use-lowpower-timer-ticker": 1
        }
    }
}
```
I basically just did the the custom_target because we want to have our own pinnames.h and use a different clock. I knwo we could have just done the same thing in the mbed_app.json, but since we will have multiple versions of this boards, this would fit our needs much better.

theotherjimmy, 2018-07-03:
> ```json
"target_overrides": {
            "events.use-lowpower-timer-ticker": 1
        }
```
This won't work.

theotherjimmy, 2018-07-03:
@EmbedEngineer on current master, with your `custom_targets.json` I get:
```
Compile [  0.2%]: mbed_tz_context.c
Compile [  0.3%]: BusInOut.cpp
[Error] objects.h@27,0:  #5: cannot open source input file "PinNames.h": No such file or directory
Failed to build library
```

Which indicates that I'm at the step where I would need parts of your board port.

EmbedEngineer, 2018-07-04:
@theotherjimmy Sure thank you for trying it out. 

My `PinNames.h` is:
```
#ifndef MBED_PINNAMES_H
#define MBED_PINNAMES_H

#include "CommonPinNames.h"

#ifdef __cplusplus
extern "C"
{
#endif

    typedef enum
    {
        EFM32_STANDARD_PIN_DEFINITIONS,

        LED0 = PF4,
        LED1 = PF5,

        /* Serial (just some usable pins) */
        SERIAL_TX = PA6,
        SERIAL_RX = PA7,

        /* Board Controller UART (USB)*/
        USBTX = PA0,
        USBRX = PA1,

        /* Board Controller */
        STDIO_UART_TX = USBTX,
        STDIO_UART_RX = USBRX
    } PinName;

#ifdef __cplusplus
}
#endif

#endif
```

Furthermore you will need a `device_peripherals.h`:
```
#ifndef MBED_DEVICE_PERIPHERALS_H
#define MBED_DEVICE_PERIPHERALS_H

/* us ticker */
#define US_TICKER_TIMER         TIMER0
#define US_TICKER_TIMER_CLOCK   cmuClock_TIMER0
#define US_TICKER_TIMER_IRQ     TIMER0_IRQn

/* PWM */
#define PWM_TIMER        TIMER1
#define PWM_TIMER_CLOCK  cmuClock_TIMER1
#define PWM_ROUTE        TIMER_ROUTE_LOCATION_LOC1

/* Crystal calibration */
#if !defined(CMU_HFXOINIT_STK_DEFAULT)
#define CMU_HFXOINIT_STK_DEFAULT                                                \
{                                                                               \
  true,         /* Low-power mode for EFM32 */                                  \
  false,        /* Disable auto-start on EM0/1 entry */                         \
  false,        /* Disable auto-select on EM0/1 entry */                        \
  false,        /* Disable auto-start and select on RAC wakeup */               \
  _CMU_HFXOSTARTUPCTRL_CTUNE_DEFAULT,                                           \
  0x142,        /* Steady-state CTUNE for STK boards without load caps */       \
  _CMU_HFXOSTEADYSTATECTRL_REGISH_DEFAULT,                                      \
  0x20,         /* Matching errata fix in CHIP_Init() */                        \
  0x7,          /* Recommended steady-state osc core bias current */            \
  0x6,          /* Recommended peak detection threshold */                      \
  _CMU_HFXOTIMEOUTCTRL_SHUNTOPTTIMEOUT_DEFAULT,                                 \
  0xA,          /* Recommended peak detection timeout  */                       \
  0x4,          /* Recommended steady timeout */                                \
  _CMU_HFXOTIMEOUTCTRL_STARTUPTIMEOUT_DEFAULT,                                  \
  cmuOscMode_Crystal,                                                           \
}
#endif

/* DCDC settings */
#if !defined(EMU_DCDCINIT_STK_DEFAULT)
#define EMU_DCDCINIT_STK_DEFAULT EMU_DCDCINIT_DEFAULT
#endif
#endif
```

I put both files in a TARGET_MY_TARGET folder.

theotherjimmy, 2018-07-05:
@EmbedEngineer I was able to compile mbed and tests with those headers. I cannot reproduce your issue.

Maybe you're placing `custom_targets.json` in the wrong location? Where have you put it in your project? does `mbed compile -m MY_TARGET` work?

EmbedEngineer, 2018-07-06:
@theotherjimmy I put the `custom_targets.json` in the root directory (my workspace folder where I start mbed compile).
`mbed compile -m MY_TARGET` works just fine. Maybe it is a problem because I am using the arm gcc compiler as I am supecting in my issue #7395?

theotherjimmy, 2018-07-06:
For good measure, I ran it with gcc_arm as well, and it works just fine for me.

aashishc1988, 2018-07-31:
The PR, https://github.com/ARMmbed/mbed-os/pull/7477, addresses the problem listed here, and on https://github.com/ARMmbed/mbed-os/issues/7395. Before that PR, i was able to reproduce the behavior as indicated by @ccli8 , and @EmbedEngineer, but not anymore. 

@ccli8 , please update your workspace to include those changes and try again. Let me know if you still the issue.

ccli8, 2018-08-01:
@aashishc1988 I test on `mbed-os-5.9.3` and master (63f62165d89f5562c529cd3ecb94823ce1dc7f13). The issue gets resolved on them.

closed 2018-08-01 by 0xc0170