ARMmbed/mbed-os #6347: Pull Request type information

bcostm, 2018-03-13:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

With the new PR template, we need to mention using a text box if the PR is a fix or a Refactor etc... The problem is that displays a "todo list" of 5 actions which is not what we want to see... and use todo list only when it is really necessary (missing commits, tests, revieweers, etc...)

[image]

**Suggested enhancement**

Use in the PR template:
```
[ ] Fix
[ ] Refactor
[ ] New target
[ ] Feature
[ ] Breaking change
```

instead of:
```
- [ ] Fix
- [ ] Refactor
- [ ] New target
- [ ] Feature
- [ ] Breaking change
```

0xc0170, 2018-03-13:
Thanks @bcostm . I've just noticed this in one of the recent issues, will review and fix

0xc0170, 2018-03-20:
Should be fixed via https://github.com/ARMmbed/mbed-os/pull/6348

closed 2018-05-22 by 0xc0170