ARMmbed/mbed-os #6355: GR-Peach / GR-LYCHEE Thread & Ticker Can't run at the same time

Meano, 2018-03-14:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue:
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
RZ_A1H|GR_LYCHEE

**Toolchain:**
GCC_ARM

**Toolchain version:**
GCC ARM 6-2017-Q2

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
af46177a6 (mbed-os-5.4 had no problem)

**DAPLink version:**
Unconcerned

**Expected behavior**
Thread and Ticker Callback Function runs Well

**Actual behavior**
Thread function stop running after start a while.

**Steps to reproduce**
compile the code and run a well, Thread function can't trigger breakpoint.

```
#include "mbed.h"
Ticker flipper;
Thread thread;
DigitalOut led1(LED1);
DigitalOut led2(LED2);
void  trd(){
    while(true){
	led1 = !led1;
	wait_ms(5);	 
    }
}
void flip() {
    led2 = !led2;
}

int main() {
    led2 = 1;
    flipper.attach(&flip, 0.001);
	thread.start(trd);

    while(1) {
        wait(0.2);
    }
}

```
@TomoYamanaka

TomoYamanaka, 2018-05-23:
Sorry for this late reply.
As you said, this will be resolve by adopting files I mentioned in #6617.
Could you please wait a little more until #6617 is fixed.

cmonr, 2018-06-21:
Closing since it appears this was resolved with https://github.com/ARMmbed/mbed-os/pull/7032

closed 2018-06-21 by cmonr