ARMmbed/mbed-os #6358: UARTSerial output does not work with REALTEK_RTL8195AM

SeppoTakalo, 2018-03-14:
### Description
- Type: Bug
- Target: REALTEK_RTL8195AM

Following example should work on all platforms, but crashes on REALTEK_RTL8195AM.

```
#include "mbed.h"

int main() {
    UARTSerial *serial = new UARTSerial(USBTX, USBRX);
    serial->set_baud(115200);

    FILE *out = fdopen(serial, "w");
    FILE *in = fdopen(serial, "r");

    fprintf(out, "Started!\r\n");

    while (true) {
        int c = fgetc(in);
        if (c != EOF) {
            fprintf(out, "Input: %c\r\n", c);
        } else {
            return 0;
        }
    }
}

```

This, however works:
```
#include "mbed.h"

int main() {
    UARTSerial *serial = new UARTSerial(USBTX, USBRX);
    serial->set_baud(115200);

    //FILE *out = fdopen(serial, "w");
    FILE *in = fdopen(serial, "r");

    printf("Started!\r\n");

    while (true) {
        int c = fgetc(in);
        if (c != EOF) {
            printf("Input: %c\r\n", c);
        } else {
            return 0;
        }
    }
}
```

Problem seems to be in the TX interrupt of the serial port.
Following code is able to crash the device using RawSerial:
```
#include "mbed.h"

int c = 0;
RawSerial *serial;

void tx_avail(void)
{
    while (serial->writeable())
        serial->putc((c++ % 95) + ' ');
}

int main() {
    serial = new RawSerial(USBTX, USBRX, 115200);

    core_util_critical_section_enter();
    serial->attach(tx_avail, SerialBase::TxIrq);
    tx_avail();
    core_util_critical_section_exit();

}
```

SeppoTakalo, 2018-03-14:
Output in Realtek is always only 4 characters. Then it crashes.

0xc0170, 2018-03-14:
@ARMmbed/team-realtek please review

M-ichae-l, 2018-03-20:
Hi, @SeppoTakalo 
I have used the latest Mbed-os to test with your first code example. There is no crashes on my side. Could you please double check and give more details on your first crash issue?

M-ichae-l, 2018-03-21:
Hi, 
I have uploaded a fix for LogUart Tx interrupt crach. 
Please refer to #6415 .

SeppoTakalo, 2018-03-21:
To replicate the issue:
```
mbed new serial-test
cd serial-test
cd mbed-os
git checkout master
cd ..
```
Now create `main.cpp` based on the last example I posted earlier.
```
mbed target REALTEK_RTL8195AM
mbed toolchain GCC_ARM
mbed compile
```

SeppoTakalo, 2018-03-21:
I can confirm that the fix is working.

To verify, follow the previous post on how to create an example application, then:
```
cd mbed-os
git fetch origin pull/6415/head:PR_6415
git checkout PR_6415
cd ..
mbed compile
```

And now the example code starts to work.

closed 2018-05-22 by 0xc0170