ARMmbed/mbed-os #6362: Waking STM32L073 from sleep

donpowrie, 2018-03-14:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2018-03-15:
@donpowrie Can you please provide more information about the bug? The target , toolchain, code snippet to reproduce it, expected vs actual behavior ?

donpowrie, 2018-03-15:
Sorry, I'm not used to this system...

I have a STM32L073RZT6 running with no external crystals. I started the project with MBED and exported to Kyle uVision.  The project works great and now I need to be able to put it to sleep for 3 seconds, wake the L073 to power up an RF transceiver to check for incoming packets, and then go back to sleep. I cannot leave the transceiver in low-power RX mode and use its interrupt output pin for incoming packets (takes too much power) so I must be able to achieve low-power (<1mA) operation via a sleep mode that the L073 can wake from using an internal peripheral. 

If I call sleep() by itself the L073 goes to sleep and current draw for the system is great at 300uA. Problem is I currently cannot find a way wake the L073 from this sleep.

With this code I can use the sleep() function and use an interrupt to wake up but the current draw is about 5mA during sleep:

bool expired; //global
void int_acq1()  {	expired=1;    } //interrupt routine

expired=0;
Ticker MyTimer;
MyTimer.attach(&int_acq1, 3.0); //3 second tick
while(expired==0) 	sleep();

Questions:
1. Why does sleep() behave differently, depending on the preceding line of code?  If there is no Ticker code preceding the call to sleep then sleep is a very deep (300uA) sleep that will require wake with interrupt via GPIO pin (since all timers/peripherals in the L073 are off).  With the Ticker code preceding the call to sleep() then current consumption is too high (5mA).
2. Where can I find documentation that explicitly explains this characteristic of sleep()?  I have searched MBED and countless other places and found much on the sleep() function - but nothing on this specific topic of preceding code.
3. Is what I'm wanting to do even possible with MBED?

Thanks,
Don

donpowrie, 2018-03-15:
I exported from MBED months ago.  How do I tell from within Keil uVision what the version is for the MBED?

0xc0170, 2018-03-15:
You are working offline then. Look at `mbed.h` or check git sha (that can also help). What is important is to know if you use mbed 2 or mbed OS 5 (basically to answer rtos included or not).

donpowrie, 2018-03-15:
/* mbed Microcontroller Library
 * Copyright (c) 2006-2013 ARM Limited
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0

I'm not intending to use an RTOS.

donpowrie, 2018-03-15:
#ifndef MBED_H
#define MBED_H

#define MBED_LIBRARY_VERSION 153

#if MBED_CONF_RTOS_PRESENT
// RTOS present, this is valid only for mbed OS 5
#define MBED_MAJOR_VERSION 5
#define MBED_MINOR_VERSION 6
#define MBED_PATCH_VERSION 2

#else
// mbed 2
#define MBED_MAJOR_VERSION 2
#define MBED_MINOR_VERSION 0
#define MBED_PATCH_VERSION MBED_LIBRARY_VERSION
#endif

0xc0170, 2018-03-15:
Using mbed 2, library version 153.

`Ticker ` is running on high speed freq clock, if you defined that object, it is enabled and running , thus going to sleep would be affected (the power consumption). If you do not high precision ticker, use low power ticker (in your example 3 seconds wake up), I would rather use Low power ticker in that case.

donpowrie, 2018-03-15:
I tried that.  When I declare LowPowerTicker the current draw is even higher in sleep mode at 9-12mA.  

LowPowerTicker MyTimer;
expired=0;
MyTimer.attach(&int_acq1, 6.0); 
while(expired==0) sleep();

Am I using LowPowerTicker correctly?

0xc0170, 2018-03-15:
> I tried that. When I declare LowPowerTicker the current draw is even higher in sleep mode at 9-12mA.

Is that entire code snippet how to reproduce this draw on `STM32L073` ? Based on the description, you have own target based on STM32L073  MCU?

@ARMmbed/team-st-mcd Please review

jeromecoutant, 2018-03-15:
Hi

Lots of work have been done since 5.6.2 version

Please rebase to the latest version

Thx

donpowrie, 2018-03-15:
Yes, this is the entire code for sleep mode.  See my first post in this thread for the ISR.

Yes, my PCB is very small and coin cell powered.  It has the STM32, an SX1280 based RF transceiver and a temperature sensor that is powered by a port pin but is powered off.  The transceiver is also in sleep mode.  If I call sleep() by itself the current draw from the coin cell is 300uA.

Please advise how I go about rebasing to to the latest version of MBED.  Is there a document I can read for the procedure?

donpowrie, 2018-03-15:
The STM32 is under the RFS1280 module.
[image]

donpowrie, 2018-03-15:
jeromecoutant:
Please advise how I go about rebasing to to the latest version of MBED. Is there a document I can read for the procedure?

bcostm, 2018-03-16:
> I started the project with MBED and exported to Kyle uVision

I think you should redo this. Take care to use the latest mbed library before running the exporter command. It will be the easiest way.

Or you can use the offline "mbed export" command to generate the Keil project.

See the documentation:
https://os.mbed.com/docs/latest/tutorials/arm-mbed-os-quickstart.html
https://os.mbed.com/docs/latest/tutorials/quick-start-offline.html

donpowrie, 2018-03-16:
bcostm-
Thank you for your reply.  I'll try this again.  Hopefully the sleep command will be working in the later version of MBED.  

I would still like to know where I might find information on the two personalities of the sleep command.  See above for details.

donpowrie, 2018-03-16:
I am now running the latest version of MBED (5.7.6 or 159 in mbed.h).  

The sleep command has not changed in its ability to enter and exit sleep mode.  If Sleep follows a timer (Ticker or LowPowerTicker) command then the lowest current draw is about 6mA.  If called all by itself, sleep() successfully enters a low-power state with my system only draws 300-500uA.  

When the ticker expires normal operation resumes which is great.  If no ticker is used I have no way (yet) to wake from sleep.  I need a lowpowertimer or LowPowerTicker or something else that actually works.

Any ideas?

0xc0170, 2018-03-21:
> The sleep command has not changed in its ability to enter and exit sleep mode. If Sleep follows a timer (Ticker or LowPowerTicker) command then the lowest current draw is about 6mA. If called all by itself, sleep() successfully enters a low-power state with my system only draws 300-500uA.

If you use LowPowerTicker, can you look at what is blocking deepsleep ? or is it actually blocked - Does your platform support tickless ? I think that is blocking the deepsleep.

What I do not yet understand, calling sleep() vs going idle (that also invokes sleep).

donpowrie, 2018-03-21:
Thank you for the reply.  Please see my questions inline:

If you use LowPowerTicker, can you look at what is blocking deepsleep ? or is it actually blocked?
-->How do I check what is blocking deepsleep? 

Does your platform support tickless ? I think that is blocking the deepsleep.
-->How would I know?  I am now running the latest version of MBED (5.7.6 or 159 in mbed.h) and compiling with the Keil uVision compiler. Is there a test I can try?

What I do not yet understand, calling sleep() vs going idle (that also invokes sleep). They should be the same regarding the draw
-->How do I enable Idle?  Is this a function call like sleep()?

jeromecoutant, 2018-03-30:
Hi

Maybe have a look on https://os.mbed.com/docs/v5.8/reference/power-management.html

You can see, for ex, that ticker is locking deep sleep, and that in most cases, you don't need to call sleep() directly, Mbed OS enters sleep mode automatically any time the system is idle.

atoy40, 2018-04-09:
@donpowrie @jeromecoutant sleep() calls explicitly allows you to go to deepsleep if nothing else locked it. But It's not the case for the (default) idle handler(for exemple if you use wait function) which lock deepsleep. I've test deepsleep with a L073RZ nucleo board and found about 70uA.

deep sleep lock code in idle handler :
https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/mbed_rtx_idle.cpp#L129

Only "MBED_TICKLESS"-ready targets are allowed to go to deepsleep when idle. Is support for TICKLESS scheduling planned for STM boards ?

Anthony.

LMESTM, 2018-04-09:
TICKLESS support on STM32 targets is planned for next mbed release (mbedOS 5.9, in June timeframe). There are ongoing modifications, not only in STM targets but also in mbed core.   For example, but not limited,   #6474 #6561 #6558 #6534 #6536  will be needed.

jeromecoutant, 2018-06-06:
Hi @donpowrie and all
All tickers updates have been merged.
Could you try again with the master branch ?
Thx

cmonr, 2018-06-21:
This appears to have been fixed with recent PRs.
Closing since no reply has been gived for a while.

closed 2018-06-21 by cmonr