ARMmbed/mbed-os #6366: I2C failure on FRDM-K82F

maclobdell, 2018-03-15:
### Description
- Type: bug
- Related issue: n/a
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
FRDM-K82F

**Toolchain:**
Tested GCC_ARM, but likely to impact others.

**mbed-os sha:**
5.7.6  (91e6db1ea251ffcc973001ed90477f42fdca5751)

**Expected behavior**
This should work.
```
#include "mbed.h"
#include "LM75B.h"

DigitalOut led1(LED1);

int main() {
  LM75B  temperature(D14, D15);
  float temp = temperature.temp();
  printf("\r\n****\r\nTEST LM75B : Temperature Read = `%f`\r\n****\r\n",temp);;
}
```

**Actual behavior**
Instead it results in an assert message to the terminal

```
assertion "masterConfig && srcClock_Hz" failed: file "..\mbed-os\targets\TARGET_Freescale\TARGET_MCUXpresso_MCUS\TARGET_K82F\drivers\fsl_i2c.c", line 474, function: I2C_MasterInit
```

**Steps to reproduce**

mbed import mbed-os-example-blinky
mbed add https://developer.mbed.org/users/neilt6/code/LM75B/#7ac462ba84ac
update main.cpp with above
mbed compile -m K82F -t GCC_ARM -f
open terminal at 9600 baud

cc @ARMmbed/team-nxp

maclobdell, 2018-03-15:
This is fixed by https://github.com/ARMmbed/mbed-os/pull/6310/.  Just tested it and solves the problem.

cmonr, 2018-03-15:
Closing as #6310 is now merged.

closed 2018-03-15 by cmonr