ARMmbed/mbed-os #6380: HAL Flash sector erase doesn'y work consistently

normads, 2018-03-16:
I'm emulating an EEPROM on STM32F746ZG (nucleo board) using sector 6 and 7 and HAL_FLASHEx_Erase(&pEraseInit, &SectorError) does not consistently erase the sector. 
checking whether the sector is erased and erasing again in a loop usually works after 1 or 2 iterations and sometimes rarely takes upto 5 iterations. 

GCC_ARM

0xc0170, 2018-03-20:
@normads What is mbed OS version (git sha?) ?

I assume you do not use FlashIAP but directly HAL_FLASHEx functions?

@ARMmbed/team-st-mcd Please review

normads, 2018-03-21:
mbed version: https://github.com/ARMmbed/mbed-os/#ca661f9d28526ca8f874b05432493a489c9671ea
Yes I'm using HAL_FLASHEx functions directly
In order to check if the sector is erased I'm reading the base address to check if it's 0xFFFFFFFF and sometimes it would be read as erased in consecutive reads even without calling sector erase again. 
But adding a delay doesn't ensure the erase either. 
Which made me think that it could be a caching issue, so i tried disabling ART with __HAL_FLASH_ART_DISABLE() with no improvement

bcostm, 2018-03-21:
I have retested FlashIAP tests to be sure everything in ok on master branch.

```
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
| target                | platform_name | test suite                  | result | elapsed_time (sec) | copy_method |
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-flashiap | OK     | 23.1               | default     |
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+-----------------------+---------------+-----------------------------+-----------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                  | test case                         | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+-----------------------------+-----------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-flashiap | FlashIAP - init                   | 1      | 0      | OK     | 0.03               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-flashiap | FlashIAP - program                | 1      | 0      | OK     | 6.9                |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-flashiap | FlashIAP - program across sectors | 1      | 0      | OK     | 2.07               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-flashiap | FlashIAP - program errors         | 1      | 0      | OK     | 0.05               |
+-----------------------+---------------+-----------------------------+-----------------------------------+--------+--------+--------+--------------------+
```

Can you share your code ?

You can also check on the STM32 forum (https://community.st.com/community/stm32-community/stm32-forum) just in case ?

normads, 2018-03-21:
Here's a minimal code to replicate: 
`
#include "mbed.h"

#define SECTOR_7_BASE_ADDRESS 0x080C0000

DigitalOut led1(LED1);

Serial pc(SERIAL_TX, SERIAL_RX, 115200);

FLASH_EraseInitTypeDef pEraseInit;
uint32_t SectorError = 0;

HAL_StatusTypeDef FlashStatus = HAL_OK;

int main() {

    __HAL_FLASH_SET_LATENCY(7U);

    HAL_FLASH_Unlock();

    for(int i=0;i<10;i++) {

        pc.printf("Erasing Sector 7 - Iteration %d\r\n", i);

        pEraseInit.TypeErase = TYPEERASE_SECTORS;
        pEraseInit.Sector = 7;
        pEraseInit.NbSectors = 1;
        pEraseInit.VoltageRange = FLASH_VOLTAGE_RANGE_3;

        FlashStatus = HAL_FLASHEx_Erase(&pEraseInit, &SectorError);

        if (FlashStatus != HAL_OK) {
            return FlashStatus;
        } else {
            pc.printf("Erase - No error flags\r\n");
        }

        while((*(__IO uint16_t*)SECTOR_7_BASE_ADDRESS) != 0xFFFF) {
            pc.printf("Sector not Erased.. Read %X\r\n",  (*(__IO uint16_t*)SECTOR_7_BASE_ADDRESS));
            FlashStatus = HAL_FLASHEx_Erase(&pEraseInit, &SectorError);
            if (FlashStatus != HAL_OK) {
                return FlashStatus;
            } else {
                pc.printf("Erase - No error flags\r\n");
            }
        }
        pc.printf("Sector Erased..\r\n");

        FlashStatus = HAL_FLASH_Program(FLASH_TYPEPROGRAM_HALFWORD, SECTOR_7_BASE_ADDRESS, 0x0000);

        if (FlashStatus != HAL_OK) {
            pc.printf("Write failed - %d\r\n", HAL_FLASH_GetError());
            return 0;
        } else  {
            pc.printf("Write - No error flags\r\n");
        }

        if((*(__IO uint16_t*)SECTOR_7_BASE_ADDRESS) != 0x0000) {
            pc.printf("Sector not Written - %d\r\n", (*(__IO uint16_t*)SECTOR_7_BASE_ADDRESS));
            return 0;
        } else {
            pc.printf("Write successful\r\n");
        }

        wait(5.0);
    }
}

`

And this is the output
`
Erasing Sector 7 - Iteration 0
Erase - No error flags
Sector Erased..
Write - No error flags
Write successful
Erasing Sector 7 - Iteration 1
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 0
Erase - No error flags
Sector not Erased.. Read 
`
However with the rest of my code running the outcome is not this consistent and it manages to erase more often.

bcostm, 2018-03-21:
First, why you don't use mbed FlashIAP API ? Can you try ?
https://os.mbed.com/docs/v5.7/reference/flash-iap.html

Second, as you use only ST HAL code, maybe some portion of code is missing. May I suggest to restart from CubeF7 FLASH_EraseProgram example. It should work. Then add on top of it your while loop and see if you have the same error ?

normads, 2018-03-22:
> First, why don't you use mbed FlashIAP API ? Can you try ?
> https://os.mbed.com/docs/v5.7/reference/flash-iap.html

I wasn't sure if you could program individual addresses with FlashIAP. According to the api doc you can only program sizes that are multiples of sector size. This is not very efficient for an emulated eeprom application

> Second, as you use only ST HAL code, maybe some portion of code is missing. May I suggest to restart from CubeF7 FLASH_EraseProgram example. It should work. Then add on top of it your while loop and see if you have the same error ?

From a glance i can't see a difference between the cube example but I'll try. Thanks

normads, 2018-03-23:
So the issue was actually in caching, while i first tried disabling ART, ART only effects the ITCM bus and i was reading the flash through the AXIM interface. So it was the L1 D-Cache that needed to be invalidated. 

It should be noted that in the HAL_Flash driver description it mentions that it implements cache resetting functions, 
> In addition to these functions, this driver includes a set of macros allowing
>       to handle the following operations:
>        (+) Set the latency
>        (+) Enable/Disable the prefetch buffer
>        (+) **Enable/Disable the Instruction cache and the Data cache**
>        (+) **Reset the Instruction cache and the Data cache**
**Enable/Disable the Instruction cache and the Data cache** & **Reset the Instruction cache and the Data cache** are actually not implemented in the HAL_Flash interface for F7 devices. This is not controlled by the FLASH_ACR register for F7 devices and has therefore been moved to the CMSIS M7 core (core_cm7.h) 

So calling **SCB_InvalidateDCache()** after flash erase solves the problem. 

This probably does not effect the FlashIAP interface because this error only occurs when a read is performed after a erase without a preceding write to that particular address. Because a write would update the D-Cache.

Thanks a bunch for the support guys!

normads, 2018-03-23:
SCB_InvalidateDCache() isn't implemented in the cortex M4 core. For a F4 device i think the call should be __HAL_FLASH_DATA_CACHE_RESET() which is implemented in the HAL flash driver. 
I'm not very familiar with the MBED architecture to suggest the best way forward. 
But i think there should atleast be a mention of this in the FlashIAP doc to guide someone who might potentially run into an issue in the right direction.

normads, 2018-03-26:
I just had a proper look at the source tree and it seems that the flash api is not implemented for F7 devices?
And in the F4 implementation cache is reset before a program operation, which doesn't make much sense. Because a write operation would anyway update the cache. Cache in my opinion should be reset after the erase operation. 
If someone could confirm my thinking i can send a PR. And perhaps port it to F7?

jeromecoutant, 2018-06-06:
@normads 
could you check https://github.com/ARMmbed/mbed-os/pull/7088
and maybe close this issue ?
Thx

normads, 2018-06-07:
@jeromecoutant I'm afraid #7088 is unrelated. 
This issue is in the flash api and i don't think there's a test case for it.

0xc0170, 2018-06-22:
@jeromecoutant If this issue about  HAL_FLASH, should it be moved somewhere else?

screamerbg, 2018-09-10:
@0xc0170 Can we remove the ST tag from this issue? Seems that this is rather an issue/feature of the Flash API.

screamerbg, 2018-09-13:
@normads Can you clarify the status of this using the FlashIAP and the latest Mbed OS (5.9.4)? Otherwise I'll have to close the issue in 1 week due to inactivity and being outdated.

PR #10248 (2019-04-01): TARGET_STM32F7: Refresh cache when erasing or programming flash
TARGET_STM32F7: Refresh cache when erasing or programming flash
    
The cache must be refreshed when we erase or program flash memory.
It fix 2 issues :
    Fix #9934
    Fix #6380
    
This solution was initially proposed in #6380.

Solution tested on mbed-os-5.12.0 and STM32F746 Disco board with this command: 
    mbed test -t ARM -m DISCO_F746NG -v -n *features-storage-tests-blockdevice-general_block_device* -v
commit 5765c4d0 (2019-04-01): TARGET_STM32F7: Refresh cache when erasing or programming flash

The cache must be refreshed when we erase or program flash memory.
It fix 2 issues :
    Fix #9934
    Fix #6380

This solution was initially proposed in #6380.
closed 2019-04-01 by cmonr