ARMmbed/mbed-os #6382: LPCXpresso54114 has no PWMOUT - but it's on the hardware

janjongboom, 2018-03-16:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
LPCXpresso54114

**Toolchain:**
N/A

**mbed-os sha:**
Mbed OS 5.7.5

**Expected behavior**

In targets.json this board is marked as not having PWM: https://github.com/ARMmbed/mbed-os/blob/7c30faf69de11c84acc5853ed322e8b67a4ce33a/targets/targets.json#L736

But that's not correct, the board should support PWM, it's even listed on the board page at https://os.mbed.com/platforms/LPCXpresso54114/. Any idea what's going on?

https://os.mbed.com/questions/80523/PWM-doesnt-work-with-LPCxpresso-54114-Po/

0xc0170, 2018-03-19:
I do not see it implemented for any of  `MCUXpresso_MCUS`, not  yet been ported.

@ARMmbed/team-nxp Please review

mmahadevan108, 2018-03-26:
Yes, this needs to be added. We will work on submitting a patch in the coming days.

0xc0170, 2018-06-22:
> Yes, this needs to be added. We will work on submitting a patch in the coming days.

I can't locate PR, not yet fixed?

janjongboom, 2018-07-09:
@mmahadevan108 ^

pleonhardt, 2019-11-17:
I'm curious if any progress has been made on this bug.  Is this something that can be added to the MCUXpresso CPUs?

Thanks.

0xc0170, 2019-11-22:
@mmahadevan108 Has this been added?

closed 2021-10-20 by ciarmcom