ARMmbed/mbed-os #6383: How can I get STM32's IRQ and re-write it

savent404, 2018-03-17:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question

## Question

**I want to write a sdio driver for SDFileSystem, How can I get SDIO's IRQ handle in mbed?**
My way is transplant from STM32F411RE's SD BSP(using SDIO DMA mode with 

I can write interrupt handle but this function is at 0x0800xxxx, and I found nvic vector is at 0x020000xx, so the function i wrote would never be called.

0xc0170, 2018-03-19:
mbed uses dynamic vectors, thus using NVIC SetVector/GetVector API should work for you. Just need to know IRQn number and set your own handler plus set properly hw to trigger interrupts. 

You can look some HAL implementation how it does handle interrupts (_init function mainly)

savent404, 2018-03-19:
That's good help, I just re-target NVIC handle in startup.s as usual.

bcostm, 2018-03-20:
Are you OK to close this issue or do you need more help ?

jeromecoutant, 2018-06-06:
@0xc0170 could you close this issue ?
Thx

closed 2018-06-22 by 0xc0170