ARMmbed/mbed-os #6385: LoRaWAN HAL does not support sub-band selection in US915 channel plan

janjongboom, 2018-03-17:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
All

**Toolchain:**
N/A

**mbed-os sha:**
Mbed OS 5.8 RC

**Expected behavior**

US915 uses 64 channels, but most gateways only support 8. Thus the network uses a sub-band of US915 to operate the LoRa network on. We should be able to specify which sub-band to use in code. If you use OTAA it will *eventually* find a right frequency (because it just hops around and at some point a join succeeds) but this is power inefficient and might take a long time. In ABP mode you'll see 7/8 packets lost, depending where we jump.

I hacked around this by removing all channels except the ones in the sub-band that I use (by modifying the source code of the stack) but that's not a feasible solution. We need an API to specify sub-band in US915 frequency plan. MT stack provides this for example, and I believe LMiC has it too.

This is easy to replicate. Take US915 gateway, connect it to TTN, use ABP to send data to the network. Only if channel index 8..15 are used do messages arrive, the rest of messages is lost.

@hasnainvirk @sg- I'd say this should be fixed in Mbed OS 5.8.1 because the stack appears broken for anyone trying to use it in the US.

AnttiKauppila, 2018-04-06:
ARM internal ref: IOTCELL-753

janjongboom, 2018-06-02:
@hasnainvirk @kivaisan Any update here?

kivaisan, 2018-06-04:
Since this is non-standard requirement, I'm now that confident that we need to support this from LoRaWANBase API level. But I think we should enable application to provide it's own region object for lora stack. That way only a specific region API can be expanded to support custom requirements.

Actually we have already planned this a bit as we need it to expand our tests.

AnttiKauppila, 2018-06-04:
Actually the regional standard (both old and new) says: "Personalized devices SHALL have all 72 channels enabled following a reset." So reading between the lines it is not a MUST to have all 72 channels.

The new spec also have changed according to CFList handling. 1.0.2rB Says that CFList is ignored if present, but 1.0.3 says that CFList is supported

kivaisan, 2018-06-05:
The options then are either to provide my earlier mentioned possibility to provide user's own region or allow modification of default channels.

kivaisan, 2018-06-05:
Here is a proposal for user provided phy object:
https://github.com/kivaisan/mbed-os/commit/d297a724266e4fd6ef9855d8ae4ba52a63ece920

I've been using this with my simulated lora test environment.

AnttiKauppila, 2018-09-13:
@janjongboom This is fixed and issue can be closed

closed 2018-09-14 by janjongboom

brady-aiello, 2018-11-19:
Where is this fixed?

AnttiKauppila, 2018-11-19:
Please see: https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/FSB_Usage.txt