ARMmbed/mbed-os #6386: Update Docs for adding to target config values from mbed_app.json

grbd, 2018-03-18:
### Description
- Type: Enhancement

Hi,
I've recently been trying to get mbed to output a hex file instead of a bin
to make it easier to flash with jlink
(since jlink flash doesn't take elf files, and with bin files you have to specify an offset)
I know there's a value of OUTPUT_EXT you can set, but it's just a case of knowing how to set it.

tried this but no result
```
{
   "config": {
       "OUTPUT_EXT": "hex"
   }
}
```

tried this but since the value OUTPUT_EXT is not already set in the targets.json it fails with
"[ERROR] Attempt to override undefined parameter 'app.OUTPUT_EXT' in 'application[*]'"
```
{
    "target_overrides": {
        "*": {
	    "OUTPUT_EXT": "hex"
        }
    }
}
```

One way around it is to manually hack the mbed-os\targets\targets.json file
then add it in there to the target needed (in this case NUCLEO_F767ZI)
but thats less than ideal if I want to update mbed later on

after digging through the source
I eventually figured out that this works
```
{
   "target_overrides": {
        "*": {
	    "target.OUTPUT_EXT": "hex"
        }
    }
}
```

It seems prefixing the attribute name with "target." allows you to add to the list of configuration values for a given target if they don't already exist
It might be an idea for this to be added to the docs since it's a bit non obvious.

0xc0170, 2018-03-20:
> It might be an idea for this to be added to the docs since it's a bit non obvious.

If this app vs target is not documented well, please send a pull request with this improvement.

I looked at Overriding target attributes section there, and it specifies target prefix for overriding target configuration. Here https://github.com/ARMmbed/Handbook/blob/new_engine/docs/tools/config_system.md#overriding-target-attributes

grbd, 2018-03-20:
Okay since it's in the handbook I'll close this down

closed 2018-03-20 by grbd