ARMmbed/mbed-os #6389: LoRaWAN receive API should not require port

janjongboom, 2018-03-19:
### Description
- Type: Bug | Enhancement

## Enhancement

**Reason to enhance or problem with existing solution**

In the LoRaWAN API you are required to specify the `port` that you can want to 'receive' a message on. Receive between quotes because you're not receiving anything as the message is already received. There is a message or there isn't. Also in LoRaWAN you can **at most** get 1 message, so it's not like you might receive a different message if you pass in a different port. Next this receive function is being called in an RX_DONE event, so we *know* that there is a message. Just give me the message then. Same for the confirmed|unconfirmed flag. This does not make sense.

**Suggested enhancement**

Either;

1. Change `receive` so it passes the port in an out parameter or so.
2. Pass the message and port to the RX_DONE function. 

I prefer 2, because this pull approach does not make any sense in class-B or class-C devices. When there's data there's data so just give it to me.

Also see https://github.com/ARMmbed/mbed-os-example-lorawan/issues/49. This is already causing confusion in users.

@hasnainvirk @kivaisan

kivaisan, 2018-03-20:
I fully agree. Current API makes no sense. In order to minimize the API change, I'd prefer changing receive() method as:

    /** Receive a message from the Network Server.
     *
     * @param port              On return the application port number to which message
     *                          was received.
     *
     * @param data              A pointer to buffer where the received data will be
     *                          stored.
     *
     * @param length            The size of data in bytes
     *
     * @param flags             On return contains message type information, for example:
     *
     *                          MSG_UNCONFIRMED_FLAG = 0x01
     *                          MSG_CONFIRMED_FLAG   = 0x02
...
int16_t receive(uint8_t& port, uint8_t* data, uint16_t length, int& flags);

@AnttiKauppila Can we still do these API changes?

janjongboom, 2018-03-20:
@kivaisan Maybe change it to a structure, because I'd also like stuff like RSSI and SNR of the received packets. If not today, then at a later point.

kivaisan, 2018-03-20:
I would like to keep receive parameter list as short as possible. Would it be better to have a separate method to query signal quality?

kjbracey-arm, 2018-03-20:
I wasn't totally sure about the request - it seemed to me like "there can be only 1 packet" was a current implementation restriction, not fundamental. Why couldn't the stack be buffering 1 or even multiple received packets for each of 2 ports?

There is already a race here - the RX_DONE callback is not synchronous, it's on an event queue. It's not necessarily true that the packet you got the RX_DONE for is still available - another packet could have pushed it out. I'm not sure how much of a problem that would be.

The basic question here is whether this API is supposed to be so low level that any socket buffering would be done at the level above it (like early Nanostack->NSAPI glue), or is it supposed to be a socket-like interface where buffering could potentially be happening in the stack? Are we expecting two totally independent people to be making the "receive" call, each specifying the port they're interested in, like a socket, or one person making it, who is then presumably acting as a socket dispatcher?

For Nanostack we originally only had this unbuffered API where the receive socket call had to be made synchronously inside the RX callback. That was synchronous, and if the callback didn't issue it, the data was chucked. It was "push pretending to be pull" To simplify the glue layers and make TCP flow control work better, we later added optional socket buffers, so it really could be used as "pull".

This is kind of somewhere in the middle - it's not synchronous, but it's not really properly buffered either.

Looking at it, I don't think you've really got enough there to properly treat this as a socket-like API - there's no mechanism to "open" a socket/port such that it starts buffering on that port, for example. Passing the port number on receive is too late, really. 

So given that, keeping it basically unbuffered and changing to telling you the port+flags on receive as suggested seems reasonable. Some layer above could arrange to start buffering received packets per socket/port - you could either invent a `LoRaWANSocket` or `LoRaWANEndpoint`, or go straight to a `Socket` implementation written on `LoRaWANInterface`.

Current receive API is already finalised for 5.8, but new receive API could be added as an overload. If doing it as a new overload, can't do it via references as it would conflict, and output references have generally been stylistically frowned on anyway. Should be pointers.

Would be nice to eliminate the race somehow.

kivaisan, 2018-04-13:
@janjongboom PR got merged, so can we close this issue? We can check the need for RSSI and SNR later like you suggested above.

closed 2018-04-17 by janjongboom