ARMmbed/mbed-os #6390: Need build profiles in Online Compiler

janjongboom, 2018-03-19:
### Description
- Type: Enhancement

## Enhancement

**Reason to enhance or problem with existing solution**

Build profiles are super useful, you can use them to enable c++11 or switch to newlib-nano. Also they are required for release builds. I want them in the online compiler too. E.g. DISCO-LRWAN1 target can basically only be built if you do some tricks like newlib-nano, which makes developing in the online compiler almost impossible.

**Suggested enhancement**

Add a dropdown in the online compiler to switch between profiles, and also allow profile customization.

**Pros**

C++11 and newlib-nano!

**Cons**

No idea how much time it'd take.

geky, 2018-03-19:
cc @screamerbg, @theotherjimmy, not sure where this issue should be filed.

theotherjimmy, 2018-03-19:
@thegecko I think this may be your department

0xc0170, 2018-05-22:
> @thegecko I think this may be your department

@thegecko Is this something we want to take further ?

loverdeg-ep, 2018-05-24:
Personally I'd prefer this be, **won't fix**, if a total replacement for the online IDE is coming, and if significant effort is required of anybody to implement it.

thegecko, 2018-05-24:
Thanks for your input. I'll discuss this with the team as it shouldn't affect any IDE frontend

thegecko, 2018-06-06:
Request tracked internally

closed 2018-06-06 by thegecko