ARMmbed/mbed-os #6391: LoRaMac.cpp resets antenna gain and max EIRP wrong

maxgerhardt, 2018-03-19:
### Description
- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug

The LoRaMac.cpp resets the parameters `_params.sys_params.max_eirp` and `_params.sys_params.antenna_gain` in the wrong way. In one part it uses `phy_param.value` (`uint32_t` interpretation of the union) and in another place it uses the `phy_param.f_value` interpretation (`float` interpretation)

LoRaMac.cpp:1745 (initialize())

```cpp
    get_phy.attribute = PHY_DEF_MAX_EIRP;
    phy_param = lora_phy->get_phy_params(&get_phy);
    _params.sys_params.max_eirp = phy_param.f_value;

    get_phy.attribute = PHY_DEF_ANTENNA_GAIN;
    phy_param = lora_phy->get_phy_params(&get_phy);
    _params.sys_params.antenna_gain = phy_param.f_value;
```

LoRaMac.cpp:1338 (reset_mac_parameters())

```cpp
    get_phy.attribute = PHY_DEF_MAX_EIRP;
    phy_param = lora_phy->get_phy_params(&get_phy);
    _params.sys_params.max_eirp = phy_param.value;

    get_phy.attribute = PHY_DEF_ANTENNA_GAIN;
    phy_param = lora_phy->get_phy_params(&get_phy);
    _params.sys_params.antenna_gain = phy_param.value;
```

**Actual behavior**

In my case (EU868 PHY selection) it causes the `LoRaMac::initialize()` function to first initialize the antenna gain correctly to `2.15f` and then reset it to the abnormally large `uint32_t` representation of 1.07*10^9, which consequently causes it to compute  `phy_tx_power` to -128dB in `LoRaPHY::tx_config`.

**Expected behaviour**
The interpretation should in both cases be `float`, i.e. the code should use `phy_param.f_value` in all cases in antenna gain and max EIRP.

**Steps to reproduce**
Use the https://github.com/ARMmbed/mbed-os-example-lorawan/ code and step through the `LoRaMac::initialize` function and watch the global `_params` object.

(Debugger screenshot: https://i.imgur.com/uQVJvRN.png)

0xc0170, 2018-03-20:
@hasnainvirk @kivaisan Please review

kivaisan, 2018-03-20:
@maxgerhardt Yes, looks like a clear bug. I'll prepare a fix for it.

maxgerhardt, 2018-05-06:
Was this fixed and merged so I can close this? I didn't see a reference to this issue number in the release notes.

kivaisan, 2018-05-07:
@maxgerhardt Yes, this has been merged.

closed 2018-05-07 by maxgerhardt