ARMmbed/mbed-os #6395: mbed-os-example-lorawan fails to export in online compiler (all make_* targets on F411RE)

janjongboom, 2018-03-20:
@theotherjimmy 

### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
NUCLEO-F411RE, others not tested.

**Toolchain:**
Online Compiler

**mbed-os sha:**
Mbed OS 5.8 RC

**Expected behavior**

Exporting works.

**Actual behavior**

Exporting fails.

See https://github.com/ARMmbed/mbed-os-example-lorawan/issues/53, https://github.com/ARMmbed/mbed-os-example-lorawan/issues/52, https://github.com/ARMmbed/mbed-os-example-lorawan/issues/51.

**Steps to reproduce**

Import https://github.com/armmbed/mbed-os-example-lorawan, then try to export it to one of the make exporters. I'm pretty sure this is supposed to work on F411RE, it's one of our most popular targets.

theotherjimmy, 2018-03-20:
Thanks for consolidating these problems @janjongboom I'll take a look today.

theotherjimmy, 2018-03-20:
Hm. I cannot reproduce this offline.

theotherjimmy, 2018-03-20:
@janjongboom I think this may affect all targets too. It failed for K64F x make_gcc_arm for me

theotherjimmy, 2018-03-20:
Oh, this might be a timeout :-1:

theotherjimmy, 2018-03-22:
upon further investigation this is https://github.com/ARMmbed/mbed-os/pull/6428

hasnainvirk, 2018-04-06:
@janjongboom This should be fixed in https://github.com/ARMmbed/mbed-os/pull/6428/commits 
Could you please test ?

kivaisan, 2018-04-12:
@janjongboom Can we close this issue?

closed 2018-04-12 by janjongboom