ARMmbed/mbed-os #6396: Is malloc() thread-safe in ARMC6?

ccli8, 2018-03-20:
### Description
- Type: Question
- Priority: Major

-----------------------------------------------------------------

## Question

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
ARM|ARMC6

**Toolchain version:**
ARM: 5.06
ARMC6: 6.9

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
f6bcbfe Merge pull request #6356 from hug-dev/asm-include-flags

**How to?**
According to **Arm C and C++ Libraries and Floating-Point Support User Guide**, `malloc()` can be made thread-safe by implementing the \_mutex\_*() functions. I want to check its validity in ARMC6.

I compile Greentea test code `mbed-os-tests-mbedmicro-rtos-mbed-malloc` with **ARM** toolchain and check generated MAP file `malloc.map`. I can find \_mutex\_*() functions (`_mutex_initialize/acquire/release/free`) are included, so I am convinced `malloc()` is thread-safe.
```
mbed test -m NUMAKER_PFM_NUC472 -t ARM -n mbed-os-tests-mbedmicro-rtos-mbed-malloc --compile
```

But then I compile the same code with **ARMC6** toolchain and check generated MAP file `malloc.map`. I just find `_mutex_initialize` and doesn't find `_mutex_acquuire/release/free`. Is `malloc` thread-safe in ARMC6, or there's some other mechanism?
```
mbed test -m NUMAKER_PFM_NUC472 -t ARMC6 -n mbed-os-tests-mbedmicro-rtos-mbed-malloc --compile
```

0xc0170, 2018-03-20:
https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/mbed_boot.c#L442

They are present for ARMCC, can you verify they are compiled for ARMC6 that you are using?

In the user guide for ARMCC 6, it is still defined same API for thread safe allocators (section 1.6.8 Management of locks in multithreaded applications there).

ccli8, 2018-03-21:
> In the user guide for ARMCC 6, it is still defined same API for thread safe allocators (section 1.6.8 Management of locks in multithreaded applications there).

@0xc0170 Yes. But per my compiling with ARMC6 (V6.9), I didn't find `_mutex_acquuire`/`_mutex_release` included in generated MAP file like ARMCC (V5.06). So I question thread-safety of `malloc` with ARMC6 in Mbed OS.

The test `mbed-os-tests-mbedmicro-rtos-mbed-malloc` is to test `malloc` in multiple threads. It is passed in **NUMAKER_PFM_NUC472** with both ARM/ARMC6. But I still want to clarify it to avoid potential thread-safety issue.

pan-, 2018-03-22:
@0xc0170 `_mutex_acquire`  is defined in [RTX](https://github.com/ARMmbed/mbed-os/blob/342e3caeb80b39c57060b9009a5339777c599af3/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.c#L625) I'm not sure to understand why the function is defined as weak; it doesn't seems right.

0xc0170, 2018-03-22:
> @0xc0170 _mutex_acquire is defined in RTX I'm not sure to understand why the function is defined as weak; it doesn't seems right.

I believe that is protected via macro that we define not to be enabled. Same is for some startup hooks that we redefine. I assume that is a reason why is not weakly defined.

cc @JonatanAntoni

ccli8, 2018-04-02:
@0xc0170 @pan- Any update?

JonatanAntoni, 2018-04-03:
Sorry, I don't understand your issue. The multi-thread library integration in RTX is defined weak to allow a redefinition if needed. In case of mbedOS I expect these function to be defined somewhere else to be compliant with your memory allocation style.

deepikabhavnani, 2018-05-23:
@JonatanAntoni - I believe the question here should be why `__USED` define is only set for ARMCC version < 6. https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.c#L623

@ccli8 has submitted a PR for this fix (https://github.com/ARMmbed/mbed-os/pull/6973), will it make sense to upstream the change in CMSIS Repo?

deepikabhavnani, 2018-05-23:
@JonatanAntoni - Sorry my bad, change is done in mbed-os repo only, CMSIS is fine.

geky, 2018-05-23:
Responding to @deepikabhavnani, I believe the `__USED` defines are behind the ifdef because I was only able to test ARMC5 at the time. (ARMC6 has frontend changes and that pr needed to go in quickly). If they work in ARMC6 we should adopt them there as well.

deepikabhavnani, 2018-07-03:
Fix merged to master https://github.com/ARMmbed/mbed-os/pull/6973

closed 2018-07-03 by deepikabhavnani