ARMmbed/mbed-os #6397: NUCLEO_L476RG AnalogIn, want to measure internal temperature sensor.

maskedw, 2018-03-20:
### Description
- Type: Question
---------------------------------------------------------------

## Question
**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc 6.3.1

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
16bac10

-----------------------------------------------------------------

## Question

**How to?**

The temperature is measured with the code below, but it is very inaccurate. (Room temperature is 23℃)

```c++
#include "mbed.h"

Serial pc(USBTX, USBRX);
AnalogIn adc_temp(ADC_TEMP);

int main()
{
    pc.baud(115200);
    int i = 0;
    while(1)
    {
        int temp12 = adc_temp.read_u16() >> 4;
        int temp = __LL_ADC_CALC_TEMPERATURE(3300, temp12, LL_ADC_RESOLUTION_12B);
        printf("TMP%04d: %d\n", i++, temp);
        wait_ms(1000);
    }

    return 0;
}
```

```
TMP0000: 18
TMP0001: 31
TMP0002: 30
TMP0003: 30
TMP0004: 30
TMP0005: 30
TMP0006: 30
TMP0007: 31
...
```

I tried the following sample code.
[STM32CubeL4](http://www.st.com/en/embedded-software/stm32cubel4.html)
STM32Cube_FW_L4_V1.11.0/Projects/NUCLEO-L476RG/Examples_LL/ADC/ADC_TemperatureSensor

About 25 ℃ was measured in this sample code. It is close to the actual temperature.

Can I use mbed::AnalogIn to measure the temperature with practical accuracy?

0xc0170, 2018-03-20:
@ARMmbed/team-st-mcd Please review

jeromecoutant, 2018-03-20:
Hi

Please check #6399 

Regards

maskedw, 2018-03-22:
Thank you. I confirmed that the AD value will be as expected.

closed 2018-03-22 by maskedw