ARMmbed/mbed-os #6398: NUCLEO_L476RG: Measuring ADC_TEMP after using ADC_VREF in AnalogIn results in abnormal results

maskedw, 2018-03-20:
### Description
- Type: Bug
- Related issue: #1685, #4326, #6397

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc 6.3.1

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
16bac10

**Steps to reproduce**
(Room temperature 23 ℃)

```c++
#include "mbed.h"

Serial pc(USBTX, USBRX);

int main()
{
    pc.baud(115200);
    while(1)
    {
        {
            AnalogIn adc_temp(ADC_TEMP);
            wait_ms(1000);
            int16_t temp12 = adc_temp.read_u16() >> 4;
            int32_t temp = __LL_ADC_CALC_TEMPERATURE(3300, temp12, LL_ADC_RESOLUTION_12B);
            printf("TMP: %d\n", temp);
        }
    }

    return 0;
}
```

This measured value is not as expected, but I have posted another issue. #6397
```
TMP: 30
```

```c++
#include "mbed.h"

Serial pc(USBTX, USBRX);

int main()
{
    pc.baud(115200);
    while(1)
    {
        {
            AnalogIn adc_vref(ADC_VREF);
            wait_ms(1000);
            uint16_t vref12 = adc_vref.read_u16() >> 4;
            int vref = (3.0f * (*VREFINT_CAL_ADDR) / vref12 * 1000);
            printf("AVDD: %d\n", vref);
        }

        {
            AnalogIn adc_temp(ADC_TEMP);
            wait_ms(1000);
            int16_t temp12 = adc_temp.read_u16() >> 4;
            int32_t temp = __LL_ADC_CALC_TEMPERATURE(3300, temp12, LL_ADC_RESOLUTION_12B);
            printf("TMP: %d\n", temp);
        }
    }

    return 0;
}
```

After measuring VREFINT, obviously an abnormal value will be returned.
```
TMP: -205
AVDD: 3306
```

0xc0170, 2018-03-20:
@ARMmbed/team-st-mcd Please review

jeromecoutant, 2018-03-21:
Hi
Issue is confirmed. ADC driver team is contacted.
Thx

jeromecoutant, 2018-03-21:
ST_INTERNAL_REF 45122

maskedw, 2018-03-22:
With # 6412 change, we confirmed that both ADC_VREF and ADC_TEMP can be measured normally.

closed 2018-03-22 by maskedw