ARMmbed/mbed-os #6405: Add a blank value API to flash_api

davidsaada, 2018-03-20:
### Description
- Type: Enhancement 
- Priority: Minor

**Reason to enhance or problem with existing solution**
We want to allow code to check whether a specific part of a sector is erased. NVStore code, for instance, does it in initialization. In the vast majority of flash components, the erase value is all ones. However, there are a few boards whose erase value is zero. The STM32L151CB is one of these. 

**Suggested enhancement**
Add a `blank_value` API to flash_api. Should be something like this:
`unsigned char blank_value();`
returning either 0xFF or 0.

**How to?**
This is quite an easy implementation, as the erase value is well known for each component. Should be a one liner function for each board. Once implemented in flash_api, it can be floated via FlashIAP driver to the user.

0xc0170, 2018-03-20:
The current workaround: write some data (not writing 0 or 0xff), and erase the sector and check what value is it.

What would be the use cases ? Will it be to check if sector was erased only? would it be better to create a method for this check rather than erased value? 

@c1728p9 What do you think?

davidsaada, 2018-03-20:
@0xc0170 This is indeed a workaround, but it is destructive. I don't think I'd want to erase a sector on each initialization. Worse than that - I may not have a free sector to check this workaround on, meaning it can destroy essential data.
Use case is not only checking if the sector is fully erased, but also if it is partially erased.  NVStore code, for instance, decides (at init) where to start writing based on the erased portion start.

0xc0170, 2018-03-20:
> Use case is not only checking if the sector is fully erased, but also if it is partially erased. NVStore code, for instance, decides (at init) where to start writing based on the erased portion start.

Then having `get_sector_erase_value()` would make sense.

dannybenor, 2018-08-21:
@davidsaada can we close this issue?

davidsaada, 2018-08-21:
> can we close this issue?

No. It still needs to be added. Not sure whose responsibility it should be though. It can be either implemented by our partner enabling teams, or one person can scan all boards and return the erase value.
In the vast majority of the cases, it should be a one liner returning 0xff (probably 100% of our boards). However, one still needs to check all boards.

closed 2018-11-08 by cmonr