ARMmbed/mbed-os #6407: wait function waits for wrong time on gcc

bmanga, 2018-03-20:
### Description
- Type: Bug 

Using the wait function on mbed-cli results in wrong timing on LPC1768 using arm-gcc, and it seems to wait ~10x the specified amount. The same code works fine via the online compiler.
---------------------------------------------------------------
## Bug

**Target**
LPC1768

**Toolchain:**
GCC_ARM

**Toolchain version:**
7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

**mbed-cli version:** 
1.3.0

**mbed-os sha:**
16bac101a (HEAD, tag: mbed-os-5.7.7, tag: latest, origin/mbed-os-5.7) Merge pull request #6364 from ARMmbed/release-candidate

**Expected behavior**
wait(0.1f) should wait for 100ms
**Actual behavior**
wait(0.1f) waits for about 1s

**Steps to reproduce**
minimal example:
```
#include "mbed.h"
 
DigitalOut led1(LED1);
int main()
{
    while(true) {
        led1 = 1;
        wait(0.1f);
        led1 = 0;
        wait(0.1f);
    }
}
```

0xc0170, 2018-04-11:
> 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

should be 6 but that I dont think would fix this issue

Can you test debug build with gcc , does it have same behavior? Is actually debuger attached?

0xc0170, 2018-05-22:
Closing as no update

closed 2018-05-22 by 0xc0170

fasthazard, 2018-08-23:
Confirmed this after fresh first-time install.
I measured on an oscilloscope that the delay is in fact exactly 10x longer than requested.

Steps to reproduce:

1. mbed import mbed-os-example-blinky
2. cd mbed-os-example-blinky
3. mbed compile -t GCC_ARM -m detect -f (-mLPC1768 no different)
4. Connect oscilloscope probe between GND and right pad of LED1. 
5. Measure positive / negative pulse width: result: 5.00s (should be 0.5s)

mbed help lists version 1.7.2
mbed update lists mbed-os at rev #f8b140f8d7cb (tags: latest, mbed-os-5.9.5)

fasthazard, 2018-08-24:
Possibly duplicates #7381, although it references the online compiler.
I was able to identify a fix for this error.

The slow blinking was observed on two 'older' lpc1768 boards.
Both boards had (factory installed) interface firmware 16457.
The interface firmware version can be found by editing MBED.HTM and examining the embedded link.
After upgrading to the latest available interface firmware (141212), blinking is back to normal speed.

FW updates can be found [here](https://os.mbed.com/handbook/Firmware-LPC1768-LPC11U24 )

Changelog indicates CMSIS-DAP was added in the newest version, so I'm guessing the problem lies in mismatch between mbed-os and interface.

Since MBED.HTM file is accessible via the USB device, it should be possible for mbed detect and/or mbed compile to detect the older firmware version and warn of incompatibility.

0xc0170, 2018-08-24:
@fasthazard Thanks for sharing ! This is helpful