ARMmbed/mbed-os #6409: NVIC_SetVector does nothing on NRF51822

Spivoxity, 2018-03-20:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF51822

**Expected behavior**
NVIC_SetVector(cmsis_nvic.c) should dynamically change an interrupt vector, allowing a handler to be installed at run time.

**Actual behavior**
I haven't tried this, but the code to copy the vector table from flash to RAM (two versions of it) is commented out, so it looks to me like the assignment to a vector slot targets flash and therefore has no effect.  I can't see any other provision being made to copy the vectors to RAM, but would appreciate having it pointed out if there is.

In fact, Nordic application notes seem to indicate that the chip does not have the resources needed to remap the vector table into RAM.  The suggestion there is to map all interrupts to a handler that looks up the interrupt in a secondary, dynamic table.

An implementation for mbed could either follow this suggestion, or not provide NCIC_SetVector at all, either giving a runtime error when it is called or (perhaps preferably) failing at link time.

mbed-os/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/device/cmsis_nvic.c

0xc0170, 2018-03-21:
@marcuschangarm Please review

marcuschangarm, 2018-03-21:
We currently only support vector table relocation using the VTOR register found on newer Cortex devices. 

The NVIC_SetVector function in question is generic for all Cortex-M0, so for platforms that can map RAM to address 0x0 the function would work as expected.

Spivoxity, 2018-03-22:
(As I said in my first post:)

mbed-os/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/device/cmsis_nvic.c

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/device/cmsis_nvic.c

It looks to be a mangled copy of the relevant functions in your one.  I don't think this OS is big enough to include both of them!

Looking back in the history, it seems someone decided to reverse a decision to put NVIC_SetVector in one place, resurrecting this (and possibly other) broken versions.  This is a nightmare for someone reading the code to try and understand the hardware!

marcuschangarm, 2018-03-22:
From the comment inside that file it looks like it's been copy-pasted from the NXP LPC11U. It should probably be removed.

@c1728p9

closed 2018-12-19 by ciarmcom