ARMmbed/mbed-os #6410: Zero delay (attach) test fails for REALTEK_RTL8195AM

studavekar, 2018-03-20:
May be related to https://github.com/ARMmbed/mbed-os/pull/5106

### Description
- Type: Bug
- Priority:   Minor

---------------------------------------------------------------
## Bug

**Target**

REALTEk_RTL8195AM

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
53cb6e0 Merge pull request #6279 from AnttiKauppila/master

**Steps**
mbed test -t GCC_ARM -m REALTEk_RTL8195AM --run -n tests-mbed_drivers-timeout -v -V

```
[1521585481.77][CONN][RXD]
[1521585481.78][CONN][RXD] >>> Running case #9: 'Zero delay (attach)'...
[1521585481.79][CONN][INF] found KV pair in stream: {{__testcase_start;Zero delay (attach)}}, queued...
[1521585481.80][CONN][RXD] :204::FAIL: Expected 1 Was 0
[1521585481.81][CONN][INF] found KV pair in stream: {{__testcase_finish;Zero delay (attach);0;1}}, queued...
[1521585481.83][CONN][RXD] >>> 'Zero delay (attach)': 0 passed, 1 failed with reason 'Assertion Failed'
[1521585481.83][CONN][RXD]
[1521585481.85][CONN][RXD] >>> Test cases: 8 passed, 1 failed with reason 'Assertion Failed'
[1521585481.86][CONN][RXD] >>> TESTS FAILED!
[1521585481.86][CONN][INF] found KV pair in stream: {{__testcase_summary;8;1}}, queued...
[1521585481.87][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1521585481.87][CONN][INF] found KV pair in stream: {{reserved_heap;0}}, queued...
[1521585481.87][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1521585481.867801
[1521585481.87][HTST][ERR] orphan event in main phase: {{reserved_heap;0}}, timestamp=1521585481.867805
[1521585481.88][CONN][RXD]
[1521585481.88][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1521585481.88][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1521585481.88][HTST][INF] __exit(0)
[1521585481.88][HTST][INF] __notify_complete(False)
[1521585481.88][HTST][INF] __exit_event_queue received
[1521585481.88][HTST][INF] test suite run finished after 1.09 sec...
[1521585481.89][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1521585481.92][HTST][INF] CONN exited with code: 0
[1521585481.92][HTST][INF] Some events in queue
[1521585481.92][HTST][INF] stopped consuming events
[1521585481.92][HTST][INF] host test result() call skipped, received: False
[1521585481.92][HTST][INF] calling blocking teardown()
[1521585481.92][HTST][INF] teardown() finished
[1521585481.92][HTST][INF] {{result;failure}}
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', u'REALTEK_RTL8195AM', '-p', '/dev/ttyACM4:9600', '-f', u'"BUILD/tests/REALTEk_RTL8195AM/GCC_ARM/TESTS/mbed_drivers/timeout/timeout.bin"', '-e', u'"TESTS/host_tests"', '-d', u'/mnt/pci-0000_00_14_0-usb-0_6_1_6_1_0-scsi-0_0_0_0', '-c', 'default', '-t', u'460000000989df4000000000000000000000000097969902', '-r', 'default', '-C', '4', '--sync', '5'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'FAIL'
mbedgt: test on hardware with target id: 460000000989df4000000000000000000000000097969902
mbedgt: test suite 'tests-mbed_drivers-timeout' ...................................................... FAIL in 75.53 sec
        test case: '1 s delay accuracy (attach)' ..................................................... SKIPPED in 0.00 sec
        test case: '1 s delay accuracy (attach_us)' .................................................. SKIPPED in 0.00 sec
        test case: '10 ms delay accuracy (attach)' ................................................... SKIPPED in 0.00 sec
        test case: '10 ms delay accuracy (attach_us)' ................................................ SKIPPED in 0.00 sec
        test case: '5 s delay accuracy (attach)' ..................................................... SKIPPED in 0.00 sec
        test case: '5 s delay accuracy (attach_us)' .................................................. SKIPPED in 0.00 sec
        test case: 'Callback called once (attach)' ................................................... OK in 0.04 sec
        test case: 'Callback called once (attach_us)' ................................................ OK in 0.04 sec
        test case: 'Callback not called when cancelled (attach)' ..................................... OK in 0.04 sec
        test case: 'Callback not called when cancelled (attach_us)' .................................. OK in 0.04 sec
        test case: 'Callback override (attach)' ...................................................... OK in 0.04 sec
        test case: 'Callback override (attach_us)' ................................................... OK in 0.05 sec
        test case: 'Multiple timeouts running in parallel (attach)' .................................. OK in 0.03 sec
        test case: 'Multiple timeouts running in parallel (attach_us)' ............................... OK in 0.04 sec
        test case: 'Timing drift (attach)' ........................................................... SKIPPED in 0.00 sec
        test case: 'Timing drift (attach_us)' ........................................................ SKIPPED in 0.00 sec
        test case: 'Zero delay (attach)' ............................................................. FAIL in 0.02 sec
        test case: 'Zero delay (attach_us)' .......................................................... SKIPPED in 0.00 sec
mbedgt: test case summary: 8 passes, 1 failure
mbedgt: utest test case summary mismatch: utest reported passes and failures miscount!
        reported by utest: passes = 8, failures 1)
        test case result count: passes = 8, failures 10)
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.0683939613
mbedgt: test suite report:
```

studavekar, 2018-03-20:
cc : @fkjagodzinski @0xc0170

0xc0170, 2018-03-21:
@ARMmbed/team-realtek Can you review please?

M-ichae-l, 2018-03-22:
Is this part of ci test or greentea test?

studavekar, 2018-03-22:
> Is this part of ci test or greentea test?

This is part of greentea

M-ichae-l, 2018-04-02:
Please see the fix #6522. Discuss about this issue.

0xc0170, 2018-06-22:
> Please see the fix #6522. Discuss about this issue.

This one was closed, but #6536 referenced instead. did this PR fixed this issue ?

samchuarm, 2018-09-26:
@studavekar @0xc0170 , do we know if this issue got fixed? Thx

samchuarm, 2018-10-03:
Hi @0xc0170 , please advise how we should proceed. Are we waiting for Realtek to verify if the issue still exists?

M-ichae-l, 2018-10-05:
This issue will be fixed when the new us/lp ticker api implementation updates (low power feature).

samchuarm, 2018-10-11:
@M-ichae-l , please let us know when you have PR up for us/lp ticker implementation. Thanks.

studavekar, 2018-10-12:
currently above test are not been build so not able to veridy the fix. Below is the list of tests which are skipped for REALTEK_RTL8195AM http://mbed-os.s3-eu-west-1.amazonaws.com/builds/8390/PASS/REALTEK_RTL8195AM/IAR/7252618b08f1db9398e64f068e02cf0d1d6c9ac5_build_log_REALTEK_RTL8195AM_IAR.txt

```
Build skips:
  * REALTEK_RTL8195AM::IAR::FEATURES-CELLULAR-TESTS-API-CELLULAR_DEVICE
  * REALTEK_RTL8195AM::IAR::FEATURES-CELLULAR-TESTS-API-CELLULAR_INFORMATION
  * REALTEK_RTL8195AM::IAR::FEATURES-CELLULAR-TESTS-API-CELLULAR_NETWORK
  * REALTEK_RTL8195AM::IAR::FEATURES-CELLULAR-TESTS-API-CELLULAR_POWER
  * REALTEK_RTL8195AM::IAR::FEATURES-CELLULAR-TESTS-API-CELLULAR_SIM
  * REALTEK_RTL8195AM::IAR::FEATURES-CELLULAR-TESTS-API-CELLULAR_SMS
  * REALTEK_RTL8195AM::IAR::FEATURES-CELLULAR-TESTS-SOCKET-UDP
  * REALTEK_RTL8195AM::IAR::FEATURES-NETSOCKET-CELLULAR-GENERIC_MODEM_DRIVER-TESTS-UNIT_TESTS-DEFAULT
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM-DIRS
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM-FILES
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM-INTERSPERSED
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM-SEEK
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_INTEGRATION-FORMAT
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RECOVERY-RESILIENCE
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RECOVERY-RESILIENCE_FUNCTIONAL
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RECOVERY-WEAR_LEVELING
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RETARGET-DIRS
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RETARGET-FILES
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RETARGET-INTERSPERSED
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RETARGET-SEEK
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-TESTS-FILESYSTEM-FAT_FILESYSTEM
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-TESTS-FILESYSTEM-GENERAL_FILESYSTEM
  * REALTEK_RTL8195AM::IAR::FEATURES-STORAGE-TESTS-FILESYSTEM-MULTIPART_FAT_FILESYSTEM
  * REALTEK_RTL8195AM::IAR::TESTS-EVENTS-QUEUE
  * REALTEK_RTL8195AM::IAR::TESTS-EVENTS-TIMING
  * REALTEK_RTL8195AM::IAR::TESTS-LORAWAN-LORARADIO
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-BASIC
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-CONDITION_VARIABLE
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-EVENT_FLAGS
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-HEAP_AND_STACK
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-MAIL
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-MALLOC
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-MUTEX
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-QUEUE
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-RTOSTIMER
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-SEMAPHORE
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-SIGNALS
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-SYSTIMER
  * REALTEK_RTL8195AM::IAR::TESTS-MBEDMICRO-RTOS-MBED-THREADS
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-LP_TICKER
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-LP_TIMEOUT
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-LP_TIMER
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-MEM_TRACE
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-RACE_TEST
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-RTC
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-SLEEP_LOCK
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-TICKER
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-TIMEOUT
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-TIMER
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_DRIVERS-TIMEREVENT
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-COMMON_TICKERS
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-COMMON_TICKERS_FREQ
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-CRC
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-LP_TICKER
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-QSPI
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-RTC
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-RTC_RESET
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-SLEEP
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-SLEEP_MANAGER
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-SLEEP_MANAGER_RACECONDITION
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_HAL-US_TICKER
  * REALTEK_RTL8195AM::IAR::TESTS-MBED_PLATFORM-STATS_CPU
  * REALTEK_RTL8195AM::IAR::TESTS-NETSOCKET-DNS
  * REALTEK_RTL8195AM::IAR::TESTS-NETSOCKET-TCP
  * REALTEK_RTL8195AM::IAR::TESTS-NETSOCKET-UDP
  * REALTEK_RTL8195AM::IAR::TESTS-NETWORK-EMAC
  * REALTEK_RTL8195AM::IAR::TESTS-NETWORK-INTERFACE
  * REALTEK_RTL8195AM::IAR::TESTS-NETWORK-WIFI
  * REALTEK_RTL8195AM::IAR::TESTS-NFC-EEPROM
  * REALTEK_RTL8195AM::IAR::TEST_APPS-DEVICE-EXAMPLEAPP
  * REALTEK_RTL8195AM::IAR::TEST_APPS-DEVICE-NANOSTACK_MAC_TESTER
  * REALTEK_RTL8195AM::IAR::TEST_APPS-DEVICE-SOCKET_APP
```